espressif/esp-matter #913: Use both esp-matter and esp-idf mDNS simultaneously

jeevithkumarnm, 2024-05-01:
Hi,

I am trying to use both esp-matter mDNS service and a separate esp-idf mDNS service for my application.
They seem to clash and only one service is visible in my LAN.
I also tried to use minimal mDNS for esp-matter and standard esp-idf mDNS for my service. This works sometimes, but not stable. Seems to kill either of the services later down the execution.

What is the best way to advertise both esp-matter and esp-idf services and work seamlessly?

I am calling mdns_init, followed by mdns_hostname_set and then mdns_service_add as given in the documentation for my service broadcast.

jonsmirl, 2024-05-02:
esp-matter is already initializing mdns
https://github.com/project-chip/connectedhomeip/blob/master/src/platform/ESP32/ESP32DnssdImpl.cpp#L138
You probably need to watch the esp-matter startup events and wait until it is running. At that point mdns will be started, then you can add your additional service.

jeevithkumarnm, 2024-05-06:
Thanks for the pointer @jonsmirl!
Waiting for CHIP server ready event and starting my mDNS service helps. Now ESP advertises both Matter UDP and my service.
I am able to access and use my service, whereas Matter commissioning is still a problem.

Before starting to debug further, wanted to know if at all esp-matter works well with esp-idf mDNS component. Surprisingly even the _esp-matter/examples/light_ does not work when I disable USE_MINIMAL_MDNS setting in menuconfig. Which did work seamlessly for commissioning with USE_MINIMAL_MDNS setting enabled.

Looks like esp-matter does not work with esp-idf mDNS component at all.

esp-matter at 'main' --> https://github.com/espressif/esp-matter/commit/b831d1051b9571d982aaa7854e6d839365a414ae
esp-idf at v5.2.1

jeevithkumarnm, 2024-05-06:
One more observation: With USE_MINIMAL_MDNS enabled, Matter advertises as TCP, whereas with USE_MINIMAL_MDNS disabled, it is UDP.
`chip[DIS]: mDNS service published: _matter._tcp`

JonathanWitthoeft, 2024-05-06:
Disabling USE_MINIMAL_MDNS and using the standard esp-idf mdns code worked for me. I am able to have concurrent services. I am using the release/1.2 branch

jeevithkumarnm, 2024-05-15:
Using release branch on esp-matter made things stable. esp-idf mDNS is working fine without issues and MATTER is also able to use it for successful commissioning.
We will stick to v1.2 release for our development. Thanks @JonathanWitthoeft!

esp-idf : a322e6b (Tag v5.2.1)
esp-matter : c5d5835(latest branch release/v1.2)

closed 2024-05-15 by jeevithkumarnm

jzwyssig, 2024-05-28:
I just tested this using release v1.2 from last week (commit 15519d13642715e058eccfa97f83b3b451818230). However, if I disable USE_MINIMAL_MDNS, I am unable to pair the device because the MDNS service is not published. I am using Ethernet, Wi-Fi station support is disabled. I noticed that there are many changes for Espressif's DNSSD in the master branch of connectedhomeip, but those changes are not yet included in the latest commit of esp-matter.