espressif/esp-matter #915: More incorrect attribute defaults

jonsmirl, 2024-05-02:
Temperature Measurement:
https://github.com/espressif/esp-matter/blob/main/components/esp_matter/esp_matter_cluster.h#L630
~~~
    config() : cluster_revision(4), measured_value(), min_measured_value(27315), max_measured_value(32767) {}
~~~
Last two should be NULL.
~~~
2.3.4.2. MinMeasuredValue Attribute
This attribute SHALL indicate the minimum value of MeasuredValue that is capable of being mea­
sured. See Measured Value for more details.
The null value indicates that the value is not available.
2.3.4.3. MaxMeasuredValue Attribute
This attribute indicates the maximum value of MeasuredValue that is capable of being measured.
See Measured Value for more details.
The null value indicates that the value is not available.
~~~

Relative Humidity
https://github.com/espressif/esp-matter/blob/main/components/esp_matter/esp_matter_cluster.h#L642
~~~
    config() : cluster_revision(3), measured_value(), min_measured_value(0), max_measured_value(10000) {}
~~~
Last two should be NULL
~~~
2.6.4.2. MinMeasuredValue Attribute
The MinMeasuredValue attribute indicates the minimum value of MeasuredValue that can be mea­sured. 
The null value means this attribute is not defined. See Measured Value for more details.
2.6.4.3. MaxMeasuredValue Attribute
The MaxMeasuredValue attribute indicates the maximum value of MeasuredValue that can be mea­sured. 
The null value means this attribute is not defined. See Measured Value for more details.
~~~

More auditing of these default values is needed, I keep finding problems.

jadhavrohit924, 2024-05-08:
Thanks for the report. I'll fix it.

dhrishi, 2024-05-15:
Fixed with https://github.com/espressif/esp-matter/commit/2ef5a6049c84c7da271061437bdb2ee50aea503e

closed 2024-05-15 by dhrishi