espressif/esp-matter #916: Matter light Example broken

Haerteleric, 2024-05-03:
**Describe the bug**
Build fails because of examples/light/CMakeLists.txt:17.
```
include(${CMAKE_CURRENT_LIST_DIR}/../common/cmake_common/components_include.cmake)
```
Removing the line fixes the problem.

**Environment**
- ESP-Matter Commit Id: b84a2cc
- ESP-IDF Commit Id: a322e6bdad4b6675d4597fb2722eea2851ba88cb
- SoC (eg: ESP32 or ESP32-C3): ESP32-C6

**Build Log**
```
Executing task: cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 -B /home/phoscon/light/build -S /home/phoscon/light 

-- Found Git: /usr/bin/git (found version "2.39.2") 
CMake Error at CMakeLists.txt:17 (include):
  include could not find requested file:

    /home/phoscon/light/../common/cmake_common/components_include.cmake

-- Component directory /home/phoscon/.espressif/esp-matter/examples/common/blemesh_platform does not contain a CMakeLists.txt file. No component will be added
-- Component directory /home/phoscon/.espressif/esp-matter/examples/common/cmake_common does not contain a CMakeLists.txt file. No component will be added
-- Component directory /home/phoscon/.espressif/esp-matter/examples/common/external_platform does not contain a CMakeLists.txt file. No component will be added
-- Component directory /home/phoscon/.espressif/esp-matter/examples/common/relinker does not contain a CMakeLists.txt file. No component will be added
-- Component directory /home/phoscon/.espressif/esp-matter/examples/common/utils does not contain a CMakeLists.txt file. No component will be added
-- The C compiler identification is GNU 13.2.0
-- The CXX compiler identification is GNU 13.2.0
-- The ASM compiler identification is GNU
-- Found assembler: /home/phoscon/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/xtensa-esp32-elf-gcc
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /home/phoscon/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/xtensa-esp32-elf-gcc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /home/phoscon/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/xtensa-esp32-elf-g++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Building ESP-IDF components for target esp32
NOTICE: Skipping optional dependency: espressif/cmake_utilities
Processing 29 dependencies:
[1/29] espressif/button (3.2.0)
[2/29] espressif/cbor (0.6.0~1)
[3/29] espressif/cmake_utilities (0.5.3)
[4/29] espressif/esp-serial-flasher (0.0.11)
[5/29] espressif/esp_bsp_generic (1.1.2)
[6/29] espressif/esp_diag_data_store (1.0.1)
[7/29] espressif/esp_diagnostics (1.0.2)
[8/29] espressif/esp_encrypted_img (2.1.0)
[9/29] espressif/esp_insights (1.0.1)
[10/29] espressif/esp_lcd_gc9a01 (1.2.0)
[11/29] espressif/esp_lcd_ili9341 (1.2.0)
[12/29] espressif/esp_lcd_touch (1.1.1)
[13/29] espressif/esp_lcd_touch_cst816s (1.0.3)
[14/29] espressif/esp_lcd_touch_ft5x06 (1.0.6)
[15/29] espressif/esp_lcd_touch_gt1151 (1.0.5~1)
[16/29] espressif/esp_lcd_touch_gt911 (1.1.0)
[17/29] espressif/esp_lcd_touch_tt21100 (1.1.0)
[18/29] espressif/esp_lvgl_port (1.4.0)
[19/29] espressif/esp_rcp_update (1.0.2)
[20/29] espressif/esp_secure_cert_mgr (2.4.1)
[21/29] espressif/jsmn (1.1.0)
[22/29] espressif/json_generator (1.1.2)
[23/29] espressif/json_parser (1.0.3)
[24/29] espressif/led_indicator (0.9.2)
[25/29] espressif/led_strip (2.5.3)
[26/29] espressif/mdns (1.3.0)
[... 22 lines cut ...]
-- Adding linker script /home/phoscon/light/build/esp-idf/esp_system/ld/memory.ld
-- Adding linker script /home/phoscon/esp/v5.2.1/esp-idf/components/esp_system/ld/esp32/sections.ld.in
-- Adding linker script /home/phoscon/esp/v5.2.1/esp-idf/components/esp_rom/esp32/ld/esp32.rom.ld
-- Adding linker script /home/phoscon/esp/v5.2.1/esp-idf/components/esp_rom/esp32/ld/esp32.rom.api.ld
-- Adding linker script /home/phoscon/esp/v5.2.1/esp-idf/components/esp_rom/esp32/ld/esp32.rom.libgcc.ld
-- Adding linker script /home/phoscon/esp/v5.2.1/esp-idf/components/esp_rom/esp32/ld/esp32.rom.newlib-data.ld
-- Adding linker script /home/phoscon/esp/v5.2.1/esp-idf/components/esp_rom/esp32/ld/esp32.rom.syscalls.ld
-- Adding linker script /home/phoscon/esp/v5.2.1/esp-idf/components/esp_rom/esp32/ld/esp32.rom.newlib-funcs.ld
-- Adding linker script /home/phoscon/esp/v5.2.1/esp-idf/components/soc/esp32/ld/esp32.peripherals.ld
-- BUTTON: 3.2.0
-- LED_INDICATOR: 0.9.2
-- ESP_LCD_GC9A01: 1.2.0
-- ESP_LCD_ILI9341: 1.2.0
-- git rev-parse returned 'fatal: not a git repository (or any of the parent directories): .git'
ESP Insights Project commit: HEAD-HASH-NOTFOUND
-- Components: app_bridge app_reset app_trace app_update bootloader bootloader_support bt chip cmock console cxx driver efuse esp-tls esp_adc esp_app_format esp_bootloader_format esp_coex esp_common esp_eth esp_event esp_gdbstub esp_hid esp_http_client esp_http_server esp_https_ota esp_https_server esp_hw_support esp_lcd esp_local_ctrl esp_matter esp_matter_bridge esp_matter_console esp_matter_controller esp_matter_rainmaker esp_matter_thread_br esp_mm esp_netif esp_netif_stack esp_partition esp_phy esp_pm esp_psram esp_ringbuf esp_rom esp_system esp_timer esp_wifi espcoredump espressif__button espressif__cbor espressif__cmake_utilities espressif__esp-serial-flasher espressif__esp_bsp_generic espressif__esp_diag_data_store espressif__esp_diagnostics espressif__esp_encrypted_img espressif__esp_insights espressif__esp_lcd_gc9a01 espressif__esp_lcd_ili9341 espressif__esp_lcd_touch espressif__esp_lcd_touch_cst816s espressif__esp_lcd_touch_ft5x06 espressif__esp_lcd_touch_gt1151 espressif__esp_lcd_touch_gt911 espressif__esp_lcd_touch_tt21100 espressif__esp_lvgl_port espressif__esp_rcp_update espressif__esp_secure_cert_mgr espressif__jsmn espressif__json_generator espressif__json_parser espressif__led_indicator espressif__led_strip espressif__mdns espressif__rmaker_common esptool_py fatfs freertos hal heap http_parser idf_test ieee802154 json log lvgl__lvgl lwip main mbedtls mqtt newlib nvs_flash nvs_sec_provider openthread partition_table perfmon protobuf-c protocomm pthread sdmmc soc spi_flash spiffs tcp_transport ulp unity usb vfs wear_levelling wifi_provisioning wpa_supplicant xtensa
-- Component paths: /home/phoscon/.espressif/esp-matter/examples/common/app_bridge /home/phoscon/.espressif/esp-matter/examples/common/app_reset /home/phoscon/esp/v5.2.1/esp-idf/components/app_trace /home/phoscon/esp/v5.2.1/esp-idf/components/app_update /home/phoscon/esp/v5.2.1/esp-idf/components/bootloader /home/phoscon/esp/v5.2.1/esp-idf/components/bootloader_support /home/phoscon/esp/v5.2.1/esp-idf/components/bt /home/phoscon/.espressif/esp-matter/connectedhomeip/connectedhomeip/config/esp32/components/chip /home/phoscon/esp/v5.2.1/esp-idf/components/cmock /home/phoscon/esp/v5.2.1/esp-idf/components/console /home/phoscon/esp/v5.2.1/esp-idf/components/cxx /home/phoscon/esp/v5.2.1/esp-idf/components/driver /home/phoscon/esp/v5.2.1/esp-idf/components/efuse /home/phoscon/esp/v5.2.1/esp-idf/components/esp-tls /home/phoscon/esp/v5.2.1/esp-idf/components/esp_adc /home/phoscon/esp/v5.2.1/esp-idf/components/esp_app_format /home/phoscon/esp/v5.2.1/esp-idf/components/esp_bootloader_format /home/phoscon/esp/v5.2.1/esp-idf/components/esp_coex /home/phoscon/esp/v5.2.1/esp-idf/components/esp_common /home/phoscon/esp/v5.2.1/esp-idf/components/esp_eth /home/phoscon/esp/v5.2.1/esp-idf/components/esp_event /home/phoscon/esp/v5.2.1/esp-idf/components/esp_gdbstub /home/phoscon/esp/v5.2.1/esp-idf/components/esp_hid /home/phoscon/esp/v5.2.1/esp-idf/components/esp_http_client /home/phoscon/esp/v5.2.1/esp-idf/components/esp_http_server /home/phoscon/esp/v5.2.1/esp-idf/components/esp_https_ota /home/phoscon/esp/v5.2.1/esp-idf/components/esp_https_server /home/phoscon/esp/v5.2.1/esp-idf/components/esp_hw_support /home/phoscon/esp/v5.2.1/esp-idf/components/esp_lcd /home/phoscon/esp/v5.2.1/esp-idf/components/esp_local_ctrl /home/phoscon/.espressif/esp-matter/components/esp_matter /home/phoscon/.espressif/esp-matter/components/esp_matter_bridge /home/phoscon/.espressif/esp-matter/components/esp_matter_console /home/phoscon/.espressif/esp-matter/components/esp_matter_controller /home/phoscon/.espressif/esp-matter/components/esp_matter_rainmaker /home/phoscon/.espressif/esp-matter/components/esp_matter_thread_br /home/phoscon/esp/v5.2.1/esp-idf/componen [... cut ...] -idf/components/wpa_supplicant /home/phoscon/esp/v5.2.1/esp-idf/components/xtensa
-- Configuring incomplete, errors occurred!
See also "/home/phoscon/light/build/CMakeFiles/CMakeOutput.log".

 *  The terminal process "/usr/bin/bash '-c', 'cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 -B /home/phoscon/light/build -S /home/phoscon/light'" terminated with exit code: 1. 
```
- Host Machine OS: debian 12
- Commissioner app and versions if present: NONE
- Commissioner's logs if present: NONE

shubhamdp, 2024-05-06:
Light example is not broken, It just not handle being built from any other place than `examples/light`. And I think you moved it from there to some other place.

Can you please replace this https://github.com/espressif/esp-matter/blob/b84a2cc62a3d04578d35b470aae4e1bd6bda9d42/examples/light/CMakeLists.txt#L17 with 
```
include(${ESP_MATTER_PATH}/examples/common/cmake_common/components_include.cmake)
```

Thanks for reporting, I'll raise a fix for this.

Haerteleric, 2024-05-06:
your suggested fix seems to work.

i did not manually copy the example, this was done by the esp-idf vscode extension "new project wizard".

shubhamdp, 2024-05-06:
@Haerteleric Thanks for verifying

closed 2024-05-09 by espressif-bot