espressif/esp-matter #917: Changing the endpoint name displayed in Apple Home

supersunil, 2024-05-04:
Hello there,

How can I change the endpoint name of the devices in the node? We tried adding the fixed label to to factory data with the key value pair of `name:presence sensor`.

[image]

We verified the written data via `chip-tool fixedlabel read <node-id> 1` and it returns the written value.

What exactly I need to change to get this take effect on Apple Home?

ronny-antoon, 2024-05-05:
check this . #914

ph2355, 2024-05-09:
Apple does not use fixed label in the Home app at this time. The only thing that worked for me was creating a bridge and exposing each endpoint as a separate node and naming each node. See the issue which @ronny-antoon mentions above.

dhrishi, 2024-05-15:
Closing as the question is answered.

closed 2024-05-15 by dhrishi