espressif/esp-matter #919: Door Lock device type, one device works good, more than one Door Lock, Got "No Response".

ronny-antoon, 2024-05-06:
**Describe the bug**
One Root Node endpoint
One Aggregator endpoint
--------------------------------
Adding 3 endpoints device type Door Lock, with set parent the aggregator.
Pairing with Apple Home APP, When controlling one "door Lock" the others goes No Response.

**Environment**
- ESP-Matter Commit Id: 1.2
- ESP-IDF Commit Id: 5.1.2
- SoC (eg: ESP32 or ESP32-C3): ESP32
- Device Logs (Please attach the log file):
- Apple Home APP. Homepod mini 17.4.1, Iphone 13.

**Any additional details**
If I make only one endpoint device type Door Lock every thing goes as expected. 

I had made one change in the ConnectedHomeIP 
"connectedhomeip\connectedhomeip\src\app\clusters\door-lock-server\door-lock-server-callback.cpp"
```

bool emberAfPluginDoorLockOnDoorLockCommand(chip::EndpointId endpointId, const Optional<chip::ByteSpan> & pinCode,
                                            DlOperationError & err) {

DoorLockServer::Instance().SetLockState(endpointId, DlLockState::kLocked);

return true;
}

bool emberAfPluginDoorLockOnDoorUnlockCommand(chip::EndpointId endpointId, const Optional<chip::ByteSpan> & pinCode,
                                              DlOperationError & err) {

DoorLockServer::Instance().SetLockState(endpointId, DlLockState::kUnlocked);

return true;
}

```

Jerry-ESP, 2024-05-23:
Hi @ronny-antoon, you mean you add three door-lock device type endpoints, right? When control on endpoint, the other two show no response? Can you share your code to create the 5 endpoints, and all the device logs from pairing to no device response. And which example you use for testing? In esp-matter or connectedhoip?

VaishaliAvhale, 2024-06-12:
Hi @ronny-antoon , I have tried same thing at end and it working I have attached logs and the recording of door-lock device type with 3 endpoint.
iPad OS version: 17.5.1
HomePod Version: 17.5 [21L569]

dhrishi, 2024-07-26:
@ronny-antoon Can you please provide an update? Please close the issue if it works

ronny-antoon, 2024-07-26:
> @ronny-antoon Can you please provide an update? Please close the issue if it works

Iam in vacation, 3 days and i will back to see it again.

ronny-antoon, 2024-07-29:
its working dont know why :( iam sorry

closed 2024-07-29 by ronny-antoon