espressif/esp-matter #920: Reading an attribute from multiple, bound servers -- receiving multiple results

jonsmirl, 2024-05-07:
When constructing a thermostat there is one client site for binding temp sensors. Boris says I should bind all of my temp sensors to that single site. So assuming I have two or three temp sensors bound to the this single temp sensor client site, what is the esp-matter API to read an attribute value from all of the bound devices? This is supposed to work like a subscription and return multiple results.

Note that I can mix unicast and group bindings in a bind. So if I have three unicast bound temp sensors at this client site, what is the code to read attrib 0 from cluster 201 and get three results back?  I believe this uses attribute paths and I should even be able to read multiple attributes from multiple sensors in a single call.

jonsmirl, 2024-05-07:
Boris provided me with some more clues. We need another version of esp_matter::client::cluster_update. This new version takes an attribute path instead of a cmd_handle.  This new version uses the binding manager to iterate over the bindings calling ReadClient for each one.  Lastly, a user supplied callback is called as each attribute value arrives.

jonsmirl, 2024-05-07:
I also suspect the code for routing events is missing too. Bound devices can generate events, and all of the events from those bound devices should get routed to a single event handler for the Binding.

wqx6, 2024-05-09:
Thanks for the suggestions. we are planing to change `command_handle` to `request_handle` which will include the interaction actions of reading attribute/event, subscribing attribute/event, writing attribute and invoking command.

jonsmirl, 2024-05-28:
Any idea when this will be ready?

wqx6, 2024-05-29:
We already have an internal MR under reviewing. I can send to you the patch file if you want to try it.

jonsmirl, 2024-05-29:
I can wait a week or so for it. I'll just work on other issues.

jonsmirl, 2024-06-08:
I synced to your changes, filing a different issue next,

closed 2024-06-08 by jonsmirl