espressif/esp-matter #921: Can't build with Ubuntu 24.04 LTS Desktop

PichuChen, 2024-05-07:
**Describe the bug**
A clear and concise description of what the bug is.

**Environment**
- ESP-Matter Commit Id: 89f47b6c8a3c4f9743d5e713536b3a23d60e8af5
- SoC (eg: ESP32 or ESP32-C3): N/A
- Host Machine OS: Ubuntu 24.04LTS 
```
No LSB modules are available.
Distributor ID: Ubuntu
Description:    Ubuntu 24.04 LTS
Release:        24.04
Codename:       noble
```
python: Python 3.12.3

$ ninja --version
1.11.1

Log:

```
Downloading and installing packages into local source directory:

  Setting up CIPD package manager...done (2.6s)
  Setting up Project actions........skipped (0.1s)
  Setting up Python environment.....[-]
ninja: Entering directory `/home/ubuntu/esp-matter/connectedhomeip/connectedhomeip/.environment/gn_out'
[1/589] python3 ../../third_party/pigweed/repo/pw_build/py/pw_build/python_runner.py --gn-root ../../ --current-path ../.. --default-toolchain=//build/toolchain/linux:linux_x64_gcc --current-toolchain=//third_party/pigweed/repo/pw_build/python_toolchain:python --capture-output --module piptools --python-virtualenv-config python/gen/matter_build_venv/venv_metadata.json --python-dep-list-files python/gen/matter_build_venv._compile_requirements_metadata_path_list.txt -- compile --resolver=backtracking --allow-unsafe --output-file python/gen/matter_build_venv/compiled_requirements.txt python/gen/matter_build_venv/generated_requirements.txt ../../scripts/setup/requirements.build.txt
FAILED: python/gen/matter_build_venv/compiled_requirements.txt
python3 ../../third_party/pigweed/repo/pw_build/py/pw_build/python_runner.py --gn-root ../../ --current-path ../.. --default-toolchain=//build/toolchain/linux:linux_x64_gcc --current-toolchain=//third_party/pigweed/repo/pw_build/python_toolchain:python --capture-output --module piptools --python-virtualenv-config python/gen/matter_build_venv/venv_metadata.json --python-dep-list-files python/gen/matter_build_venv._compile_requirements_metadata_path_list.txt -- compile --resolver=backtracking --allow-unsafe --output-file python/gen/matter_build_venv/compiled_requirements.txt python/gen/matter_build_venv/generated_requirements.txt ../../scripts/setup/requirements.build.txt
    error: subprocess-exited-with-error

    × Getting requirements to build wheel did not run successfully.
    │ exit code: 1
    ╰─> [54 lines of output]
        running egg_info
        writing lib/PyYAML.egg-info/PKG-INFO
        writing dependency_links to lib/PyYAML.egg-info/dependency_links.txt
        writing top-level names to lib/PyYAML.egg-info/top_level.txt
        Traceback (most recent call last):
          File "/home/ubuntu/esp-matter/connectedhomeip/connectedhomeip/.environment/gn_out/python-venv/lib/python3.12/site-packages/pip/_vendor/pyproject_hooks/_in_process/_in_process.py", line 353, in <module>
            main()
          File "/home/ubuntu/esp-matter/connectedhomeip/connectedhomeip/.environment/gn_out/python-venv/lib/python3.12/site-packages/pip/_vendor/pyproject_hooks/_in_process/_in_process.py", line 335, in main
            json_out['return_val'] = hook(**hook_input['kwargs'])
                                     ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
          File "/home/ubuntu/esp-matter/connectedhomeip/connectedhomeip/.environment/gn_out/python-venv/lib/python3.12/site-packages/pip/_vendor/pyproject_hooks/_in_process/_in_process.py", line 118, in get_requires_for_build_wheel
            return hook(config_settings)
                   ^^^^^^^^^^^^^^^^^^^^^
          File "/tmp/pip-build-env-bcem_9zo/overlay/lib/python3.12/site-packages/setuptools/build_meta.py", line 325, in get_requires_for_build_wheel
            return self._get_build_requires(config_settings, requirements=['wheel'])
                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
          File "/tmp/pip-build-env-bcem_9zo/overlay/lib/python3.12/site-packages/setuptools/build_meta.py", line 295, in _get_build_requires
            self.run_setup()
          File "/tmp/pip-build-env-bcem_9zo/overlay/lib/python3.12/site-packages/setuptools/build_meta.py", line 311, in run_setup
            exec(code, locals())
          File "<string>", line 288, in <module>
          File "/tmp/pip-build-env-bcem_9zo/overlay/lib/python3.12/site-packages/setuptools/__init__.py", line 104, in setup
            return distutils.core.setup(**attrs)
                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
          File "/tmp/pip-build-env-bcem_9zo/overlay/lib/python3.12/site-packages/setuptools/_distutils/core.py", line 184, in setup
            return run_commands(dist)
                   ^^^^^^^^^^^^^^^^^^
          File "/tmp/pip-build-env-bcem_9zo/overlay/lib/python3.12/site-packages/setuptools/_distutils/core.py", line 200, in run_commands
            dist.run_commands()
          File "/tmp/pip-build-env-bcem_9zo/overlay/lib/python3.12/site-packages/setuptools/_distutils/dist.py", line 969, in run_commands
            self.run_command(cmd)
          File "/tmp/pip-build-env-bcem_9zo/overlay/lib/python3.12/site-packages/setuptools/dist.py", line 967, in run_command
            super().run_command(command)
          File "/tmp/pip-build-env-bcem_9zo/overlay/lib/python3.12/site-packages/setuptools/_distutils/dist.py", line 988, in run_command
[... 129 lines cut ...]

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/ubuntu/esp-matter/connectedhomeip/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/env_setup.py", line 995, in <module>
    sys.exit(main())
             ^^^^^^
  File "/home/ubuntu/esp-matter/connectedhomeip/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/env_setup.py", line 987, in main
    return EnvSetup(**vars(parse())).setup()
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/ubuntu/esp-matter/connectedhomeip/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/env_setup.py", line 577, in setup
    result = step(spin)
             ^^^^^^^^^^
  File "/home/ubuntu/esp-matter/connectedhomeip/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/env_setup.py", line 766, in virtualenv
    if not virtualenv_setup.install(
           ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/ubuntu/esp-matter/connectedhomeip/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/virtualenv_setup/install.py", line 443, in install
    install_packages(gn_target)
  File "/home/ubuntu/esp-matter/connectedhomeip/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/virtualenv_setup/install.py", line 424, in install_packages
    raise subprocess.CalledProcessError(
subprocess.CalledProcessError: Command '['ninja', '-C', '/home/ubuntu/esp-matter/connectedhomeip/connectedhomeip/.environment/gn_out', '-v', ':python_packages.install']' returned non-zero exit status 1.
```

**Any additional details**
...

jonsmirl, 2024-05-07:
I am running 22.04LTS too and the first time you ran that was something newer than 22.04. 22.04 is still on python 3.10
~~~
jonsmirl@ares:~$ python --version
Python 3.10.12
jonsmirl@ares:~$ ninja --version
1.10.1
~~~

shubhamdp, 2024-05-08:
@PichuChen right now the setup is only tested on Ubuntu 22.04 LTS and it may take some time to test it on 24.04 LTS. https://docs.espressif.com/projects/esp-matter/en/latest/esp32/developing.html#host-setup

PichuChen, 2024-05-08:
> you ran that was something newer

Yes, I known 24.04 didn't tested right now, but I didn't found the ticket related on it, so I open a ticket for this issue.

dhrishi, 2024-05-28:
Even the upstream connectedhomeip does not mention 24.04 in the tested environment - https://github.com/project-chip/connectedhomeip/blob/master/docs/guides/BUILDING.md#tested-operating-systems

baylf2000, 2024-07-21:
Is there any update on this at all? Has anyone managed to make the required changes to make 24.04 work? Ubuntu 22.04 is several years old now, and 24.04 has many improvements.

KUAPT, 2024-08-10:
Is there any new progress on this issue now? I also encountered the same problem with ubuntu24.04-LTS and encountered the same error message

KUAPT, 2024-08-12:
python --version 3.11.x
ninja --version  1.10.1
After testing, reverting to the above version can pass
I guess the problem may occur in the Python version

baylf2000, 2024-08-13:
@KUAPT I opened an issue regarding this over at the matter project. Testing with Fedora 40 shows no issues, and that uses Python 3.12, so I don't think the issue is with Python. You can view the details here: [https://github.com/project-chip/connectedhomeip/issues/34432](url)

jonsmirl, 2024-08-13:
It appears to be missing a python package. Fedora may have added that package and Ubuntu didn't. Maybe you can identify what is missing and manually add it.

dhrishi, 2024-09-26:
@PichuChen I am closing this one as it is tracked through the upstream issue https://github.com/project-chip/connectedhomeip/issues/34432

closed 2024-09-26 by dhrishi