espressif/esp-matter #922: Unable to use the matter control command to generate a QR code that can be matched correctly

limengdu, 2024-05-08:
Hi guys, I'm currently trying to develop a Matter device using a development board with the ESP32-C6 chip, the host used is Ubuntu 22.04, the IDF version is v5.2.1, and Matter is the latest master branch.
I'm a first time user of Matter, and I'm confused about the things that are listed for in the provided Matter dev docs, resulting in my current inability to generate a QR code that can connect to a device using the matter app. There are two main issues I am having.

Q1: The development documentation mentions "The console on the device can be used to run commands for testing. It is configurable through menuconfig and enabled by default in the Here are some useful commands". 
My confusion is, which of the menuconfig settings mentioned in this sentence is set in order to enable the console for Matter? I am now getting an unresponsive state with any message sent to the device using minicom. Or is there an environment in which the commands for matter need to be available?

Q2: I'm trying to generate a pairable QR code based on the [documentation](https://docs.espressif.com/projects/esp-matter/en/latest/esp32/faq.html#a1-10-how-to-generate-matter-onboarding-codes-qr-code-and-manual-pairing-code) here, I'm using the light from the example by the way, but no matter how much I change the values of discriminator, passcode, vendor-id and product-id I can't find the device with my iPhone. So I would like to know where to look up these values or how to set them?
```
chip-tool payload generate-qrcode --discriminator 3131 --setup-pin-code 20201111 \
                                  --vendor-id 0xFFF1 --product-id 0x8004 \
                                  --version 0 --commissioning-mode 0 --rendezvous 2
```
I can use this command `chip-tool pairing onnetwork-long 0x12 20202021 3840` to connect to my device, but I really don't know where the values inside this command come from and if they can be changed at will.

Requests for your replies are appreciated.

shubhamdp, 2024-05-08:
@limengdu I'd suggest you to please go through this [commissioning and control](https://docs.espressif.com/projects/esp-matter/en/latest/esp32/developing.html#commissioning-and-control) guide.

Regarding QRCode, please take a look at this guide: https://docs.espressif.com/projects/esp-matter/en/latest/esp32/developing.html#commissioning.

For console, are you using an `UART` or `USB_JTAG` port?
Most of the dev-kits have UART port so we have console enabled over that, and all of the examples have `CONFIG_ENABLE_CHIP_SHELL` option enabled so by default they have the console enabled.

If you are using the `USB_JTAG` port  then you will have to enable the `CONFIG_ESP_CONSOLE_USB_SERIAL_JTAG` option.

If you just want to try out an example, you can give a try and flash the example through Google Chrome: https://espressif.github.io/esp-launchpad/?flashConfigURL=https://espressif.github.io/esp-matter/launchpad.toml.

limengdu, 2024-05-09:
> @limengdu I'd suggest you to please go through this [commissioning and control](https://docs.espressif.com/projects/esp-matter/en/latest/esp32/developing.html#commissioning-and-control) guide. [...]

I have read both of the links you gave in their entirety. But still have the same question, how can I get the **Vendor ID**, **ProductID** of the device? These numbers don't match up to any other numbers in the tutorial.

I'll recheck the question about whether UART or USB_JATG is being used, I suspect that's what's causing the problem as well, either way, thanks for the reply.

limengdu, 2024-05-09:
[image]
I don't know if I'm doing this right, but I'll document what I'm doing.

limengdu, 2024-05-09:
[image]
I can send commands now, but I don't understand how to change the values, are they hidden somewhere in the code? Or am I using the wrong command?

limengdu, 2024-05-09:
[image]

I'm using the `matter onboardingcodes onnetwork` command to see a link to a QR code in the console, and when I go to scan this QR code using my iPhone, the app gets stuck.
[image]

 I restarted the phone and still nothing new happens. But I used the chip-tool command: `chip-tool pairing code ${NODE_ID_TO_ASSIGN} MT:#######` on my computer and it worked perfectly. What I can make sure of is that my phone, computer and device are all on the same network.

shubhamdp, 2024-05-09:
As your device is advertising over ble, you need to set the transport/rendezvous as BLE in the qr code. If you are using the default example then qrcode in this [commissioning](https://docs.espressif.com/projects/esp-matter/en/latest/esp32/developing.html#commissioning) section works.

For generating and using different values for these, please check, [mfg-tool](https://github.com/espressif/esp-matter/blob/9180a3be/tools/mfg_tool/README.md)

limengdu, 2024-05-09:
Sorry, I still can't understand what you said "As your device is advertising over ble, you need to set the transport/rendezvous as BLE in the qr code. if you are using the default example then qrcode in this commissioning section works.", my device is supposed to be a WiFi device, how do I set the transport/rendezvous?

shubhamdp, 2024-05-09:
Which example are you using, are you using example from the SDK without any change? If yes then the QRCode in [this](https://docs.espressif.com/projects/esp-matter/en/latest/esp32/developing.html#commissioning) section works.

Now, did you explicitly connected the device to wifi, (either using some menuconfig options, or cli)?
If yes, then the on-network stuff makes sense and device will be advertising its commissionable state using mdns service.
If no, then the device is advertising as commissionable device using the ble service.

> my device is supposed to be a WiFi device

Yes, your device is wifi, but for commissioning, it uses BLE if not already connected to wifi.
[Here](https://docs.espressif.com/projects/esp-matter/en/latest/esp32/developing.html#device-console), you can find some console commands. But, to connect the dev-kit to wifi you can use `matter esp wifi connect SSID PASSPHRASE`.

> how do I set the transport/rendezvous?

Do you want your device to be discoverable on the BLE or mdns? For BLE, you have to do nothing, flashing examples with default configuration works. For mdns, connect it to wifi and you are good to go. In both cases you can use that default QRCode, in the documentation.

If you want to use a different qrcode, you need to read the mfg_tool docs or also read this [production consideration](https://docs.espressif.com/projects/esp-matter/en/latest/esp32/production.html) guide.

limengdu, 2024-05-09:
Forgive me for not providing more detail, the example I used is: https://github.com/espressif/esp-matter/tree/main/examples/light
I only changed the GPIO for the LEDs and enable USB_JTAG in menuconfig, nothing else.
And WiFi has been configured with an IP address.
So earlier I said that I can debug my device by using this command, which is fine, no problem.
```
chip-tool pairing code 0x1234 MT:Y.K90AFN00KA0648G00
```
However, when I use the ESP Matter APP on my iPhone and click on the plus sign in the top right corner and try to scan the code to bind the device for debugging, the APP gets stuck, like in my picture above, without the next menu.

limengdu, 2024-05-09:
I see that the APP was last updated a year ago and I'm not sure if this is a compatibility issue. Maybe I should use an older version of Matter?

shubhamdp, 2024-05-09:
> Forgive me for not providing more detail

Thats why we have templates when filing an issue, but, I see no one wants to fill in the details... :-(

Ahh, I see, I'm not sure about the Matter support in older apps and it may be flaky, can you please update the iOS app and give a try?

Also, can you please check the internet connection to the phone, we had seen the same problem, that the app being stuck on this screen.

For phone app, can you share more details
- Phone being used: (eg: iPhone 12)
- iOS version:
- [ESP Rainmaker app](https://apps.apple.com/us/app/esp-rainmaker/id1497491540) version being used:

limengdu, 2024-05-09:
I am using ESP Matter APP instead of ESP Rainmaker, do I need to change the software? I'm using an iPhone 12, iOS 17.4.1. internet connection works fine.

limengdu, 2024-05-09:
I see that the current year old app is the latest version with no available updates. I am using a mainland China apple id, do I need a US apple id?

limengdu, 2024-05-09:
I used the Rainmaker APP, but again I can't add a device. the RainMaker software version number is 3.1.4 and the Matter software version number is 1.0.4.

limengdu, 2024-05-09:
I wanted to try it with v1.1 of esp-matter, but after I switched branches, I got stuck inside Setting up Python enviroment for a long time. Do I need to clear the cache of the latest matter I was using before? How do I need to do that?
[image]

limengdu, 2024-05-09:
[image]
All over again, but still stuck in the Python environment.

limengdu, 2024-05-09:
[image]
I don't know what to do anymore ......

shubhamdp, 2024-05-09:
try `rm -r connectedhomeip/connectedhomeip/.environment` and give a try again. I'm not sure about ESP Matter app, you should use the ESP Rainmaker app to add a matter device.

limengdu, 2024-05-09:
It doesn't seem to work, I even deleted the whole matter folder and re-cloned it, but I still get stuck at the python environment step...
Is there any way to uninstall matter so I can reinstall it? I don't want to reinstall the system again.
[image]

shubhamdp, 2024-05-09:
The complete environment is inside `connectedhomeip/connectedhomeip/.environment` so ideally deleting it is sort of matter uninstall.

Are you using release/v1.1? Can you try on the latest main branch?

limengdu, 2024-05-10:
I've seen some videos from other people on YouTube and I saw that one of the fellows was using the Home APP for the iPhone and was using the Homepod as a home hub and then the Home APP scanned to tether the device.

But I haven't seen anyone who has successfully used the ESP RainMaker APP to bind a device and debug it. I understand that ESP RainMaker has a [separate development framework and Github repository](https://github.com/espressif/esp-rainmaker), is this something I am currently doing with Matter that is relevant?

shubhamdp, 2024-05-20:
ESP Rainmaker iOS app has the support to commission the Matter devices. You should be able to add/control the device from the app. Its simple, you just have to scan the qrcode, select the group you'd like to add that device, and If its the test device(non certified matter accessory) then please proceed further with that warning, and it should commission the device.

Can you try a simple thing first before doing anything? Please visit https://espressif.github.io/esp-launchpad/?flashConfigURL=https://espressif.github.io/esp-matter/launchpad.toml from Google Chrome.

Please check the tutorial video for flashing/commissioning esp-matter light example with esp-launchpad [here](https://www.youtube.com/live/cgcImPtbnPQ?t=764&feature=shared)

limengdu, 2024-05-24:
Thank you for your help, the problem has been solved.

closed 2024-05-24 by limengdu