espressif/esp-matter #925: Endpoint ids should be recycled

maenkai, 2024-05-11:
**Is your feature request related to a problem? Please describe.**
Description of what the problem is.
At present, every creation of an endpoint is to add a new id, which means that nvs will consume one more namespace, when the consumption to a certain extent, nvs will appear 0x1105 problem, even if I use nvs_erase can not solve, it seems that nvs has an upper limit mechanism

**Describe the solution you'd like**
Description of what you want to happen.
For example, if you delete endpoint 1, you should recycle endpoint 1 instead of endpoint 2 next time

**Describe alternatives you've considered**
Description of any alternative solutions or features you've considered.
Recycle the deleted endpoint id

**Additional context**
- At present, this logic has gone through more than 240 loops and started to fail
```c
  while (true) {
    esp_err_t    err = ESP_OK;
    nvs_handle_t handle;
    char         namespace_name[16] = { 0 };
    snprintf(namespace_name, 16, "endpoint_%X", counts);
    err = nvs_open_from_partition("nvs", namespace_name, NVS_READWRITE, &handle);
    if (err != ESP_OK) {
      ESP_LOGE(TAG, "Error opening partition %s namespace %s. Err: %d", CONFIG_ESP_MATTER_BRIDGE_INFO_PART_NAME,
               namespace_name, err);
      break;
    }

    err = nvs_set_u8(handle, "u8", 1);

    if (err != ESP_OK) {
      printf("set fail = %d \r\n", err);
      break;
    }

    nvs_commit(handle);
    err = nvs_erase_all(handle);
    if (err != ESP_OK) {
      printf("erase fail = %d \r\n", err);
      break;
    }
    nvs_close(handle);

    printf("test ======= %d  name %s\r\n", ++counts , namespace_name);
    vTaskDelay(300);
  }
```

- This is fine if you change it to the following logic
```c
  while (true) {
    esp_err_t    err = ESP_OK;
    nvs_handle_t handle;
    char         namespace_name[16] = { 0 };
    snprintf(namespace_name, 16, "endpoint_%X", counts % 100);
    err = nvs_open_from_partition("nvs", namespace_name, NVS_READWRITE, &handle);
    if (err != ESP_OK) {
      ESP_LOGE(TAG, "Error opening partition %s namespace %s. Err: %d", CONFIG_ESP_MATTER_BRIDGE_INFO_PART_NAME,
               namespace_name, err);
      break;
    }

    err = nvs_set_u8(handle, "u8", 1);

    if (err != ESP_OK) {
      printf("set fail = %d \r\n", err);
      break;
    }

    nvs_commit(handle);
    err = nvs_erase_all(handle);
    if (err != ESP_OK) {
      printf("erase fail = %d \r\n", err);
      break;
    }
    nvs_close(handle);

    printf("test ======= %d  name %s\r\n", ++counts , namespace_name);
    vTaskDelay(300);
  }
```

- Currently expect to fix the logic
```c
namespace endpoint {

endpoint_t *create(node_t *node, uint8_t flags, void *priv_data)
{
    /* Find */
    if (!node) {
        ESP_LOGE(TAG, "Node cannot be NULL");
        return NULL;
    }
    _node_t *current_node = (_node_t *)node;

    /* Allocate */
    _endpoint_t *endpoint = (_endpoint_t *)esp_matter_mem_calloc(1, sizeof(_endpoint_t));
    if (!endpoint) {
        ESP_LOGE(TAG, "Couldn't allocate _endpoint_t");
        return NULL;
    }

    /* Set */
    endpoint->endpoint_id = current_node->min_unused_endpoint_id++;  // ********** It should not be increased unchecked
    endpoint->device_type_count = 0;
    endpoint->parent_endpoint_id = chip::kInvalidEndpointId;
    endpoint->flags = flags;
    endpoint->priv_data = priv_data;

    /* Store */
    if (esp_matter_started) {
        node::store_min_unused_endpoint_id();
    }

    /* Add */
    _endpoint_t *previous_endpoint = NULL;
    _endpoint_t *current_endpoint = current_node->endpoint_list;
    while (current_endpoint) {
        previous_endpoint = current_endpoint;
        current_endpoint = current_endpoint->next;
    }
    if (previous_endpoint == NULL) {
        current_node->endpoint_list = endpoint;
    } else {
        previous_endpoint->next = endpoint;
    }

    return (endpoint_t *)endpoint;
}
```

jonsmirl, 2024-05-11:
You can use this tool to decode nvs and see what it is doing.
https://github.com/AFontaine79/Espressif-NVS-Analyzer

wqx6, 2024-05-13:
The spec says 
[image]

So we will use the minimum unused endpoint id as the endpoint id of new created endpoint. And for the nvs issue, I think https://github.com/espressif/esp-matter/commit/98c6e8070b582c9b18058256f490e6b02a4264e1#diff-fd1d9115adb31f9533d3585e893878020375905e1a324269d7f2ab1aa3f245c8 has already changed the namespace name to esp_matter_kvs so all the endpoints will use one namespace now.

dhrishi, 2024-05-15:
@maenkai Does the above comment answer your questions? Please close if it does.

maenkai, 2024-05-23:
> The spec says ![image](https://private-user-images.githubusercontent.com/43193572/329952988-5f0a4b5b-ee7f-48a4-b287-ed5da566d3d4.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTY0NDQzNzQsIm5iZiI6MTcxNjQ0NDA3NCwicGF0aCI6Ii80MzE5MzU3Mi8zMjk5NTI5ODgtNWYwYTRiNWItZWU3Zi00OGE0LWIyODctZWQ1ZGE1NjZkM2Q0LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA1MjMlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNTIzVDA2MDExNFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTViZGU2N2VmOWI3OTdmYTZiYTZiYTFhMmMzNTcyYjkyNzdmYzc1ZGMyNGE3NzJkZmU4NjQzMjA2N2E3ODIxOGQmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.vhVXYKaGh7mjZQOlHnqkdVhncxc_Fa3f4pCWEY3xSkA)
> 
> So we will use the minimum unused endpoint id as the endpoint id of new created endpoint. And for the nvs issue, I think [98c6e80#diff-fd1d9115adb31f9533d3585e893878020375905e1a324269d7f2ab1aa3f245c8](https://github.com/espressif/esp-matter/commit/98c6e8070b582c9b18058256f490e6b02a4264e1#diff-fd1d9115adb31f9533d3585e893878020375905e1a324269d7f2ab1aa3f245c8) has already changed the namespace name to esp_matter_kvs so all the endpoints will use one namespace now.

Yeah, just solve my problem
I want to know when the changes will be merged into the branch release/v1.1

closed 2024-05-23 by maenkai