espressif/esp-matter #926: Link failure with BT

jonsmirl, 2024-05-11:
I just updated by esp-matter and I can't link.  This is because I am allowing BSS in external memory
~~~
/home/jonsmirl/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld: esp-idf/bt/libbt.a(bt.c.obj):(.literal.esp_bt_mem_release+0x0): undefined reference to `_bt_bss_start'
/home/jonsmirl/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld: esp-idf/bt/libbt.a(bt.c.obj):(.literal.esp_bt_mem_release+0x4): undefined reference to `_bt_bss_end'
/home/jonsmirl/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld: esp-idf/bt/libbt.a(bt.c.obj):(.literal.esp_bt_mem_release+0x20): undefined reference to `_bt_data_start'
/home/jonsmirl/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld: esp-idf/bt/libbt.a(bt.c.obj):(.literal.esp_bt_mem_release+0x24): undefined reference to `_bt_data_end'
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
ninja failed with exit code 1, output of the command is in the /home/jonsmirl/l6/build/log/idf_py_stderr_output_1242968 and /home/jonsmirl/l6/build/log/idf_py_stdout_output_1242968
~~~
I need this fix before I could link again
~~~
If you are having problems at link time, can you try removing the following lines from the component/bt/linker.lf file:

    if ESP_ALLOW_BSS_SEG_EXTERNAL_MEMORY = y:
        * (extram_bss)
and try compiling again.
~~~

shubhamdp, 2024-05-13:
This is something that is broken in IDF and we somehow pulled this in when updating the connectedhomeip submodule to fix crashes during ble-deinit sequence.

Can you please try to disable `ESP_ALLOW_BSS_SEG_EXTERNAL_MEMORY` for short run? This will be fixed in next IDF release or I'll check how this can be fixed in connectedhomeip.

tanishqmanuja, 2024-05-15:
I am also having same issue with light example .. i am new to such big build systems, can someone point towards the right file to file. where should i define this ESP_ALLOW_BSS_SEG_EXTERNAL_MEMORY to 0 ?

shubhamdp, 2024-08-12:
This is fixed with esp-idf v5.2.2, can be used with `CONFIG_ ESP_ALLOW_BSS_SEG_EXTERNAL_MEMORY=y`. Please close the issue if resolved.

dhrishi, 2024-09-09:
Closing it now. Please re-open if you see this

closed 2024-09-09 by dhrishi