espressif/esp-matter #927: Simple way to generate TLV encode/decode logic

jonsmirl, 2024-05-11:
I would like to be able to write a small XML file describing my custom clusters and then run a tool to generate the TLV encode/decode logic.  zap can do that, but ... how can I run zap over a small amount of external XML without rebuilding all of CHIP?

jadhavrohit924, 2024-05-15:
@jonsmirl This is how you can add custom cluster.

1. Write data model xml in required format.
2. Add entry of the xml/s to the src/app/zap-templates/zcl/zcl-with-test-extensions.json, src/app/zap-templates/zcl/zcl.json and src/app/zap-templates/zcl/data-model/all.xml.
3. Use script to regenerate zap. ./scripts/tools/zap_regen_all.py

jonsmirl, 2024-05-15:
There should be a simpler way to do this. Wouldn't you like to put the XML for the Rainmaker cluster into the build, and then when you build, the build process generates the Encode/Decode file?

In the short term... can you check an empty XML file into CHIP named custom.xml and edit src/app/zap-templates/zcl/zcl-with-test-extensions.json, src/app/zap-templates/zcl/zcl.json and src/app/zap-templates/zcl/data-model/all.xml to process it? Then all you have to do is replace custom.xml with your own XML.  But that still leaves the manual process of digging the generated code out of the zzz directories.

jadhavrohit924, 2024-05-16:
As of now this is the way to add custom cluster.