espressif/esp-matter #928: esp matter controller

vishnuprasadvp12345, 2024-05-15:
Using the ESP Matter controller example code, the command for controlling a Matter device isn't functioning as expected. Despite sending the command matter esp controller invoke-cmd 2 1 6 0 "{\"0:BOOL\":true/flase}", there's no change in the attribute values. Upon inspection, the attribute value remains unchanged. Additionally, attempting to modify other attributes within Cluster 6, specifically attribute ID 0x4000 Series, isn't supported by the controller device side response in both false and true same device side 
esp_matter_attribute: ********** R : Endpoint 0x0001's Cluster 0x00000006's Attribute 0x00000000 is 0 **********

sayondeep, 2024-05-15:
@vishnuprasadvp12345 on,off and toggle command for OnOff cluster does not require data.
you can simply use 
`matter esp controller invoke-cmd <node-id> 0x1 0x06 0x02` for toggle
`matter esp controller invoke-cmd <node-id> 0x1 0x06 0x01` for on
`matter esp controller invoke-cmd <node-id> 0x1 0x06 0x00` for off

vishnuprasadvp12345, 2024-05-15:
thank you for your response

closed 2024-05-15 by vishnuprasadvp12345