espressif/esp-matter #929: Multiple On Off Plugin Unit Endpoints

tanishqmanuja, 2024-05-16:
**Describe the question/query that you have**
I am trying to replicate a very common use case based device, i.e. an esp32 connected to a 4 channel relay module. I cannot find any example with multiple server clusters/endpoints for on_off_plugin_unit.
I tried to search the docs, issues on this repo and other github repos for the same. At the end i was able to create the device which shows N number of plugin units on alexa but the issue is after following the generic_switch example the GPIO Pins are messed after initialization, which will be clear in the logs.

On program start
```
[0;32mI (2411) esp_matter_attribute: ********** W : Endpoint 0x0001's Cluster 0x00000003's Attribute 0x00000001 is 0 **********[0m
[0;31mE (2421) app_driver: toggling GPIO 13[0m
[0;31mE (2431) app_driver: SET Endpoint 1 to 0[0m
[0;32mI (2431) esp_matter_attribute: ********** W : Endpoint 0x0001's Cluster 0x00000004's Attribute 0x00000000 is 128 **********[0m
[0;31mE (2451) app_driver: toggling GPIO 13[0m
[0;31mE (2451) app_driver: SET Endpoint 1 to 128[0m
[0;32mI (2451) esp_matter_attribute: ********** W : Endpoint 0x0001's Cluster 0x00000004's Attribute 0x0000FFFC is <invalid type: 0> **********[0m
[0;31mE (2471) app_driver: toggling GPIO 13[0m
[0;31mE (2471) app_driver: SET Endpoint 1 to 0[0m
``` 

After few seconds
```
[0;32mI (25511) esp_matter_attribute: ********** W : Endpoint 0x0001's Cluster 0x00000006's Attribute 0x00004000 is 1 **********[0m
[0;31mE (25521) gpio: gpio_set_level(238): GPIO output gpio_num error[0m
[0;31mE (25521) app_driver: toggling GPIO 1073462352[0m
[0;31mE (25541) app_driver: SET Endpoint 1 to 1[0m
```

**Additional context**
HERE'S MY ATTEMPT: https://github.com/tanishqmanuja/matter.four-plug-device

I took reference from these sources as well:
- https://github.com/oleksiikutuzov/esp32-matter-plugin-unit
- https://github.com/espressif/esp-matter/tree/main/examples/light
- https://github.com/espressif/esp-matter/tree/main/examples/generic_switch
- https://github.com/mrezzaaa/light_relay

Log:
[log.txt](https://github.com/espressif/esp-matter/files/15338877/log.txt)

shubhamdp, 2024-05-20:
Does this comment wasn't helpful https://github.com/espressif/esp-matter/blob/3ddea0493b3cb779414442787bec925d244de086/examples/generic_switch/main/app_main.cpp#L220-L225

tanishqmanuja, 2024-05-20:
That works fine for creating the plugs but i am unable to access plug_list.plug->GPIO_PIN_VALUE inside the attribute change callback .. always return junk value instead of actual gpio pin number.

I guess that's why mrezzaaa's light relay repo also duplicates the gpio pin inside both app_main and app_driver.

shubhamdp, 2024-05-20:
If you check [create_button()](https://github.com/espressif/esp-matter/blob/main/examples/generic_switch/main/app_main.cpp#L128), it passes the `button` as a pointer, and that [pointer](https://github.com/espressif/esp-matter/blob/main/examples/generic_switch/main/app_main.cpp#L221) is local to `app_main()` and hence you may be getting garbage.

For now, can you dynamically allocate it? We'll fix this internally and let you know here.

tanishqmanuja, 2024-05-20:
Thanks, If you can push an example similar to generic switch but for an output device instead of input that would be really helpful.

PS:
Also i tried zero code and with some manual json modification you can have 4 relay device with it. But since it is closed source I won't be able to see the source code generated by it, but they may already have the solution for dynamic endpoints

tanishqmanuja, 2024-08-07:
Any update on this?

pimpalemahesh, 2024-08-26:
@tanishqmanuja The example is ready and will be available soon.

tanishqmanuja, 2024-08-26:
@pimpalemahesh, cannot find. Is it a new directory in examples or have you modified an existing one ?

pimpalemahesh, 2024-09-03:
@tanishqmanuja The example hasn’t been merged with the main branch yet. It will be available in the examples directory as a separate example soon.

dhrishi, 2024-09-09:
https://github.com/espressif/esp-matter/tree/main/examples/multiple_on_off_plugin_units

closed 2024-09-09 by dhrishi