espressif/esp-matter #930: GN, ninja build system to cmake, make build system for linux platform.

gaganpydi, 2024-05-17:
**Describe the question/query that you have**
I have a project that uses cmake and make for building. I want to integrate Matter project into mine. I am using Linux platform, but in matter they are using gn and ninja build system. Do you have source code for Linux platform that uses cmake and make for building. or is there any other way to integrate?
**Additional context**
...

sayondeep, 2024-05-17:
@gaganpydi you can use cmake and make to build any example.
Steps:
1) go to example directory `path/to/esp-matter/examples/light/`
2) `mkdir build`
3) `cd build`
4) `cmake ..`
5) `make`

P.S. make sure to export esp-idf and esp-matter before you build

gaganpydi, 2024-05-20:
I am not using esp32, i want to run Linux examples, for eg. in /connectedhomeip/examples/lighting-app/linux, in this directory there is only build.gn, which gn build system uses to generate ninja files. But i want to build cmake, so i need CMakeLists.txt in this directory, is there any way i can build my example using cmake build system.

dhrishi, 2024-05-20:
@gaganpydi Please raise an issue [here](https://github.com/project-chip/connectedhomeip/issues) for this request instead.

closed 2024-05-20 by dhrishi