espressif/esp-matter #931: WIFI change

stefano055415, 2024-05-21:
it is possible to change the wifi network of an already paired device, for example switching from WIFI_X to WIFI_Y ?

jadhavrohit924, 2024-05-22:
If you change the wifi on the device, then device will be unreadable for the controller.

jadhavrohit924, 2024-05-22:
If you change the wifi for controller too, then yes device is controllable.

stefano055415, 2024-05-22:
Hi @jadhavrohit924  obviously I would change the wifi on both the device and the controller.

I'll try to explain myself better, via a REST-API command, so no matter I would like to send the new ssid and password to the device.

Is there an esp-matter function that allows me to set up this new network?

jadhavrohit924, 2024-05-23:
@stefano055415 You can use [AddOrUpdateNetwork](https://github.com/project-chip/connectedhomeip/blob/master/src/platform/ESP32/NetworkCommissioningDriver.h#L112) API.

jadhavrohit924, 2024-05-23:
[Here](https://github.com/espressif/esp-matter/blob/main/components/esp_matter_console/esp_matter_console_wifi.cpp#L29-#L42) are the ESP APIs that you can use to update wifi network.

closed 2024-05-27 by stefano055415