espressif/esp-matter #932: chip[DL]: Chip stack locking error at 'src/transport/Session.h:178'. Code is unsafe/racy<0x1b>[0m <0x1b>[0;31mE (47179) chip[-]: chipDie chipDie chipDie<0x1b>[0m   abort() was called at PC 0x4204eb0f on core 0

vishnuprasadvp12345, 2024-05-22:
*
any one help me to solve this issue

  uint64_t destination_id = uint64_t(2) ;
    uint16_t endpoint_id = uint16_t(1);
    uint32_t cluster_id = uint32_t(6);
    uint32_t command_id = uint32_t(0);
    const char *command_data_field = "{\"0:BOOL\":false}";
   
     esp_matter::controller::send_invoke_cluster_command(destination_id, endpoint_id, cluster_id, command_id, command_data_field);

shubhamdp, 2024-05-22:
@vishnuprasadvp12345 Please take a look at https://docs.espressif.com/projects/esp-matter/en/latest/esp32/faq.html#a1-11-chip-stack-locking-error-code-is-unsafe-racy

vishnuprasadvp12345, 2024-05-22:
I given data from UART without encountering errors, but when the same function is invoked via MQTT, errors occur. This issue arises specifically when #ifdef CONFIG_ESP_MATTER_COMMISSIONER_ENABLE is enabled. How can I send data through MQTT using the function esp_matter::controller::send_invoke_cluster_command(destination_id, endpoint_id, cluster_id, command_id, command_data_field)?

shubhamdp, 2024-05-22:
@vishnuprasadvp12345 The FAQ also answers how to solve that, it listed two ways to deal with the assert: 1. either use the PostEvent or 2. Use locks.

Do you find any problem using any of the methods?

vishnuprasadvp12345, 2024-05-23:
thank you for your reply its working fine

closed 2024-05-23 by vishnuprasadvp12345