espressif/esp-matter #933: How to enable flash encryption in esp-matter?

PhLuReh, 2024-05-22:
I read the documentation about [Flash Encryption](https://docs.espressif.com/projects/esp-idf/en/v5.1.4/esp32/security/flash-encryption.html) and [Matter platform security](https://docs.espressif.com/projects/esp-matter/en/latest/esp32h2/security.html#secure-storage-guide). It is stated, that partitions marked with `encrypted` will be self-encrypted during initial encryption phase on first boot.

But there's also some limitation when using `NVS` partition. Matter credentials as generated by `mfg_tool.py` seem to be NVS partitions, as normal (not pre-encrypted) usage fails certainly as reported here: https://github.com/espressif/esp-matter/issues/726

As stated in the documentation and also in https://github.com/espressif/esp-matter/issues/716#issuecomment-1794228624, it is stated to use Secure Boot and Flash Encryption.

What is the proposed way to enable Flash Encryption in production, without having the need to use Host Generated Keys and a pre-encrypted key material partition? Can you provide a step-by-step guide?

shubhamdp, 2024-05-24:
@PhLuReh We have a guide for [flash and nvs encryption](https://github.com/project-chip/connectedhomeip/blob/master/docs/guides/esp32/flash_nvs_encryption.md). Also, please take a look at [secure cert partition usage guide](https://github.com/project-chip/connectedhomeip/blob/master/docs/guides/esp32/secure_cert_partition.md).

What is the use case here? Are you talking about in-field upgrades or these devices are brand new?

For new devices:
For Matter, it needs to have some manufacturing information in the fctry partition (it is of type nvs). For securing, it needs flash and nvs encryption enabled. This information has to be pre-flashed.

NVS partition is generated and encrypted on the host, and then flashed on the device (two things are flashed, partition and key-partition).

Other steps that needs to be carried our are explained in [host based workflow](https://docs.espressif.com/projects/esp-idf/en/stable/esp32/security/host-based-security-workflows.html)

For no-host case, we do not have an example, but it will be as follows, in firmware:
1. A secure out of band way to get the device attestation info, parameters for onboarding data, any other manufacturing data to the device.
2. You will also need some way to display the onboarding payload (qrcode/manualcode) to the user
3. If [nvs encryption](https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-reference/storage/nvs_encryption.html) is enabled, if nvs keys are not present then they will be generated on the first boot. And then this key can be used for further encrypting/decrypting.
4. When using nvs_get/set APIs to write the data, it will detect if encryption is enabled or not and then write/read accordingly.

shubhamdp, 2024-05-24:
If you are using mfg_tool.py script then, [`--encrypt`](https://github.com/espressif/esp-matter/blob/main/tools/mfg_tool/mfg_tool.py#L533) option will generate both items.

Yes, a different key partition on different will be more secure.

> Is there any further explanation available to the steps necessary according to [Enable flash and NVS encryption](https://github.com/project-chip/connectedhomeip/blob/master/docs/guides/esp32/flash_nvs_encryption.md#enable-flash-and-nvs-encryption-some-factory-settings-using-idfpy-menuconfig)? I really don't understand the technical aspect behind this...

You can take a look at [host based security workflow](https://docs.espressif.com/projects/esp-idf/en/stable/esp32/security/host-based-security-workflows.html) for more details.

dhrishi, 2024-05-29:
@PhLuReh Please close the issue if your questions are answered

closed 2024-05-31 by dhrishi

PhLuReh, 2024-06-05:
I needed to work on this first. It is still quite unclear, because a lot of additional steps are necessary. And I think one hits me hard. Production process currently uses JTAG with OpenOCD. Am I wrong, or is writing fuses currently not really supported with OpenOCD? And there is no way to let the device burn it's fuses on its own, when having host based security?

AdityaHPatwardhan, 2024-06-06:
Hi @PhLuReh,
I dont think writing eFuses is supported in openOCD, this functionality is only exposed by the `espefuse.py` utility
For your case the problem with NVS encryption is as follows
1) When you want to enable NVS encryption through the device firmware. The NVS partition must be empty at the time of enabling
2) In your case the NVS partition contains some data so it is not empty. Due to this you need to encrypt the NVS partition outside and flash the encrypted NVS partition and the nvs keys.

as @shubhamdp has already conveyed. For this you need to generate unique nvs keys partition for each device, then encrypt the NVS partition using that and flash both of them on the device. Using same nvs keys partition is not recommended.

Apart from this all the other stuff (writing security efuses, enabling flash encryption, enabling secure boot etc.) can be enabled on the device through bootloader. For this you just need to enable respective config options for your project.

PhLuReh, 2024-06-06:
Thank you for explaining, @AdityaHPatwardhan. So using openOCD should work with pre-encrypted NVS partitions? I will test it.

PhLuReh, 2024-06-11:
I'm still struggling and hoping for further advice. Why is it such a complex process???

What I'm trying to establish: I will receive CD (from CSA) and DAC key + cert + PAI (from cert provider) in the near future, which I would be able to use in a Matter firmware, having flash encryption turned on.

What I think I have to do is:
- turn on flash encryption via `menuconfig`
- create encryption key and flash it
- generate the sec_cert partition and encrypt
- generate factory_data partition and encrypt

Up to now, each tool has a different problem, I'm struggling with:
- `configure_esp_secure_cert.py` creates `esp_secure_cert` partition, but also flashes. I need to pre-generate the partitions.
- `mfg_tool.py` generates encryption key, but also generates DACs (wich will be provided).
- `generate_esp32_chip_factory_bin` with `-e` throws error: `Error. --keygen and --inputkey both are not allowed.` which were not specified.

Do I have to choose discriminator and passcode on my own or is there a tool, that does it like `mfg_tool`?

I'm quite lost, as there is no step-by-step guide on how to progress. Please provide any, if there is.

PhLuReh, 2024-06-12:
how can I create the `esp_secure_cert` peratition without relying on `configure_esp_secure_cert.py` in order to prevent the required port argument? I do not want to configure the DS nor want to flash the partition directly.

PhLuReh, 2024-06-12:
I now wrote my own python script that does:
- pack `factory_partition.bin` using the `generate_esp32_chip_factory_bin.py` tool
- generate `esp_secure_cert` partition using `nvs_format` from the `esp_secure_cert` package
- generate a private key for nvs encryption and store it to `nvs_keys.bin` using `espsecure.py`

Up to now it is untested, but I would appreciate, if it get's confirmed by Espressif...

AdityaHPatwardhan, 2024-06-12:
Hi @PhLuReh,
Regarding your issue 

> configure_esp_secure_cert.py creates esp_secure_cert partition, but also flashes. I need to pre-generate the partitions without flashing (and without the need to specify a port) for production mode.

Please use the `--skip_flash` option available with the script. That should just generate the binary locally without flashing.
The port is a mandatory requirement but it will be a no-op when you use the --skip_flash command.

> generate esp_secure_cert partition using nvs_format from the esp_secure_cert package

I would avoid using nvs_format and only recommend tlv format. this partition shall be consumed by the esp_secure_cert_mgr. In that case the NVS support is marked as legacy. Please use the default configuration. Let me know if you have any trouble with using the TLV format.

Regarding mfg_tool.py and generate_esp32_chip_factory_bin issue
@shubhamdp Can you please take a look

Otherwise the content that has been generated by your script seem fine as long as you are generating unique NVS encryption key for each device.

PhLuReh, 2024-06-13:
> Please use the `--skip_flash` option available with the script. That should just generate the binary locally without flashing.
> The port is a mandatory requirement but it will be a no-op when you use the --skip_flash command.

As `--port` still is required, even if using `--skip_flash` someone may use `--port /dev/null`. Just in case.

But, I now recognized, that my current `sdkconfig`/`partitions.csv` do not use the `èsp_secure_cert`. All data will be bundled in the `factory_data.bin` and read from there. I think it is because all the [config switches](https://docs.espressif.com/projects/esp-matter/en/latest/esp32/developing.html#configuration-options) are set to `FACTORY`, which is perfectly fine for now.

PhLuReh, 2024-06-14:
Are these assumptions correct for the TLV format:
- encryption is done by the flash encryption? no further method needed?
- `config_esp_secure_cert.py` does not store device information (like `generate_esp32_chip_factory_bin` does)? why not?
- even if I would chose to use TLV, I still would need NVS for device information?

AdityaHPatwardhan, 2024-06-18:
@PhLuReh, Sorry for the delayed reply.

> encryption is done by the flash encryption? no further method needed?

For certificates only flash encryption is applied as they are anyways public information and dont need to be protected at the cost of additional computation.
However, for private keys we have a couple of security schemes that have their own implementation for the protection of the private key.
You can find more details about the secure storage algorithms supported in esp_secure_cert_mgr [here](https://github.com/espressif/esp_secure_cert_mgr/blob/main/docs/format.md#tlv-storage-algorithms)

> config_esp_secure_cert.py does not store device information (like generate_esp32_chip_factory_bin does)? why not?

So `esp_secure_cert` partition is designed to only store critical and read only data. 
configure_esp_secure_cert.py script only stores the minimum required data. It is a development script that generates the esp_secure_cert partition in a way that can be parsed by the firmware. The support has not been updated further to add additional functionality to store things such as device type. Some progress has been made in that context where we used a csv format to tell what data to store but it is still long from completion.

> even if I would chose to use TLV, I still would need NVS for device information?

So TLV format is only specific to esp_secure_cert partition where the certificates and keys are stored. Yes you would still need other NVS partitions for generic usage by other components present in esp-idf.