espressif/esp-matter #935: GPIO and light sleep

Kristian8606, 2024-05-25:
I use the example of a lock I am trying to use a GPIO to trigger a relay when I unlock the lock but the GPIO is not responding.
How do I disable light sleep while using the relay and then enable light sleep again.

shubhamdp, 2024-05-27:
You can find more details about the sleep mode here: https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-reference/system/sleep_modes.html#sleep-modes

Also, could you elaborate more on your problem, and help with few details
**Environment**
- ESP-Matter Commit Id:
- ESP-IDF Commit Id:
- SoC (eg: ESP32 or ESP32-C3):
- Device Logs (Please attach the log file):
- Host Machine OS:

Kristian8606, 2024-05-28:
ESP-IDF v5.2.1
SoC  ESP32-H2
Host Machine OS: Mac OS 14.5
ESP-Matter Commit Id: -
Thanks for responding. It turned out that GPIO actually works, but I can't implement GPIO correctly.
I get choppy output, I use an LED, but when I set gpio_set_level(gpio, 1);
lights up and starts flashing rapidly.

Kristian8606, 2024-05-28:
I was able to get a steady state of the GPIO using the function gpio_hold_en();

closed 2024-06-06 by Kristian8606