espressif/esp-matter #936: Matter Light Commissioning Fail

RoootNoodle, 2024-05-26:
I am trying to create a matter light using the Matter Light example. It uploads and runs fine until commissioning. I use the Rainmaker app for commissioning but keeps saying connecting and eventually gives the message Unable to add accessory.  When monitoring the terminal of the ESP32, the following message is received:

I (35374) chip[NP]: ESP NetworkCommissioningDelegate: SSID: Oak View 3_5G
W (35374) wifi:Haven't to connect to a suitable AP now!
I (35384) chip[DL]: WiFi station mode change: Enabled -> Disabled
I (35454) chip[DL]: WiFi station mode change: Disabled -> Enabled
W (35454) wifi:Haven't to connect to a suitable AP now!
I (35454) chip[DL]: Attempting to connect WiFi station interface
I (35464) chip[DL]: WiFi station state change: NotConnected -> Connecting
I (35474) chip[DL]: Done driving station state, nothing else to do...
W (35484) wifi:Haven't to connect to a suitable AP now!
I (35484) chip[DL]: Attempting to connect WiFi station interface
E (35494) wifi:sta is connecting, return error
E (35494) chip[DL]: esp_wifi_connect() failed: ESP_ERR_WIFI_CONN
W (35504) wifi:Haven't to connect to a suitable AP now!
I (35504) chip[DL]: Attempting to connect WiFi station interface
E (35514) wifi:sta is connecting, return error
E (35514) chip[DL]: esp_wifi_connect() failed: ESP_ERR_WIFI_CONN
W (35524) wifi:Haven't to connect to a suitable AP now!
E (35524) chip[DL]: Failed to get configured network when updating network status: Error ESP32:0x0500300F
I (37854) NimBLE: GATT procedure initiated: indicate; 
I (37854) NimBLE: att_handle=14

This keeps looping until the message "Unable to add accessory" appears in the Rainmaker app
I have tried it with different boards and different examples and get the same problem. 
As far as I am concerned an Apple hub is not needed when adding it with the Rainmaker app. Am I wrong?

Using:
 ESP32C3 Wroom-02 
ESP-IDF extension in VScode on a WSL

dhrishi, 2024-05-26:
@RoootNoodle
>I (35374) chip[NP]: ESP NetworkCommissioningDelegate: SSID: Oak View 3_5G

Looks like your phone is connected to a 5 GHz AP. The ESP32-C3 does not support 5 GHz and hence the Wi-Fi connection error and the commissioning failure.

Please move your phone to a 2.4 GHz and retry commissioning. You can reset to factory your device using the CLI
`matter device factoryreset`

RoootNoodle, 2024-05-26:
Perfect, it works. 

Thank you very much

closed 2024-05-26 by dhrishi