espressif/esp-matter #938: Blank Matter Project

RoootNoodle, 2024-05-28:
Is there a way to create a blank matter project that includes all the necessary libraries? I would prefer not to work from an example as they include a lot of unwanted code. 
I have tried to add the components using the component manager in the VS code extension as well as through the terminal. But inside the connecthomeip folder, there are a lot of undefined errors.

jonsmirl, 2024-05-29:
Just start from the light example in esp-matter/examples/light/main, it's only 400 lines of simple code. Then delete what you don't like.

dhrishi, 2024-05-29:
@RoootNoodle A reference application (like light) can ideally be helpful for you to build the data model, handle the events from the stack and update the driver code. Please let us know if there's anything that you find incorrect/irrelevant and we will get it reviewed.

closed 2024-05-31 by dhrishi