espressif/esp-matter #939: Debug string getting clipped.

jonsmirl, 2024-05-28:
That should say "Load Type". The attribute appears to be ok, the debug output is just clipped.
~~~
I (14111) esp_matter_attribute: ********** R : Endpoint 0x0001's Cluster 0x00000050's Attribute 0x0000FFFC is 0 **********
I (14122) esp_matter_attribute: ********** R : Endpoint 0x0001's Cluster 0x00000050's Attribute 0x0000FFFD is 2 **********
I (14123) esp_matter_attribute: ********** R : Endpoint 0x0001's Cluster 0x00000050's Attribute 0x00000000 is Load Ty **********
I (14124) esp_matter_attribute: ********** R : Endpoint 0x0001's Cluster 0x00000050's Attribute 0x00000001 is null **********
I (14125) esp_matter_attribute: ********** R : Endpoint 0x0001's Cluster 0x00000050's Attribute 0x00000003 is 1 **********
I (14126) esp_matter_attribute: ********** R : Endpoint 0x0001's Cluster 0x00000050's Attribute 0x00000004 is 1 **********
~~~

Are we going to get the ability to set Mode values and Fixed values from code instead of mfg_tool?

jadhavrohit924, 2024-05-30:
> Are we going to get the ability to set Mode values and Fixed values from code instead of mfg_tool?

Yes. We are working on it. For Mode values app should implementa its own SupportedModesManager and store/access data. We have added providing delegate/application impl while creating cluster for some clusters. Please check https://docs.espressif.com/projects/esp-matter/en/latest/esp32/app_guide.html#delegate-implementation

> That should say "Load Type". The attribute appears to be ok, the debug output is just clipped.

I'll look into the issue.

jadhavrohit924, 2024-06-14:
Fixed with [0b7221e](https://github.com/espressif/esp-matter/commit/0b7221ecf1516da16611f1fcdf8c9719d10ee0cf)

closed 2024-06-14 by jonsmirl