espressif/esp-matter #940: Select Mode

Leczok12, 2024-05-28:
Hi
I'm trying to do a select mode on my device but I can't find anything about how do it. I read docs and I found a poor description how to do it with mfg_tool but I can't undrstand it. For example I want add modes like "Solid", "Fade", "Rainbow" to extended_light endpoint.

jonsmirl, 2024-05-29:
~~~
./mfg_tool.py -cn "Test" -v 0xFFF2 -p 0x8001 --pai \
-k ../../connectedhomeip/connectedhomeip/credentials/test/attestation/Chip-Test-PAI-FFF2-8001-Key.pem \
-c ../../connectedhomeip/connectedhomeip/credentials/test/attestation/Chip-Test-PAI-FFF2-8001-Cert.pem \
-cd ../../connectedhomeip/connectedhomeip/credentials/test/certification-declaration/Chip-Test-CD-FFF2-8001.der \
--vendor-name "Company" --product-name "Light" --serial-num "12345678" \
--hw-ver-str "1.0.0" \
--passcode 20202021 --discriminator 3840 \
--supported-modes 0/Solid/1 1/Fade/1 2/Rainbow/1 
~~~
you need to use esp-tool to write this into the nvs partition.

Add this in your code after making the light endpoint
~~~
    mode_select::config_t mode_select_config;
    mode_select_config.current_mode = 0;
    strncpy(mode_select_config.mode_select_description, "Effect Type", sizeof(mode_select_config.mode_select_description));
    cluster_t *mode_select_cluster = mode_select::create(getEndpoint(), &mode_select_config, CLUSTER_FLAG_SERVER, ESP_MATTER_NONE_FEATURE_ID);

~~~

Leczok12, 2024-05-29:
Hi! 
Thanks for answer, I done all with your instruction, but I don't have any widget to control in google home app. Do you know if it's suported by google or just I done somthing wrong?

jonsmirl, 2024-05-29:
It doesn't work in Google Home.

Leczok12, 2024-05-29:
Ok thanks

closed 2024-05-29 by Leczok12