espressif/esp-matter #942: There are only 4 types of chip tool reading attributes

MaplestoryAlen, 2024-05-29:
I added multiple attributes to the fancontrol device, but in reality, there are only four types ,why can't other attributes be read out?

  source code:

	fan::config_t fan_config;
	endpoint_t *endpoint_1 = fan::create(node, &fan_config, ENDPOINT_FLAG_NONE, NULL);
        esp_matter::cluster_t *fan_cluster = esp_matter::cluster::get(endpoint_1, chip::app::Clusters::FanControl::Id);
	
	cluster::fan_control::feature::multi_speed::config_t multi_speed_config;
	cluster::fan_control::feature::multi_speed::add(fan_cluster, &multi_speed_config);
	
	cluster::fan_control::feature::fan_auto::add(fan_cluster);
	
	cluster::fan_control::feature::rocking::config_t rocking_config;
	cluster::fan_control::feature::rocking::add(fan_cluster, &rocking_config);	
	
	cluster::fan_control::feature::wind::config_t wind_config;
	cluster::fan_control::feature::wind::add(fan_cluster, &wind_config);

	cluster::fan_control::feature::airflow_direction::config_t airflow_direction_config;
	cluster::fan_control::feature::airflow_direction::add(fan_cluster, &airflow_direction_config);

chip-tool command:

     ./chip-tool descriptor read attribute-list 0x12 1

chip-tool log:

[1716973622.356532][4831:4833] CHIP:DMG: ReportDataMessage =
[1716973622.356598][4831:4833] CHIP:DMG: {
[1716973622.356659][4831:4833] CHIP:DMG: 	AttributeReportIBs =
[1716973622.356724][4831:4833] CHIP:DMG: 	[
[1716973622.356785][4831:4833] CHIP:DMG: 		AttributeReportIB =
[1716973622.356853][4831:4833] CHIP:DMG: 		{
[1716973622.356914][4831:4833] CHIP:DMG: 			AttributeDataIB =
[1716973622.356977][4831:4833] CHIP:DMG: 			{
[1716973622.357040][4831:4833] CHIP:DMG: 				DataVersion = 0x92624101,
[1716973622.357100][4831:4833] CHIP:DMG: 				AttributePathIB =
[1716973622.357163][4831:4833] CHIP:DMG: 				{
[1716973622.357228][4831:4833] CHIP:DMG: 					Endpoint = 0x1,
[1716973622.357290][4831:4833] CHIP:DMG: 					Cluster = 0x1d,
[1716973622.357354][4831:4833] CHIP:DMG: 					Attribute = 0x0000_FFFB,
[1716973622.357414][4831:4833] CHIP:DMG: 				}
[1716973622.357490][4831:4833] CHIP:DMG: 					
[1716973622.357551][4831:4833] CHIP:DMG: 				Data = [
[1716973622.357655][4831:4833] CHIP:DMG: 						0, 1, 2, 3, 65528, 65529, 65531, 65532, 65533, 
[1716973622.357752][4831:4833] CHIP:DMG: 				],
[1716973622.357814][4831:4833] CHIP:DMG: 			},
[1716973622.357879][4831:4833] CHIP:DMG: 			
[1716973622.357939][4831:4833] CHIP:DMG: 		},
[1716973622.358004][4831:4833] CHIP:DMG: 		
[1716973622.358064][4831:4833] CHIP:DMG: 	],
[1716973622.358130][4831:4833] CHIP:DMG: 	
[1716973622.358191][4831:4833] CHIP:DMG: 	SuppressResponse = true, 
[1716973622.358255][4831:4833] CHIP:DMG: 	InteractionModelRevision = 11
[1716973622.358319][4831:4833] CHIP:DMG: }
[1716973622.358514][4831:4833] CHIP:TOO: Endpoint: 1 Cluster: 0x0000_001D Attribute 0x0000_FFFB DataVersion: 2455912705
[1716973622.358666][4831:4833] CHIP:TOO:   AttributeList: 9 entries
[1716973622.358753][4831:4833] CHIP:TOO:     [1]: 0
[1716973622.358834][4831:4833] CHIP:TOO:     [2]: 1
[1716973622.358900][4831:4833] CHIP:TOO:     [3]: 2
[1716973622.358965][4831:4833] CHIP:TOO:     [4]: 3
[1716973622.359030][4831:4833] CHIP:TOO:     [5]: 65528
[1716973622.359095][4831:4833] CHIP:TOO:     [6]: 65529
[1716973622.359160][4831:4833] CHIP:TOO:     [7]: 65531
[1716973622.359225][4831:4833] CHIP:TOO:     [8]: 65532
[1716973622.359468][4831:4833] CHIP:TOO:     [9]: 65533

**Environment**
- ESP-Matter Commit Id:
- ESP-IDF Commit Id: v5.1.2
- SoC (eg: ESP32 or ESP32-C3): ESP32S3
- Device Logs (Please attach the log file):
- Host Machine OS:Linux 20.04
- Commissioner app and versions if present:
- Commissioner's logs if present:

wqx6, 2024-05-29:
You should use 
```
./chip-tool fancontrol read attribute-list 0x12 1
```
to read the attributes list in fan control cluster.

MaplestoryAlen, 2024-05-30:
> attribute-list 0x12 1

The problem has been resolved. Thank you！

closed 2024-05-30 by MaplestoryAlen