espressif/esp-matter #943: How to read pairing codes and QR codes in a program

MaplestoryAlen, 2024-05-30:
I saw that there are pairing codes and QR codes printed in the log, but I know where to read them. Please let me know
 
 chip[SVR]: SetupQRCode: [MT:Y.K9042C00KA0648G00]
 ... 
 chip[SVR]: Manual pairing code: [34970112332]
 ..
 log:

I (1533) chip[DL]: CHIPoBLE advertising started
I (1533) chip[DL]: Starting ESP WiFi layer
I (1553) wifi:mode : sta (34:85:18:52:51:c4)
I (1553) wifi:enable tsf
W (1553) wifi:Haven't to connect to a suitable AP now!
I (1553) chip[DL]: Done driving station state, nothing else to do...
W (1563) wifi:Haven't to connect to a suitable AP now!
I (1563) chip[DL]: Done driving station state, nothing else to do...
I (1583) chip[SVR]: SetupQRCode: [MT:Y.K9042C00KA0648G00]
I (1583) chip[SVR]: Copy/paste the below URL in a browser to see the QR Code:
I (1593) chip[SVR]: https://project-chip.github.io/connectedhomeip/qrcode.html?data=MT%3AY.K9042C00KA0648G00
I (1593) chip[SVR]: Manual pairing code: [34970112332]
I (1603) chip[SVR]: Initializing subscription resumption storage...
I (1613) chip[SVR]: Server initializing...
I (1623) chip[TS]: Last Known Good Time: [unknown]
I (1623) chip[TS]: Setting Last Known Good Time to firmware build time 2024-05-28T02:03:22
I (1633) chip[DMG]: AccessControl: initializing
I (1643) chip[DMG]: Examples::AccessControlDelegate::Init
I (1643) chip[DMG]: AccessControl: setting
I (1653) chip[DMG]: DefaultAclStorage: initializing
I (1653) chip[DMG]: DefaultAclStorage: 0 entries loaded

dhrishi, 2024-05-31:
@MaplestoryAlen You can find the details about the onboarding payloads (QR code and manual code) [here](https://github.com/project-chip/connectedhomeip/blob/master/src/app/server/OnboardingCodesUtil.cpp)

Although, note that, due to security considerations listed in Matter Specification `section 5.1.7. Generation of the Passcode`, we do not store/generate the pairing code in the firmware in esp-matter

MaplestoryAlen, 2024-05-31:
Thank you very much. The issue has been resolved

MaplestoryAlen, 2024-06-03:
I am sending a question now. If I use the home app to add devices on two phones, the data below will be read out when the first device is added
log:
I (28907) app_driver: QRCode: MT:Y.K90AFN00KA0648G00
I (28913) app_driver: manualPairingCode: 34970112332
I (28918) app_driver: VendorId: 65521
I (28923) app_driver: ProductID:32768
I (28927) app_driver: Discriminator: 3840
I (28932) app_driver: PinCode: 20202021

The first phone has been added, and the data read by the second phone is the same
log:

I (98541) app_main: Commissioning window opened
I (98545) app_driver: QRCode: MT:Y.K90AFN00KA0648G00
I (98551) app_driver: manualPairingCode: 34970112332
I (98556) app_driver: VendorId: 65521
I (98561) app_driver: ProductID:32768
I (98565) app_driver: Discriminator: 3840
I (98570) app_driver: PinCode: 20202021

Actually, the pairing code has changed at this time, why is the data read still the same?

API interface read：PrintOnboardingCodes(chip::RendezvousInformationFlag::kOnNetwork);

VaishaliAvhale, 2024-06-14:
Can you please tell me which example are you trying to commission and have you made any changes in a code if yes please mention it here.
Home App means Apple Home App?  
Please attach complete logs and the SDK commit and also tell me the other 2 phone applications which you are using for commissiong the device.

dhrishi, 2024-06-14:
>The first phone has been added, and the data read by the second phone is the same

@MaplestoryAlen  Are you trying to add the device to a new fabric using the second phone, using the add pairing option? In that case, a new and random onboarding payload (QR code) will be generated by the second phone. The print you are using on the device console won't correspond to that. The print function just prints the default onboarding payload values.

MaplestoryAlen, 2024-06-15:
> @MaplestoryAlen Are you trying to add the device to a new fabric using the second phone, using the add pairing option? In that case, a new and random onboarding payload (QR code) will be generated [...]

After adding the device, retrieve the pairing code again in HOME to pair the second phone. However, this code has already changed, not 34970112332. However, what I read in the program is still 34970112332. Is it possible to read this changed pairing code?
[image]

MaplestoryAlen, 2024-06-15:
> Can you please tell me which example are you trying to commission and have you made any changes in a code if yes please mention it here.
> Home App means Apple Home App?
> Please attach complete logs [...]

It is an Apple Home App. Reopen pairing mode to generate a pairing code, and select 'light' as an example in the SDK

dhrishi, 2024-06-20:
>However, what I read in the program is still 34970112332. Is it possible to read this changed pairing code?

No. Because that is randomly generated by the phone app. But, why do you need that on the device?

MaplestoryAlen, 2024-06-20:
> No. Because that is randomly generated by the phone app. But, why do you need that on the device?

Does the SDK for Matter have a function similar to randomly generating pairing codes on mobile phones? Or I can read this pairing code, not the default 34970112332, because I want to display the QR code every time I turn on pairing mode, in order to provide users with network configuration

dhrishi, 2024-06-20:
But, the phone app, that has generated the manual pairing code, can also show the QR code? Why is that required to be displayed on the device?

MaplestoryAlen, 2024-06-20:
> But, the phone app, that has generated the manual pairing code, can also show the QR code? Why is that required to be displayed on the device?

Okay, I understand! I should have made a mistake in the process. Thank you very much

closed 2024-06-20 by MaplestoryAlen