espressif/esp-matter #944: Light example not building on Ubuntu 22.04, ESP-Matter 1.2 and ESP-IDF 5.2.1

igorLjubinGorenjeCom, 2024-05-30:
**Describe the bug**
Light example not building on Ubuntu 22.04, ESP-Matter 1.2 and ESP-IDF 5.2.1

**Environment**
- ESP-Matter Commit Id: f21874cbfba575d5325c7cafe04bfdda491ed962
- ESP-IDF Commit Id: a322e6bdad4b6675d4597fb2722eea2851ba88cb
- SoC (eg: ESP32 or ESP32-C3): ESP32s3
- Device Logs (Please attach the log file): 
[logfile.log](https://github.com/espressif/esp-matter/files/15496803/logfile.log)
- Host Machine OS: Ubuntu 22.04
- Commissioner app and versions if present:
- Commissioner's logs if present:

**Any additional details**
...

```
$ cat sdkconfig.defaults.esp32s3
#enable BT
# CONFIG_BT_ENABLED=y
# CONFIG_BT_NIMBLE_ENABLED=y

# disable BT connection reattempt
# CONFIG_BT_NIMBLE_ENABLE_CONN_REATTEMPT=n

#
# Bluetooth
#
CONFIG_BT_ENABLED=y
CONFIG_BT_BLUEDROID_ENABLED=y
# CONFIG_BT_NIMBLE_ENABLED is not set
# CONFIG_BT_CONTROLLER_ONLY is not set
CONFIG_BT_CONTROLLER_ENABLED=y
# CONFIG_BT_CONTROLLER_DISABLED is not set

# enable lwip ipv6 autoconfig
CONFIG_LWIP_IPV6_AUTOCONFIG=y

# Use a custom partition table
CONFIG_PARTITION_TABLE_CUSTOM=y
CONFIG_PARTITION_TABLE_FILENAME="partitions.csv"
CONFIG_PARTITION_TABLE_OFFSET=0xC000

# Enable chip shell
CONFIG_ENABLE_CHIP_SHELL=y

#enable lwIP route hooks
CONFIG_LWIP_HOOK_IP6_ROUTE_DEFAULT=y
CONFIG_LWIP_HOOK_ND6_GET_GW_DEFAULT=y

# Button
CONFIG_BUTTON_PERIOD_TIME_MS=20
CONFIG_BUTTON_LONG_PRESS_TIME_MS=5000

# disable softap by default
CONFIG_ESP_WIFI_SOFTAP_SUPPORT=n

# Disable DS Peripheral
CONFIG_ESP_SECURE_CERT_DS_PERIPHERAL=n

# Use compact attribute storage mode
CONFIG_ESP_MATTER_NVS_USE_COMPACT_ATTR_STORAGE=y

# Enable HKDF in mbedtls
CONFIG_MBEDTLS_HKDF_C=y

# Increase LwIP IPv6 address number to 6 (MAX_FABRIC + 1)
# unique local addresses for fabrics(MAX_FABRIC), a link local address(1)
CONFIG_LWIP_IPV6_NUM_ADDRESSES=6

# ESP32-S3-DevKitC-1 Settings
# Buttons
CONFIG_BSP_BUTTONS_NUM=1
CONFIG_BSP_BUTTON_1_TYPE_GPIO=y
CONFIG_BSP_BUTTON_1_GPIO=0
CONFIG_BSP_BUTTON_1_LEVEL=0
# LEDs
CONFIG_BSP_LEDS_NUM=1
CONFIG_BSP_LED_TYPE_RGB=y
CONFIG_BSP_LED_RGB_GPIO=48
CONFIG_BSP_LED_RGB_BACKEND_RMT=y

```

igorLjubinGorenjeCom, 2024-06-03:
[full.log](https://github.com/user-attachments/files/15529304/full.log)

As seen in the full.log file (attached), containing clean, set-target, and build commands error is:

```
[1544/1801] Performing build step for 'chip_gn'
FAILED: esp-idf/chip/chip_gn-prefix/src/chip_gn-stamp/chip_gn-build esp-idf/chip/lib/libCHIP.a /home/igor/workspace/esp/matter/ui-esp32/build/esp-idf/chip/chip_gn-prefix/src/chip_gn-stamp/chip_gn-build /home/igor/workspace/esp/matter/ui-esp32/build/esp-idf/chip/lib/libCHIP.a 
cd /home/igor/workspace/esp/matter/ui-esp32/build/esp-idf/chip && ninja esp32
[204/474] c++ obj/third_party/connectedhomeip/src/platform/ESP32/bluedroid/ESP32.BLEManagerImpl.cpp.o
FAILED: obj/third_party/connectedhomeip/src/platform/ESP32/bluedroid/ESP32.BLEManagerImpl.cpp.o 
/home/igor/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/xtensa-esp32s3-elf-g++ -MMD -MF obj/third_party/connectedhomeip/src/platform/ESP32/bluedroid/ESP32.BLEManagerImpl.cpp.o.d -Wconversion -Os -g2 -fno-common -ffunction-sections -fdata-sections -fno-exceptions -fno-unwind-tables -fno-asynchronous-unwind-tables -Wall -Werror -Wstack-usage=8192 -Wextra -Wshadow -Wunreachable-code -Wvla -Wformat -Wformat-nonliteral -Wformat-security -Wno-deprecated-declarations -Wno-missing-field-initializers -Wno-unknown-warning-option -Wno-unused-parameter -Wno-unused -Wno-cast-function-type -Wno-psabi -Wno-maybe-uninitialized -fdiagnostics-color -fno-strict-aliasing -fmacro-prefix-map=../../../../../../../esp/esp-matter/connectedhomeip/connectedhomeip/config/esp32/third_party/connectedhomeip/= -std=gnu++17 -fno-rtti -Wnon-virtual-dtor -DESP_PLATFORM -DLV_CONF_INCLUDE_SIMPLE -DLV_KCONFIG_IGNORE -DLV_USE_CHART=0 -DLV_USE_GIF=1 -DLV_USE_KEYBOARD=0 -DLV_USE_PERF_MONITOR=0 -DLV_USE_SNAPSHOT=1 -DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\" -DSOC_MMU_PAGE_SIZE=CONFIG_MMU_PAGE_SIZE -DSOC_XTAL_FREQ_MHZ=CONFIG_XTAL_FREQ -D_GLIBCXX_HAVE_POSIX_SEMAPHORE -D_GLIBCXX_USE_POSIX_SEMAPHORE -D_GNU_SOURCE -D_POSIX_READER_WRITER_LOCKS -I/home/igor/workspace/esp/matter/ui-esp32/build/config -isystem/home/igor/esp/esp-idf/components/freertos/include/freertos -isystem/home/igor/esp/esp-idf/components/freertos/FreeRTOS-Kernel/include/freertos -isystem/home/igor/esp/esp-idf/components/newlib/platform_include -isystem/home/igor/esp/esp-idf/components/freertos/config/include -isystem/home/igor/esp/esp-idf/components/freertos/config/include/freertos -isystem/home/igor/esp/esp-idf/components/freertos/config/xtensa/include -isystem/home/igor/esp/esp-idf/components/freertos/FreeRTOS-Kernel/include -isystem/home/igor/esp/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/include -isystem/home/igor/esp/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/include/freertos -isystem/home/igor/esp/esp-idf/components/freertos/esp_additions/include -isystem/home/igor/esp/esp-idf/components/esp_hw_support/include -isystem/home/igor/esp/esp-idf/components/esp_hw_support/include/soc -isystem/home/igor/esp/esp-idf/components/esp_hw_support/include/soc/esp32s3 -isystem/home/igor/esp/esp-idf/components/esp_hw_support/port/esp32s3/. -isystem/home/igor/esp/esp-idf/components/heap/include -isystem/home/igor/esp/esp-idf/components/log/include -isystem/home/igor/esp/esp-idf/components/soc/include -isystem/home/igor/esp/esp-idf/components/soc/esp32s3 -isystem/home/igor/esp/esp-idf/components/soc/esp32s3/include -isystem/home/igor/esp/esp-idf/components/hal/platform_port/include -isystem/home/igor/esp/esp-idf/components/hal/esp32s3/include -isystem/home/igor/esp/esp-idf/components/hal/include -isystem/home/igor/esp/esp-idf/components/esp_rom/include -isystem/home/igor/esp/esp-idf/components/esp_rom/include/esp32s3 -isystem/home/igor/esp/esp-idf/components/esp_rom/esp32s3 -isystem/home/igor/esp/esp-idf/components/esp_common/include -isystem/home/igor/esp/esp-idf/components/esp_system/include -isystem/home/igor/esp/esp-idf/components/esp_system/port/soc -isystem/home/igor/esp/esp-idf/components/esp_system/port/include/private -isystem/home/igor/esp/esp-idf/components/xtensa/esp32s3/include -isystem/home/igor/esp/esp-idf/components/xtensa/include -isystem/home/igor/esp/esp-idf/components/xtensa/deprecated_include -isystem/home/igor/esp/esp-idf/components/esp_timer/include -isystem/home/igor/esp/esp-idf/components/lwip/include -isystem/home/igor/esp/esp-idf/components/lwip/include/apps -isystem/home/igor/esp/esp-idf/components/lwip/include/apps/sntp -isystem/home/igor/esp/esp-idf/components/lwip/lwip/src/include -isystem/home/igor/esp/esp-idf/components/lwip/port/include -isystem/home/igor/esp/esp-idf/components/lwip/port/freertos/include -isystem/home/igor/esp/esp-idf/components/lwip/port/esp32xx/include -isystem/home/igor/esp/esp-idf/components/lwip/port/esp32xx/include/arch -isystem/home/igor/esp/esp-idf/components/lwip/port/esp32xx/include/sys -I/home/igor/workspace/esp/matter/ui-esp32/managed_components/espressif__esp_encrypted_img/include -I/home/igor/workspace/esp/matter/ui-esp32/managed_components/espressif__esp_insights/include -I/home/igor/workspace/esp/matter/ui-esp32/managed_components/espressif__esp_diagnostics/include -isystem/home/igor/esp/esp-idf/components/esp_event/include -isystem/home/igor/esp/esp-idf/components/bt/include/esp32c3/include -isystem/home/igor/esp/esp-idf/components/bt/common/osi/include -isystem/home/igor/esp/esp-idf/components/bt/common/api/include/api -isystem/home/igor/esp/esp-idf/components/bt/common/btc/profile/esp/blufi/i [... cut ...]
cc1plus: note: unrecognized command-line option '-Wno-unknown-warning-option' may have been intended to silence earlier diagnostics
[217/474] c++ obj/third_party/connectedhomeip/zzz_generated/app-common/app-common/zap-generated/libClusterObjects.cluster-objects.cpp.o
ninja: build stopped: subcommand failed.
ninja: build stopped: subcommand failed.
ninja failed with exit code 1, output of the command is in the /home/igor/workspace/esp/matter/ui-esp32/build/log/idf_py_stderr_output_31119 and /home/igor/workspace/esp/matter/ui-esp32/build/log/idf_py_stdout_output_31119

VaishaliAvhale, 2024-06-03:
Hi @igorLjubinGorenjeCom,I was able to reproduce this issue, and we are currently working on it.

igorLjubinGorenjeCom, 2024-06-05:
Would it be possible to provide us with any time frame on this?

shripad621git, 2024-06-06:
@igorLjubinGorenjeCom , we have been able to resolve the build issue ,but we are encountering few bluedroid specific issues after flashing the application. We will get back to you once the issue is resolved.

shripad621git, 2024-06-12:
@igorLjubinGorenjeCom , can you please apply the patch below to resolve your issue.
Please apply this patch at `esp-matter/connectedhomeip/connectedhomeip path.`
Along with the patch please set the two options using` idf.py menuconfig`:
```
CONFIG_BT_BLE_42_FEATURES_SUPPORTED=y
CONFIG_BT_BLE_50_FEATURES_SUPPORTED=n
```

[bluedroid.txt](https://github.com/user-attachments/files/15802344/bluedroid.txt)

dhrishi, 2024-06-20:
@igorLjubinGorenjeCom Can you please confirm if the patch works fine?

igorLjubinGorenjeCom, 2024-07-08:
Dear @dhrishi the code is building and operational. Thank you!

Could I be building this code in an docker environment (I would like to have hardware in the loop testing)?

igorLjubinGorenjeCom, 2024-11-25:
Similar problem on 1.3
../../../../../../../esp/esp-matter/connectedhomeip/connectedhomeip/config/esp32/third_party/connectedhomeip/src/platform/ESP32/bluedroid/BLEManagerImpl.cpp:150:34: error: 'constexpr const chip::System::Clock::Timeout chip::DeviceLayer::Internal::BLEManagerImpl::kFastAdvertiseTimeout' is not a static data member of 'class chip::DeviceLayer::Internal::BLEManagerImpl'
  150 | constexpr System::Clock::Timeout BLEManagerImpl::kFastAdvertiseTimeout;
      |                                  ^~~~~~~~~~~~~~
../../../../../../../esp/esp-matter/connectedhomeip/connectedhomeip/config/esp32/third_party/connectedhomeip/src/platform/ESP32/bluedroid/BLEManagerImpl.cpp:150:34: error: uninitialized 'const chip::DeviceLayer::Internal::BLEManagerImpl::kFastAdvertiseTimeout' [-fpermissive]
In file included from /home/igor/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/xtensa-esp-elf/include/c++/13.2.0/chrono:41,
                 from ../../../../../../../esp/esp-matter/connectedhomeip/connectedhomeip/config/esp32/third_party/connectedhomeip/src/system/SystemClock.h:43,
                 from ../../../../../../../esp/esp-matter/connectedhomeip/connectedhomeip/config/esp32/third_party/connectedhomeip/src/system/SystemLayer.h:39,
                 from ../../../../../../../esp/esp-matter/connectedhomeip/connectedhomeip/config/esp32/third_party/connectedhomeip/src/lib/core/CHIPCore.h:30,
                 from ../../../../../../../esp/esp-matter/connectedhomeip/connectedhomeip/config/esp32/third_party/connectedhomeip/src/include/platform/CHIPDeviceLayer.h:25,
                 from ../../../../../../../esp/esp-matter/connectedhomeip/connectedhomeip/config/esp32/third_party/connectedhomeip/src/include/platform/internal/CHIPDeviceLayerInternal.h:22,
                 from ../../../../../../../esp/esp-matter/connectedhomeip/connectedhomeip/config/esp32/third_party/connectedhomeip/src/platform/ESP32/bluedroid/BLEManagerImpl.cpp:26:
/home/igor/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/xtensa-esp-elf/include/c++/13.2.0/bits/chrono.h:523:13: note: 'const chip::System::Clock::Timeout' {aka 'const class std::chrono::duration<long unsigned int, std::ratio<1, 1000> >'} has no user-provided default constructor
  523 |       class duration
      |             ^~~~~~~~
/home/igor/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/xtensa-esp-elf/include/c++/13.2.0/bits/chrono.h:575:19: note: constructor is not user-provided because it is explicitly defaulted in the class body
  575 |         constexpr duration() = default;
      |                   ^~~~~~~~
/home/igor/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/xtensa-esp-elf/include/c++/13.2.0/bits/chrono.h:696:13: note: and the implicitly-defined constructor does not initialize 'std::chrono::duration<long unsigned int, std::ratio<1, 1000> >::rep std::chrono::duration<long unsigned int, std::ratio<1, 1000> >::__r'
  696 |         rep __r;
      |             ^~~
../../../../../../../esp/esp-matter/connectedhomeip/connectedhomeip/config/esp32/third_party/connectedhomeip/src/platform/ESP32/bluedroid/BLEManagerImpl.cpp: In member function 'CHIP_ERROR chip::DeviceLayer::Internal::BLEManagerImpl::_SetAdvertisingEnabled(bool)':
../../../../../../../esp/esp-matter/connectedhomeip/connectedhomeip/config/esp32/third_party/connectedhomeip/src/platform/ESP32/bluedroid/BLEManagerImpl.cpp:239:9: error: 'mAdvertiseStartTime' was not declared in this scope
  239 |         mAdvertiseStartTime = System::SystemClock().GetMonotonicTimestamp();
      |         ^~~~~~~~~~~~~~~~~~~
../../../../../../../esp/esp-matter/connectedhomeip/connectedhomeip/config/esp32/third_party/connectedhomeip/src/platform/ESP32/bluedroid/BLEManagerImpl.cpp:239:9: note: maximum limit of 1000 namespaces searched for 'mAdvertiseStartTime'
In file included from ../../../../../../../esp/esp-matter/connectedhomeip/connectedhomeip/config/esp32/third_party/connectedhomeip/src/system/SystemLayer.h:36:
../../../../../../../esp/esp-matter/connectedhomeip/connectedhomeip/config/esp32/third_party/connectedhomeip/src/platform/ESP32/bluedroid/BLEManagerImpl.cpp:240:68: error: 'kFastAdvertiseTimeout' was not declared in this scope
  240 |         ReturnErrorOnFailure(DeviceLayer::SystemLayer().StartTimer(kFastAdvertiseTimeout, HandleFastAdvertisementTimer, this));
      |                                                                    ^~~~~~~~~~~~~~~~~~~~~
../../../../../../../esp/esp-matter/connectedhomeip/connectedhomeip/config/esp32/third_party/connectedhomeip/src/lib/support/CodeUtils.h:176:23: note: in definition of macro 'ReturnErrorOnFailure'
  176 |         auto __err = (expr);                                                                                                       \
      |                       ^~~~
../../../../../../../esp/esp-matter/connectedhomeip/connectedhomeip/config/esp32/third_party/connectedhomeip/src/platform/ESP32/bluedroid/BLEManagerImpl.cpp:240:68: note: maximum limit of 1000 namespaces searched for 'kFastAdvertiseTimeout'
  240 |         ReturnErrorOnFailure(DeviceLayer::SystemLayer().StartTimer(kFastAdvertiseTimeout, HandleFastAdvertisementTimer, this));
      |                                                                    ^~~~~~~~~~~~~~~~~~~~~
[... cut ...]
cc1plus: note: unrecognized command-line option '-Wno-unknown-warning-option' may have been intended to silence earlier diagnostics
cc1plus: all warnings being treated as errors
[342/513] c++ obj/third_party/connectedhomeip/zzz_generated/app-common/app-common/zap-generated/libClusterObjects.cluster-objects.cpp.o
ninja: build stopped: subcommand failed.
ninja: build stopped: subcommand failed.
ninja failed with exit code 1, output of the command is in the /home/igor/workspace/esp/matter/ui-esp32/build/log/idf_py_stderr_output_6402 and /home/igor/workspace/esp/matter/ui-esp32/build/log/idf_py_stdout_output_6402

igorLjubinGorenjeCom, 2024-12-11:
@shripad621git 

We would like to switch to Matter 1.3 but the problem persists when utilizing the Bluedroid. Since this issue is not closed, I believe that suggested changes have not been merged to the main branch.

For your information the suggested patch (see above) does not apply after the commit f4363a016e 
commit f4363a016e3001638abeeb9048ad943c4953f905 (HEAD)
Author: shripad621git <79364691+shripad621git@users.noreply.github.com>
Date:   Thu Jun 27 16:56:27 2024 +0530

    Fix the hardware mac address isssue for thread network (#34081) (#34106)

Can you, please, give us, any prediction on this? Is Bluedroid an option for the future?