espressif/esp-matter #945: ON_OFF Cluster

RoootNoodle, 2024-06-01:
I am trying to create a custom cluster as shown by the guide https://docs.espressif.com/projects/esp-matter/en/latest/esp32/developing.html#defining-your-own-data-model . But the on-off config_t is not recognize, am I missing an include or how can this be fixed?
[image]

jonsmirl, 2024-06-02:
Are you using the namespaces?
~~~
using namespace esp_matter;
using namespace esp_matter::attribute;
using namespace esp_matter::endpoint;
using namespace esp_matter::cluster;
~~~

RoootNoodle, 2024-06-02:
Just needed to add the cluster namespace, thanks.

Why would adding a custem cluster change the icon to a light? Is there a way to set the icon to something else?

jadhavrohit924, 2024-06-03:
>  Is there a way to set the icon to something else?

No, by default every ecosystem identifies every device as light only and at the time of commissioning based on endpoint composition ecosystem identify and create particular device types.

RoootNoodle, 2024-06-03:
The end point is defined as on of light, without adding a custom on of cluster, the icon is different from when the custom cluster is added. Why would this be, since the endpoint did not change?

jadhavrohit924, 2024-06-03:
What was the old icon and what new icon? Please attached some snapshots. What custom cluster you have added?

RoootNoodle, 2024-06-03:
[image]

On the right is a on_off_switch endpoint with no custom clusters configure, on the left is when a on_off cluster was added to the endpoint. How do I prevent it from changing?

The endpoint and node was created as stated in the documents as specified in the first comment.

jadhavrohit924, 2024-06-05:
@RoootNoodle This should not happen. We are looking into the issue.

jonsmirl, 2024-06-05:
What app is that screen shot from?

RoootNoodle, 2025-01-06:
Unfortunately it still shows the same. Once the on off cluster is added the icon changes to a light icon as shown in previous comments.

App version v3.3.2

ravencarcass, 2025-02-24:
I was wondering if you could show you compete program. I tried it also to make that thing but I get some errors on my code. Did you made BTW also other projects with matter? I also use matter for a school project and I'm a little bit stuck on it.

RoootNoodle, 2025-02-25:
@ravencarcass I used the light example and changed it to fit my needs. This is the only project I did with matter. 

The only code changed is the code in the screenshot and then in the app_driver.cpp file you have to ensure the variable names correspond with what was declared in app_main.cpp. 

In my case it is btn_endpoint_id, since app_driver.cpp reference that variable. You'll see in the image below

[image]

[image]

ravencarcass, 2025-02-25:
Thx, I had yesterday evening a Eureka moment. I tought I saw it. The only thing I need to do is putting this I2C code in the driver. 

[image]

[image]

[image]

shubhamdp, 2025-02-26:
@RoootNoodle why are you adding `on_off` cluster in `on_off_switch` device type? It would be added by default already? https://github.com/espressif/esp-matter/blob/main/components/esp_matter/esp_matter_endpoint.cpp#L369.

```
on_off::config_t on_off_config;
cluster_t *of_cluster = on_off:: create(sendpoint, &on_off_config, CLUSTER_FLAG_SERVER, NULL);
```

This code would have no effect, as it is trying to add a cluster which already exists. You should also have a error log stating the same. https://github.com/espressif/esp-matter/blob/main/components/esp_matter/esp_matter_core.cpp#L1443

Can you please re-check this one? Or what actually is the problem?

ravencarcass, 2025-02-26:
Thx, for your feedback but. I don't understand it. For my project do I'need in total 3 endpoints. 2 dimming lights and 1 for mode_select_device. But I don't actually know what I'm doing. Pure because I don't know where I need to place everything. 

I normal program microcontrollers from STMicroelectronics, where every thing is explained in a way easier way. Sorry for this side note , back to the question. 

so if you could help me. It would mean a lot to me.

shubhamdp, 2025-02-26:
@ravencarcass Heres the doc that would help in product development: https://docs.espressif.com/projects/esp-matter/en/latest/esp32/developing.html#developing-your-product

And for starter please refer examples/light, it has basic stuff that any project would need: 1. creating matter device, 2. updating driver when data is received from matter, 3. updating matter when data is received from non-matter (i.e. interrupt in this case).

For mode-select please check this doc: https://docs.espressif.com/projects/esp-matter/en/latest/esp32/developing.html#mode-select

ravencarcass, 2025-02-26:
Fun fact, I read part 2.5 now 10 times even more and I still don't understand.

shubhamdp, 2025-02-27:
@ravencarcass fun fact from my side, the code looks an exact copy of light example and the unindented code is NOT AT ALL READABLE, also screenshots do not help code does so that it can be tried. I'd be more interested in your efforts and I see none.

Also, this issue is to solve the original authors problem, and I think we are bloating it.

May be I'd suggest you to open up the new one, with exact details about your problem, your effort, the code that you tried, what really did not work? Is the build failing or you are getting a crash or have no clue what to do, etc. Please.

RoootNoodle, 2025-02-28:
@shubhamdp The goal was to get the icon of the on_off_switch, but the on_off_switch can be interacted with from the app. It only shows whether a switch is in an on or off state. I would like to turn the switch either on, or off via the app. Therefore, I added the on_off_cluster to give that functionality to the on_off_switch endpoint.

shubhamdp, 2026-04-07:
@RoootNoodle Closing, please re-open if required!