espressif/esp-matter #947: temperaturemeasurement cluster measured value is returned as an unsigned value when it should be signed

rlaybourn, 2024-06-03:
I have setup an temperature sensor with code bellow
`    /* Create a Matter node and add the mandatory Root Node device type on endpoint 0 */
    node::config_t node_config;

    // node handle can be used to add/modify other endpoints.
    node_t *node = node::create(&node_config, app_attribute_update_cb, app_identification_cb);
    ABORT_APP_ON_FAILURE(node != nullptr, ESP_LOGE(TAG, "Failed to create Matter node"));

    dimmable_plugin_unit::config_t switch_config; 
    switch_config.on_off.on_off = DEFAULT_POWER;
    switch_config.level_control.current_level = DEFAULT_BRIGHTNESS;

    endpoint1 = dimmable_plugin_unit::create(node, &switch_config, ENDPOINT_FLAG_NONE,NULL);
    ABORT_APP_ON_FAILURE(endpoint1 != nullptr, ESP_LOGE(TAG, "Failed to create extended color light endpoint"));
    temperature_sensor::config_t temperature_sensor_config; 
    temperature_sensor_config.temperature_measurement.measured_value = (int16_t)2000;
    temperature_sensor_config.temperature_measurement.min_measured_value = (int16_t)0000;
    temperature_sensor_config.temperature_measurement.max_measured_value = (int16_t)4000;
    temperature_sensor::add(endpoint1, &temperature_sensor_config);`

I then set a negative value to the temperature 

`        temperature_int = -1000;
        cluster_t *tempSensor_cluster = cluster::get(endpoint1, TemperatureMeasurement::Id);
        attribute_t *tempSensor_attribute = attribute::get(tempSensor_cluster, TemperatureMeasurement::Attributes::MeasuredValue::Id);
        esp_matter_attr_val_t val =  esp_matter_nullable_int16(temperature_int);
        attribute::set_val(tempSensor_attribute, &val);`

when this value is read via chip tool it returns and unsigned value " 64536"  not the signed value expected. Because of this the value will not be accepted by a thermostat I am using, the read fails with error given as 
" Error: src/lib/core/TLVReader.cpp:201: CHIP Error 0x00000026: Wrong TLV type"

Am I setting this cluster up incorrectly or am I encountering a bug ?

**Environment**
- ESP-Matter Commit Id: a322e6bdad4b6675d4597fb2722eea2851ba88cb
- ESP-IDF Commit Id: f21874cbfba575d5325c7cafe04bfdda491ed962
- SoC: ESP32

jadhavrohit924, 2024-06-03:
@rlaybourn Can you please share log files (do not paste)? I have tested this by setting `measured_value to -1000`, chip-tool was able to read it back successfully.
```
[1717399085249] [49267:759556] [TOO] Endpoint: 1 Cluster: 0x0000_0402 Attribute 0x0000_0000 DataVersion: 3556183510
[1717399085250] [49267:759556] [TOO]   MeasuredValue: -1000
```

rlaybourn, 2024-06-03:
[log.light.20240603085140.txt](https://github.com/user-attachments/files/15530528/log.light.20240603085140.txt)

I have attached a log from the monitor of a read event . While in the log it does correctly see the value as -1000 in chip tool the output seen is

`[1717401126.644592][495912:495915] CHIP:DMG: ReportDataMessage =
[1717401126.644598][495912:495915] CHIP:DMG: {
[1717401126.644602][495912:495915] CHIP:DMG:    AttributeReportIBs =
[1717401126.644611][495912:495915] CHIP:DMG:    [
[1717401126.644619][495912:495915] CHIP:DMG:            AttributeReportIB =
[1717401126.644628][495912:495915] CHIP:DMG:            {
[1717401126.644633][495912:495915] CHIP:DMG:                    AttributeDataIB =
[1717401126.644639][495912:495915] CHIP:DMG:                    {
[1717401126.644644][495912:495915] CHIP:DMG:                            DataVersion = 0xc3102575,
[1717401126.644650][495912:495915] CHIP:DMG:                            AttributePathIB =
[1717401126.644657][495912:495915] CHIP:DMG:                            {
[1717401126.644664][495912:495915] CHIP:DMG:                                    Endpoint = 0x1,
[1717401126.644669][495912:495915] CHIP:DMG:                                    Cluster = 0x402,
[1717401126.644676][495912:495915] CHIP:DMG:                                    Attribute = 0x0000_0000,
[1717401126.644681][495912:495915] CHIP:DMG:                            }
[1717401126.644688][495912:495915] CHIP:DMG:
[1717401126.644695][495912:495915] CHIP:DMG:                            Data = 64536,
[1717401126.644701][495912:495915] CHIP:DMG:                    },
[1717401126.644707][495912:495915] CHIP:DMG:
[1717401126.644713][495912:495915] CHIP:DMG:            },
[1717401126.644721][495912:495915] CHIP:DMG:
[1717401126.644726][495912:495915] CHIP:DMG:    ],
[1717401126.644733][495912:495915] CHIP:DMG:
[1717401126.644739][495912:495915] CHIP:DMG:    SuppressResponse = true,
[1717401126.644743][495912:495915] CHIP:DMG:    InteractionModelRevision = 11
[1717401126.644748][495912:495915] CHIP:DMG: }
[1717401126.644817][495912:495915] CHIP:TOO: Endpoint: 1 Cluster: 0x0000_0402 Attribute 0x0000_0000 DataVersion: 3272615285
[1717401126.644846][495912:495915] CHIP:TOO: Response Failure: Can not decode Data`

The same output is seen from a thermostat based on the linux thermostat example in the connectedhomeip sdk

jadhavrohit924, 2024-06-03:
From logs, attribute is set successfully. So there is no problem with attribute setting. Are you using latest chip-tool? If not please build latest chip-tool.

rlaybourn, 2024-06-03:
I did build this chip tool a while ago , although it does work with everything else including a similar temperature sensor device that I built using nordic's sdk. I will try and build the latest chip-tool.   As I simply modified this from the light example It is entirely possible and probably that I have made mistakes along the way, is there a more appropriate example for a temperature sensor that I could start from ?

rlaybourn, 2024-06-03:
As an update to this I just want back and checked out the esp-matter release/v1.2 and quickly added the temperature-measurement cluster to the light example and ported my previous code into it to update the temperature. This is now working as expected and I can read the values correctly with chip tool and a thermostat. I cannot be absolutely sure the I didn't do anything silly the first time round but the same code appears to work fine when using release/v1.2  so that is an acceptable base to start  from

jonsmirl, 2024-06-03:
~~~
    cluster_t *tempSensor_cluster = cluster::get(endpoint1, TemperatureMeasurement::Id);
    attribute_t *tempSensor_attribute = attribute::get(tempSensor_cluster,TemperatureMeasurement::Attributes::MeasuredValue::Id);
    esp_matter_attr_val_t val =  esp_matter_nullable_int16(temperature_int);
~~~

It is not nullable.
~~~
       val = esp_matter_int16 (event.sensors.temperature);
~~~

rlaybourn, 2024-06-03:
Hi , thanks jon. My initial try was not using the nullable version , that was actually a change I made as a bit of a punt when I dug into the matter code and it seemed to be using it in there. Using the v1.2 branch it works despite that mistake although it is something that I will correct.

rlaybourn, 2024-06-05:
moving to branch v1.2 resolves this issue for me

closed 2024-06-05 by rlaybourn