espressif/esp-matter #949: panic -- bleAdv Timeout : Start slow advertisement

jonsmirl, 2024-06-05:
If I let my device sit a while without commissioning, I hit this. Looks like something to do with trying to print debug from a timer interrupt. My esp-matter is b74daacc4408cceefc5b377f2fc87ea43312490a

~~~
I (31670) chip[DL]: bleAdv Timeout : Start slow advertisement

***ERROR*** A stack overflow in task Tmr Svc has been detected.

Backtrace: 0x40375bfa:0x3fcb8460 0x40382939:0x3fcb8480 0x40382f41:0x3fcb84a0 0x403837ea:0x3fcb8520 0x40382a38:0x3fcb8550 0x40382a2e:0x3fcb8580 0x420dd461:0x3fc9dd24 |<-CORRUPTED
0x40375bfa: panic_abort at /home/jonsmirl/esp/esp-idf/components/esp_system/panic.c:466
0x40382939: esp_system_abort at /home/jonsmirl/esp/esp-idf/components/esp_system/port/esp_system_chip.c:93
0x40382f41: vApplicationStackOverflowHook at /home/jonsmirl/esp/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:553
0x403837ea: vTaskSwitchContext at /home/jonsmirl/esp/esp-idf/components/freertos/FreeRTOS-Kernel/tasks.c:3630 (discriminator 7)
0x40382a38: _frxt_dispatch at /home/jonsmirl/esp/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/portasm.S:451
0x40382a2e: _frxt_int_exit at /home/jonsmirl/esp/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/portasm.S:246
0x420dd461: chip::DeviceLayer::Internal::BLEManagerImpl::BleAdvTimeoutHandler(tmrTimerControl*) at /home/jonsmirl/l6/build/esp-idf/chip/../../../../aosp/esp-matter/connectedhomeip/connectedhomeip/config/esp32/third_party/connectedhomeip/src/platform/ESP32/nimble/BLEManagerImpl.cpp:301 (discriminator 1)
 (inlined by) chip::DeviceLayer::Internal::BLEManagerImpl::BleAdvTimeoutHandler(tmrTimerControl*) at /home/jonsmirl/l6/build/esp-idf/chip/../../../../aosp/esp-matter/connectedhomeip/connectedhomeip/config/esp32/third_party/connectedhomeip/src/platform/ESP32/nimble/BLEManagerImpl.cpp:297 (discriminator 1)

ELF file SHA256: b7f98f5e11b9d56e

Rebooting...
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x40375b7c
0x40375b7c: esp_restart_noos at /home/jonsmirl/esp/esp-idf/components/esp_system/port/soc/esp32s3/system_internal.c:159

~~~
another variation of the same failure
~~~
bort() was called at PC 0x40376cf3 on core 0
0x40376cf3: lock_acquire_generic at /home/jonsmirl/esp/esp-idf/components/newlib/locks.c:130

Backtrace: 0x40375bfa:0x3fcb7ed0 0x40382939:0x3fcb7ef0 0x403877d9:0x3fcb7f10 0x40376cf3:0x3fcb7f90 0x40376e29:0x3fcb7fc0 0x40376efa:0x3fcb7fe0 0x42177737:0x3fcb8010 0x42186895:0x3fcb8040 0x421839e3:0x3fcb8160 0x42178f01:0x3fcb8480 0x421a04a9:0x3fcb84b0 0x4200a5a3:0x3fcb84e0 0x420dcb0e:0x3fcb8510 0x420c3adc:0x3fcb8550 0x420c3b16:0x3fcb8580 0x420dd461:0x3fcb85d0 0x4219f3f1:0x3fcb85f0 0x4219f4cd:0x3fcb8610 0x40382cda:0x3fcb8650
0x40375bfa: panic_abort at /home/jonsmirl/esp/esp-idf/components/esp_system/panic.c:466
0x40382939: esp_system_abort at /home/jonsmirl/esp/esp-idf/components/esp_system/port/esp_system_chip.c:93
0x403877d9: abort at /home/jonsmirl/esp/esp-idf/components/newlib/abort.c:38
0x40376cf3: lock_acquire_generic at /home/jonsmirl/esp/esp-idf/components/newlib/locks.c:130
0x40376e29: _lock_acquire_recursive at /home/jonsmirl/esp/esp-idf/components/newlib/locks.c:158
0x40376efa: __retarget_lock_acquire_recursive at /home/jonsmirl/esp/esp-idf/components/newlib/locks.c:314
0x42177737: _fflush_r at /builds/idf/crosstool-NG/.build/xtensa-esp-elf/src/newlib/newlib/libc/stdio/fflush.c:277 (discriminator 2)
 (inlined by) _fflush_r at /builds/idf/crosstool-NG/.build/src/newlib-git-af24ceb7/newlib/libc/include/stdio.h:417 (discriminator 2)
0x42186895: __sbprintf at /builds/idf/crosstool-NG/.build/xtensa-esp-elf/src/newlib/newlib/libc/stdio/vfprintf.c:477 (discriminator 1)
0x421839e3: _vfprintf_r at /builds/idf/crosstool-NG/.build/xtensa-esp-elf/src/newlib/newlib/libc/stdio/vfprintf.c:861
0x42178f01: vprintf at /builds/idf/crosstool-NG/.build/xtensa-esp-elf/src/newlib/newlib/libc/stdio/vprintf.c:34
0x421a04a9: esp_log_writev at /home/jonsmirl/esp/esp-idf/components/log/log.c:212
0x4200a5a3: __wrap_esp_log_writev at /home/jonsmirl/l6/managed_components/espressif__esp_diagnostics/src/esp_diagnostics_log_hook.c:468
0x420dcb0e: chip::Logging::Platform::LogV(char const*, unsigned char, char const*, __va_list_tag) at /home/jonsmirl/l6/build/esp-idf/chip/../../../../aosp/esp-matter/connectedhomeip/connectedhomeip/config/esp32/third_party/connectedhomeip/src/platform/ESP32/Logging.cpp:53
0x420c3adc: chip::Logging::LogV(unsigned char, unsigned char, char const*, __va_list_tag) at /home/jonsmirl/l6/build/esp-idf/chip/../../../../aosp/esp-matter/connectedhomeip/connectedhomeip/config/esp32/third_party/connectedhomeip/src/lib/support/logging/TextOnlyLogging.cpp:207
0x420c3b16: chip::Logging::Log(unsigned char, unsigned char, char const*, ...) at /home/jonsmirl/l6/build/esp-idf/chip/../../../../aosp/esp-matter/connectedhomeip/connectedhomeip/config/esp32/third_party/connectedhomeip/src/lib/support/logging/TextOnlyLogging.cpp:193
0x420dd461: chip::DeviceLayer::Internal::BLEManagerImpl::BleAdvTimeoutHandler(tmrTimerControl*) at /home/jonsmirl/l6/build/esp-idf/chip/../../../../aosp/esp-matter/connectedhomeip/connectedhomeip/config/esp32/third_party/connectedhomeip/src/platform/ESP32/nimble/BLEManagerImpl.cpp:301 (discriminator 1)
 (inlined by) chip::DeviceLayer::Internal::BLEManagerImpl::BleAdvTimeoutHandler(tmrTimerControl*) at /home/jonsmirl/l6/build/esp-idf/chip/../../../../aosp/esp-matter/connectedhomeip/connectedhomeip/config/esp32/third_party/connectedhomeip/src/platform/ESP32/nimble/BLEManagerImpl.cpp:297 (discriminator 1)
0x4219f3f1: prvProcessExpiredTimer at /home/jonsmirl/esp/esp-idf/components/freertos/FreeRTOS-Kernel/timers.c:602
0x4219f4cd: prvProcessTimerOrBlockTask at /home/jonsmirl/esp/esp-idf/components/freertos/FreeRTOS-Kernel/timers.c:663
 (inlined by) prvTimerTask at /home/jonsmirl/esp/esp-idf/components/freertos/FreeRTOS-Kernel/timers.c:634
0x40382cda: vPortTaskWrapper at /home/jonsmirl/esp/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:134

ELF file SHA256: b7f98f5e11b9d56e

Rebooting...
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x40375b7c
0x40375b7c: esp_restart_noos at /home/jonsmirl/esp/esp-idf/components/esp_system/port/soc/esp32s3/system_internal.c:159
~~~
...

jonsmirl, 2024-06-09:
Commenting out this Log statement fixes the stackover flow
[image]

shubhamdp, 2024-06-25:
@jonsmirl We are working on this PTAL https://github.com/project-chip/connectedhomeip/pull/34050

dhrishi, 2024-08-14:
The connectedhomeip submodule is updated

closed 2024-08-14 by dhrishi