espressif/esp-matter #950: How to change commissioning values and generating new QR code?

amroncz, 2024-06-05:
Hello,

I have an ESP32-C6 and flashed it using the following documentation with the “light” example:
- https://docs.espressif.com/projects/esp-matter/en/latest/esp32c6/developing.html#esp-idf-setup
- https://docs.espressif.com/projects/esp-matter/en/latest/esp32c6/developing.html#esp-matter-setup

I then took the QR code from the commissioning documentation (https://docs.espressif.com/projects/esp-matter/en/latest/esp32c6/developing.html#commissioning-and-control) and was able to successfully connect the device to both Home Assistant and Alexa. Alexa briefly complained that it was not a real Matter product when I added it, but the setup was still successful and the device could be controlled successfully.

The QR code from the documentation mentioned above only worked, of course, because I used the exact “Light” example. Hence my question:
How do I change the values for commissioning and generate a new QR code for my device?

> 
> Version:             0
> Vendor ID:           65521    (0xFFF1)
> ProductID:           32768    (0x8000)
> Custom flow:         0        (STANDARD)
> Discovery Bitmask:   0x02     (BLE)
> Long discriminator:  3840     (0xf00)
> Passcode:            20202021

I have already read the following:
- https://docs.espressif.com/projects/esp-matter/en/latest/esp32c6/faq.html#a1-10-how-to-generate-matter-onboarding-codes-qr-code-and-manual-pairing-code

And I have tried mfg_tool, chip-tool and setup_payload:
- https://github.com/espressif/esp-matter/tree/main/tools/mfg_tool#readme
- https://github.com/project-chip/connectedhomeip/blob/master/docs/guides/chip_tool_guide.md
- https://github.com/project-chip/connectedhomeip/tree/master/src/setup_payload/python  

But I can't figure it out.

I tried mfg_tool (I don't understand the whole certificate thing. Do I have to have any? my ESP32 is only for my personal smart home), chip-tool and setup_payload:
- https://github.com/espressif/esp-matter/tree/main/tools/mfg_tool#readme
- https://github.com/project-chip/connectedhomeip/blob/master/docs/guides/chip_tool_guide.md
- https://github.com/project-chip/connectedhomeip/tree/master/src/setup_payload/python

I also found this: https://github.com/project-chip/connectedhomeip/issues/7015#issuecomment-863459390

But what is node-id?

In addition, my device was automatically connected to Home Assistant and Alexa via WIFI. However, I would like to connect it via Thread.

Is the value “Discovery Bitmask” for this? Can I just set it to Thread? Or do I have to set “rendezvous” for this? And if so, how?

Please can someone give me step-by-step instructions on how to easily change the values, generate a new QR code and connect my ESP32-C6 to Home Assistant and Alexa via Thread? Gladly using the “light” example.

amroncz, 2024-06-06:
> Is the value “Discovery Bitmask” for this? Can I just set it to Thread? Or do I have to set “rendezvous” for this? And if so, how?

Okay sorry I was stupid and apparently missed the following:

> ESP32-C6 supports both the Wi-Fi and IEEE 802.15.4 radio, so you can run Wi-Fi or Thread matter example on it.
>
>To enable Thread, you should change the menuconfig options to CONFIG_OPENTHREAD_ENABLED=y, CONFIG_ENABLE_WIFI_STATION=n, and CONFIG_USE_MINIMAL_MDNS=n.

So I set the values and flashed the device again. Now, however, I can no longer connect my ESP32-C6 to Home Assistant and also not directly to Alexa, which acts as a Thread Border Router.

shripad621git, 2024-06-06:
@amroncz , please follow the steps below to commission and control the device using custom generated QR code:
1. Please generate the factory partition using 
` ./mfg_tool.py --discriminator 3867 --passcode 20202028 --vendor-id 0xFFF2 --product-id 0x8001 
`
The above command is an example command for explanation purpose. You can change the values of these arguments as per your need.
2. In the output of mfg_tool command , you will get some output like
`[2024-06-06 15:27:18,051] [   INFO] - Generated output files at: out/fff2_8001/50b6c62b-18ae-456e-97da-a465e4122afd
`
i.e the file path at which the output files are located.
You will find all the `onboarding payloads including discriminator, passcode, manual pairing code and qrcode in a onboarding-codes.csv file and also a qrcode png` generated for this input parameters.

3. You will need to enable these options in the menuconfig  by `idf.py menuconfig`  of light example.
```
CONFIG_ENABLE_ESP32_FACTORY_DATA_PROVIDER=y
ENABLE_ESP32_DEVICE_INSTANCE_INFO_PROVIDER=y
CHIP_FACTORY_NAMESPACE_PARTITION_LABEL=nvs
```
4.  Please flash the factory-partition generated at the outptut path by the following command:
for ex: if the output path is ` out/fff2_8001/50b6c62b-18ae-456e-97da-a465e4122afd`, then the command will be
`esptool.py -p [port]  write_flash 0x10000  path/to/out/fff2_8001/b4886fe6-833e-46e8-a0fa-a249147b300a/b4886fe6-833e-46e8-a0fa-a249147b300a-partition.bin `

5. Flash the application using `idf.py flash` and now you can scan the QRCode  generated at the output path mentioned in step 2  as you did previously with the QRCode in esp-matter documentation and commission it with various ecosystems.

6. You can explore more options to set and flash in mfg_tool using `mfg_tool.py --help` and information about the factory data [here](https://github.com/project-chip/connectedhomeip/blob/master/docs/guides/esp32/factory_data.md).

dhrishi, 2024-06-10:
@amroncz Please close this if your issue is resolved

amroncz, 2024-06-10:
> @amroncz Please close this if your issue is resolved

I wasn't at home this weekend. I'll take another look at the issue in the next 2-3 days and leave feedback here.

amroncz, 2024-06-12:
> @amroncz , please follow the steps below to commission and control the device using custom generated QR code:
> 
>     1. Please generate the factory partition using
>        `./mfg_tool.py --discriminator 3867 --passcode 20202028 --vendor-id 0xFFF2 --product-id 0x8001 `
>        The above command is an example command for explanation purpose. You can change the values of these arguments as per your need.
> 
>     2. In the output of mfg_tool command , you will get some output like
>        `[2024-06-06 15:27:18,051] [   INFO] - Generated output files at: out/fff2_8001/50b6c62b-18ae-456e-97da-a465e4122afd `
>        i.e the file path at which the output files are located.
>        You will find all the `onboarding payloads including discriminator, passcode, manual pairing code and qrcode in a onboarding-codes.csv file and also a qrcode png` generated for this input parameters.
> 
>     3. You will need to enable these options in the menuconfig  by `idf.py menuconfig`  of light example.
> 
> 
> ```
> CONFIG_ENABLE_ESP32_FACTORY_DATA_PROVIDER=y
> ENABLE_ESP32_DEVICE_INSTANCE_INFO_PROVIDER=y
> CHIP_FACTORY_NAMESPACE_PARTITION_LABEL=nvs
> ```
> 
>     4. Please flash the factory-partition generated at the outptut path by the following command:
>        for ex: if the output path is ` out/fff2_8001/50b6c62b-18ae-456e-97da-a465e4122afd`, then the command will be
>        `esptool.py -p [port]  write_flash 0x10000  path/to/out/fff2_8001/b4886fe6-833e-46e8-a0fa-a249147b300a/b4886fe6-833e-46e8-a0fa-a249147b300a-partition.bin `
> 
>     5. Flash the application using `idf.py flash` and now you can scan the QRCode  generated at the output path mentioned in step 2  as you did previously with the QRCode in esp-matter documentation and commission it with various ecosystems.
>        You can verify the discriminator, vendor and product_id in firmware logs once you flash the application
>        `Advertise commission parameter vendorID=65521 productID=32768 discriminator=3867/15 cm=1 cp=0 `
> 
>     6. You can explore more options to set and flash in mfg_tool using `mfg_tool.py --help` and information about the factory data [here](https://github.com/project-chip/connectedhomeip/blob/master/docs/guides/esp32/factory_data.md).
> 
>     7. You can test out the entire procedure without using the certs mentioned in mfg_tool. But for more information about certificates you can refer [here](https://docs.espressif.com/projects/esp-matter/en/latest/esp32c6/certification.html#matter-factory-partition-binary).

Sorry for the late reply.
This step-by-step guide works like a charm. Thank you very much.

closed 2024-06-12 by amroncz

ronny-antoon, 2024-07-03:
Iam having trouble making this.
my commands in this oder:-

python -m esptool erase_flash

idf.py build flash

docker run --rm -v C:\Users\RonnyPC\Documents\MatterApplication:/project -w /project espressif/esp-matter:release-v1.3_idf_v5.2.1 esp-matter-mfg-tool --passcode 20202028 --vendor-id 0xFFF2 --product-id 0x8001

python -m esptool -b 921600 write_flash 0x14000 out\fff2_8001\ab406e7c-dbd8-435d-b96b-b074de82d557\ab406e7c-dbd8-435d-b96b-b074de82d557-partition.bin

and my partitions.csv
# Name,   Type, SubType, Offset,  Size, Flags
# Note: Firmware partition offset needs to be 64K aligned, initial 36K (9 sectors) are reserved for bootloader and partition table
esp_secure_cert,  0x3F, ,0x10000,    0x4000, encrypted
nvs,      data, nvs,     0x14000,   0x1A000,
nvs_keys, data, nvs_keys,0x2E000,   0x2000, encrypted
otadata,  data, ota,     0x30000,   0x2000,
phy_init, data, phy,     0x32000,   0x2000,
fctry,    data, nvs,     0x34000,  0xC000,
ota_0,    app,  ota_0,   0x40000,   0x5E0000,
ota_1,    app,  ota_1,   0x620000,  0x5E0000,
frontend, data, spiffs,  0xC00000,  0x350000,
coredump, data, coredump,0xF50000,  0x20000,

and in the sdconfig.default i add this 3 lines 
CONFIG_ENABLE_ESP32_FACTORY_DATA_PROVIDER=y
CONFIG_ENABLE_ESP32_DEVICE_INSTANCE_INFO_PROVIDER=y
CONFIG_CHIP_FACTORY_NAMESPACE_PARTITION_LABEL=nvs

does the order command write
is thier something else i should do ?