espressif/esp-matter #951: The device adds the local domain name of mdns

WOLEN6914183, 2024-06-06:
I found that the bottom layer of matter uses minmdns. Now I want to build a webserver control web page for my device. I need a domain name of Abc.local for browser access. How should I initialize a domain name with minmdns?

jonsmirl, 2024-06-06:
CONFIG_LWIP_LOCAL_HOSTNAME in menuconfig

jonsmirl, 2024-06-06:
I have already built this and didn't find it be useful. It is too confusing when there are multiple device. It is better to use a phone app with custom clusters.

WOLEN6914183, 2024-06-07:
[image]
Okay, thank you for your answer. I tried the link you gave to add a domain name in the early days and it didn't work very well. Later I directly added the Advertise function under the official underlying file of matter: connectedhomeip/src/lib/dnssd/Advertiser_ImplMinimalMdns.cpp.
The addition can be successful, but memory leaks occasionally occur. I don’t know if Matter officially has an interface for adding new domain names to minmdns. At the same time, I also saw an example of minmdns in the official example folder. I I feel like I can give it a try

jonsmirl, 2024-06-07:
You probably need to turn this off, CONFIG_USE_MINIMAL_MDNS=y,  so that it will use the Espressif MDNS. With that set it is using MDNS from CHIP code.

WOLEN6914183, 2024-06-07:
Okay, I just tried this and it works, thank you very much

closed 2024-06-07 by WOLEN6914183