espressif/esp-matter #952: I have an issue adding BLE to my current project. I have added everything in the CMakeLists file and specified the path to the files

vishnuprasadvp12345, 2024-06-06:
/home/vishnu/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld: esp-idf/main/libmain.a(ble_commission.cpp.obj):(.literal._ZL17gap_event_handler22esp_gap_ble_cb_event_tP22esp_ble_gap_cb_param_t+0x18): undefined reference to `esp_ble_gap_start_advertising'
/home/vishnu/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld: esp-idf/main/libmain.a(ble_commission.cpp.obj):(.literal._Z23example_write_event_envhP18prepare_type_env_tP24esp_ble_gatts_cb_param_t+0xc): undefined reference to `esp_ble_gatts_send_response'
/home/vishnu/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld: esp-idf/main/libmain.a(ble_commission.cpp.obj):(.literal._ZL29gatts_profile_b_event_handler20esp_gatts_cb_event_thP24esp_ble_gatts_cb_param_t+0x5c): undefined reference to `esp_ble_gatts_create_service'
/home/vishnu/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld: esp-idf/main/libmain.a(ble_commission.cpp.obj):(.literal._ZL29gatts_profile_b_event_handler20esp_gatts_cb_event_thP24esp_ble_gatts_cb_param_t+0x60): undefined reference to `esp_ble_gatts_send_indicate'
/home/vishnu/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld: esp-idf/main/libmain.a(ble_commission.cpp.obj):(.literal._ZL29gatts_profile_b_event_handler20esp_gatts_cb_event_thP24esp_ble_gatts_cb_param_t+0x64): undefined reference to `esp_ble_gatts_start_service'
/home/vishnu/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld: esp-idf/main/libmain.a(ble_commission.cpp.obj):(.literal._ZL29gatts_profile_b_event_handler20esp_gatts_cb_event_thP24esp_ble_gatts_cb_param_t+0x68): undefined reference to `esp_ble_gatts_add_char'
/home/vishnu/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld: esp-idf/main/libmain.a(ble_commission.cpp.obj):(.literal._ZL29gatts_profile_b_event_handler20esp_gatts_cb_event_thP24esp_ble_gatts_cb_param_t+0x6c): undefined reference to `esp_ble_gatts_add_char_descr'
/home/vishnu/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld: esp-idf/main/libmain.a(ble_commission.cpp.obj):(.literal._ZL29gatts_profile_a_event_handler20esp_gatts_cb_event_thP24esp_ble_gatts_cb_param_t+0x54): undefined reference to `esp_ble_gap_set_device_name'
/home/vishnu/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld: esp-idf/main/libmain.a(ble_commission.cpp.obj):(.literal._ZL29gatts_profile_a_event_handler20esp_gatts_cb_event_thP24esp_ble_gatts_cb_param_t+0x58): undefined reference to `esp_ble_gap_config_adv_data'
/home/vishnu/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld: esp-idf/main/libmain.a(ble_commission.cpp.obj):(.literal._ZL29gatts_profile_a_event_handler20esp_gatts_cb_event_thP24esp_ble_gatts_cb_param_t+0x5c): undefined reference to `esp_ble_gatts_get_attr_value'
/home/vishnu/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld: esp-idf/main/libmain.a(ble_commission.cpp.obj):(.literal._ZL29gatts_profile_a_event_handler20esp_gatts_cb_event_thP24esp_ble_gatts_cb_param_t+0x60): undefined reference to `esp_ble_gap_update_conn_params'
/home/vishnu/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld: esp-idf/main/libmain.a(ble_commission.cpp.obj):(.literal._Z8ble_callv+0x4c): undefined reference to `esp_bluedroid_init_with_cfg'
/home/vishnu/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld: esp-idf/main/libmain.a(ble_commission.cpp.obj):(.literal._Z8ble_callv+0x50): undefined reference to `esp_bluedroid_enable'
/home/vishnu/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld: esp-idf/main/libmain.a(ble_commission.cpp.obj):(.literal._Z8ble_callv+0x54): undefined reference to `esp_ble_gatts_register_callback'
/home/vishnu/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld: esp-idf/main/libmain.a(ble_commission.cpp.obj):(.literal._Z8ble_callv+0x58): undefined reference to `esp_ble_gap_register_callback'
/home/vishnu/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld: esp-idf/main/libmain.a(ble_commission.cpp.obj):(.literal._Z8ble_callv+0x5c): undefined reference to `esp_ble_gatts_app_register'
/home/vishnu/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld: esp-idf/main/libmain.a(ble_commission.cpp.obj):(.literal._Z8ble_callv+0x60): undefined reference to `esp_ble_gatt_set_local_mtu'
/home/vishnu/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld: esp-idf/main/libmain.a(ble_commission.cpp.obj): in function `_ZL17gap_event_handler22esp_gap_ble_cb_event_tP22esp_ble_gap_cb_param_t':
/home/vishnu/esp-matter/examples/matterhub/main/ble_commission.cpp:221:(.text._ZL17gap_event_handler22esp_gap_ble_cb_event_tP22esp_ble_gap_cb_param_t+0x36): undefined reference to `esp_ble_gap_start_advertising'
/home/vishnu/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld: esp-idf/main/libmain.a(ble_commission.cpp.obj): in function `_Z23example_write_event_envhP18prepare_type_env_tP24esp_ble_gatts_cb_param_t':
/home/vishnu/esp-matter/examples/matterhub/main/ble_commission.cpp:283:(.text._Z23example_write_event_envhP18prepare_type_env_tP24esp_ble_gatts_cb_param_t+0x97): undefined reference to `esp_ble_gatts_send_response'
/home/vishnu/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld: /home/vishnu/esp-matter/examples/matterhub/main/ble_commission.cpp:301:(.text._Z23example_write_event_envhP18prepare_type_env_tP24esp_ble_gatts_cb_param_t+0x108): undefined reference to `esp_ble_gatts_send_response'
/home/vishnu/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld: esp-idf/main/libmain.a(ble_commission.cpp.obj): in function `_ZL29gatts_profile_b_event_handler20esp_gatts_cb_event_thP24esp_ble_gatts_cb_param_t':
/home/vishnu/esp-matter/examples/matterhub/main/ble_commission.cpp:558:(.text._ZL29gatts_profile_b_event_handler20esp_gatts_cb_event_thP24esp_ble_gatts_cb_param_t+0x8a): undefined reference to `esp_ble_gatts_create_service'
/home/vishnu/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld: /home/vishnu/esp-matter/examples/matterhub/main/ble_commission.cpp:571:(.text._ZL29gatts_profile_b_event_handler20esp_gatts_cb_event_thP24esp_ble_gatts_cb_param_t+0xf2): undefined reference to `esp_ble_gatts_send_response'
/home/vishnu/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld: /home/vishnu/esp-matter/examples/matterhub/main/ble_commission.cpp:603:(.text._ZL29gatts_profile_b_event_handler20esp_gatts_cb_event_thP24esp_ble_gatts_cb_param_t+0x1ee): undefined reference to `esp_ble_gatts_send_indicate'
/home/vishnu/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld: /home/vishnu/esp-matter/examples/matterhub/main/ble_commission.cpp:620:(.text._ZL29gatts_profile_b_event_handler20esp_gatts_cb_event_th [... cut ...]
/home/vishnu/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld: /home/vishnu/esp-matter/examples/matterhub/main/ble_commission.cpp:802:(.text._Z8ble_callv+0x1ba): undefined reference to `esp_ble_gatt_set_local_mtu'
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
ninja failed with exit code 1, output of the command is in the /home/vishnu/esp-matter/examples/matterhub/build/log/idf_py_stderr_output_237760 and /home/vishnu/esp-matter/examples/matterhub/build/log/idf_py_stdout_output_237760

dhrishi, 2024-06-06:
@vishnuprasadvp12345 Please provide details as requested in the issue template. Please upload logs and not paste. Also, for any changes you've done on top of any default code/examples, please attach a patch.

Please create a new issue with the all the details

closed 2024-06-06 by dhrishi

vishnuprasadvp12345, 2024-06-06:
I am currently working on the Matter controller example code and want to add BLE to that project. I have provided all the necessary paths in my CMakeLists file, but I still encounter this error.