espressif/esp-matter #953: Unable to add ESP32-C6 to an existing Thread network

amroncz, 2024-06-06:
Hello,

I have an ESP32-C6 and flashed it with the “light” example.

I used the exact example and didn't change anything. I was able to integrate my device with the example QR code (https://docs.espressif.com/projects/esp-matter/en/latest/esp32c6/developing.html#commissioning-and-control) into both Home Assistant and Alexa.

After I adjusted the values for the device for Thread as described here (https://docs.espressif.com/projects/esp-matter/en/latest/esp32c6/developing.html#flashing-the-firmware)

> ESP32-C6 supports both the Wi-Fi and IEEE 802.15.4 radio, so you can run Wi-Fi or Thread matter example on it.
>
>To enable Thread, you should change the menuconfig options to CONFIG_OPENTHREAD_ENABLED=y, CONFIG_ENABLE_WIFI_STATION=n, and CONFIG_USE_MINIMAL_MDNS=n.

 and flashed it again, I can neither integrate it into Home Assistant nor directly into Alexa via my Echo, which acts as a Thread border router.

shripad621git, 2024-06-06:
@amroncz , can you please provide the below details
ESP-Matter Commit Id: 
ESP-IDF Commit Id: 
SoC (eg: ESP32 or ESP32-C3): ESP32s3
Device Logs (Please attach the log file):
Host Machine OS: 
Commissioner app and versions if present:
Commissioner's logs if present:

dhrishi, 2024-06-06:
@amroncz If you want to quickly try the light example with ESP32-C6 Thread enabled, you can choose `ESP32C6` in the `thread_light` option in [ESP Matter Launchpad](https://espressif.github.io/esp-launchpad/?flashConfigURL=https://espressif.github.io/esp-matter/launchpad.toml)

amroncz, 2024-06-12:
> @amroncz , can you please provide the below details

ESP-Matter Commit Id: cd83d634248cdf142c6898c61719a60f981c0622
ESP-IDF Commit Id: 3b8741b172dc951e18509698dee938304bcf1523
SoC (eg: ESP32 or ESP32-C3): ESP32s3
Device Logs (Please attach the log file): [esp32-c6_logs.txt](https://github.com/user-attachments/files/15811056/esp32-c6_logs.txt)
Host Machine OS: Microsoft Windows (Version 10.0.19045.4412) with WSL (Ubuntu 22.04.4 LTS)
Commissioner app and versions if present: Amazon Alexa (Version: 2.2.567992.0), Home Assistant Companion App (2024.6.1-full)
Commissioner's logs if present: -

> @amroncz If you want to quickly try the light example with ESP32-C6 Thread enabled, you can choose `ESP32C6` in the `thread_light` option in [ESP Matter Launchpad](https://espressif.github.io/esp-launchpad/?flashConfigURL=https://espressif.github.io/esp-matter/launchpad.toml)

Unfortunately, that won't help me any further. I'm trying to learn how to use ESP devices and Matter using the example. So I'm trying to understand step by step what I need to do.

shripad621git, 2024-06-14:
Can you share your sdkconfig in the above case where you want to commission `esp32c6 over thread`. From the logs, it seems that the options you had set are not enabled in the sdkconfig.

amroncz, 2024-06-17:
> Can you share your sdkconfig in the above case where you want to commission `esp32c6 over thread`. From the logs, it seems that the options you had set are not enabled in the sdkconfig.

Here is my sdkconfig file. I use the Visual Studio esp-idf extension for the configuration.

[sdkconfig.txt](https://github.com/user-attachments/files/15875200/sdkconfig.txt)

shripad621git, 2024-06-18:
@amroncz , I have tried to reproduce the issue with esp32c6 and thread options enabled and my device got successfully commissioned over thread. You seem to be using `idf v5.2.2.` I have tested the things with the` idf v5.2.2` setup as well. The device gets commissioned successfully over thread without any wifi logs. 
You can try the steps below and get back:
i) Please do a clean build using `rm -rf build sdkconfig sdkconfig.old managed_components dependencies.lock`.
ii) set the target as esp32c6 using `idf.py set-target esp32c6.`
iii) `Enable the thread related options` you mentioned above.
iv) Flash the app using ` idf.py erase-flash flash monitor` and `commission the device using alexa or apple.`

shripad621git, 2024-06-20:
@amroncz , any update on the issue? Please close the issue if it has been addressed.

amroncz, 2024-06-20:
> @amroncz , I have tried to reproduce the issue with esp32c6 and thread options enabled and my device got successfully commissioned over thread. You seem to be using `idf v5.2.2.` I have tested the [...]

That worked for me. I was able to control the device via Matter over Thread with both Alexa and Home Assistant.
Alexa indicated that it was not a valid Matter device, but I had that with Matter over Wifi too and the configuration still worked.

Probably the clean build or erase-flash did it...

Many thanks for your help @shripad621git