espressif/esp-matter #954: vscode espressif extension not able to show esp-matter

cnuber14624, 2024-06-06:
after selecting show examples from espressif extension screen remains empty
in vscode server log on wsl exception is found

softlinks are pointing to non existing targets. After I've deleted failed links Wifi* in that folder "Show examples"is working again for esp-matter

2024-06-06 16:31:53.765 [info] Extension host with pid 699 started
2024-06-06 16:31:53.947 [info] ExtensionService#_doActivateExtension vscode.git-base, startup: true, activationEvent: '*', root cause: vscode.git
2024-06-06 16:31:54.157 [info] ExtensionService#_doActivateExtension vscode.git, startup: true, activationEvent: '*'
2024-06-06 16:31:54.159 [info] ExtensionService#_doActivateExtension vscode.github, startup: true, activationEvent: '*'
2024-06-06 16:31:55.153 [info] Eager extensions activated
2024-06-06 16:31:55.153 [info] ExtensionService#_doActivateExtension vscode.debug-auto-launch, startup: false, activationEvent: 'onStartupFinished'
2024-06-06 16:31:55.154 [info] ExtensionService#_doActivateExtension vscode.merge-conflict, startup: false, activationEvent: 'onStartupFinished'
2024-06-06 16:31:57.370 [info] ExtensionService#_doActivateExtension espressif.esp-idf-extension, startup: false, activationEvent: 'onView:idfCommands'
2024-06-06 16:35:13.178 [error] Error: ENOENT: no such file or directory, stat '/home/chris/esp-matter/examples/common/blemesh_platform/platform/ESP32_custom/WiFiDnssdImpl.cpp'
        at Object.statSync (node:fs:1690:3)
        at Object.statSync (/home/chris/.vscode-server/bin/dc96b837cf6bb4af9cd736aa3af08cf8279f7685/node_modules/graceful-fs/polyfills.js:319:16)

jonsmirl, 2024-06-06:
I have this in cpp_properties.json
~~~
{
    "configurations": [
        {
            "name": "ESP-IDF",
            "cStandard": "c11",
            "cppStandard": "c++17",
            "includePath": [
                "${default}",
                "/home/jonsmirl/aosp/esp-matter",
            ],
            "compileCommands": "${workspaceFolder}/build/compile_commands.json",
            "compilerPath": "/home/jonsmirl/.espressif/tools/xtensa-esp32s3-elf/esp-12.2.0_20230208/xtensa-esp32s3-elf/bin/xtensa-esp32s3-elf-gcc"
        }
    ],
    "version": 4
}
~~~

cnuber14624, 2024-06-07:
this is mine
{
    "configurations": [
        {
            "name": "ESP-IDF",
            "compilerPath": "${config:idf.toolsPath}/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/xtensa-esp32-elf-gcc",
            "compileCommands": "${workspaceFolder}/build/compile_commands.json",
            "includePath": [
                "${config:idf.espIdfPath}/components/**",
                "${config:idf.espIdfPathWin}/components/**",
                "${config:idf.espAdfPath}/components/**",
                "${config:idf.espAdfPathWin}/components/**",
                "${workspaceFolder}/**"
            ],
            "browse": {
                "path": [
                    "${config:idf.espIdfPath}/components",
                    "${config:idf.espIdfPathWin}/components",
                    "${config:idf.espAdfPath}/components/**",
                    "${config:idf.espAdfPathWin}/components/**",
                    "${workspaceFolder}"
                ],
                "limitSymbolsToIncludedHeaders": false
            }
        }
    ],
    "version": 4
}

sayondeep, 2024-06-10:
@cnuber14624 can you please raise the issue [here](https://github.com/espressif/vscode-esp-idf-extension/issues), for better support and resolution.

dhrishi, 2024-06-20:
Closing this here. Please raise an issue with the repo mentioned above.

closed 2024-06-20 by dhrishi