espressif/esp-matter #955: Generic switch states

Leczok12, 2024-06-07:
Hi, I flashed a generic switch example on my esp32 and i added it to my home assistant. When I press button states change to fast if even change, becous in event type proposition i have only unknow and initial state. I done somthing wrong or its just a bug.

**Additional context**
- Nothing changed in example 
- MFG binaries are flashed to

jadhavrohit924, 2024-06-11:
@Leczok12 What are you trying to achieve and what are the issues that you are facing? Can you please help me to understand?

Leczok12, 2024-06-11:
I want achive somthing like smart button and I flashed a generic_switch example and connected device to my home assistant. I don't know why I recive only **initial press** event.
In terminal I see a lot of events like long button release etc.

Nothing changed without changing gpio from 0 to 17
##### before 
```cpp
err = create_button(NULL, node);
```
##### after
```cpp
struct gpio_button button;
button.GPIO_PIN_VALUE = GPIO_NUM_17;
create_button(&button, node);
```
I flashed mfg_binaries too

dhrishi, 2024-06-11:
Is this same as https://github.com/espressif/esp-matter/issues/962 ? If yes, please close that. We can continue the discussion on this issue

jadhavrohit924, 2024-06-11:
Hi @Leczok12, Apple only supports single press event in their home app. That's you you receive only one event.

Leczok12, 2024-06-11:
And even home assistant can't handle those events?

jadhavrohit924, 2024-06-12:
Most likely, Yes.

closed 2024-06-12 by Leczok12

PeterKjaerFisker, 2024-11-01:
Hi. 
I am new here and I don't know if I am supposed to write on a closed issue, but I had the same issue and it was solved by this. 
https://github.com/home-assistant/core/issues/117584#issuecomment-2177287369

It seems like the example code for the generic switch does not set the FeatureMap attribute correctly right now, but by adding the lines mentioned in the link above, I got the FeatureMap to be correct and now Home Assistant can decode how the switch is supposed to work. 

I also had to change a line the app_driver.cpp code for multi-press to work correctly in Home Assistant, but I don't know why my changes fixed it, so that is probably not relevant for the discussion here. 

I hope it is okay to write here, if not then please guide me to what I should do instead.