espressif/esp-matter #959: Generic Switch, sending an event - getting error and restating the esp32

ronny-antoon, 2024-06-09:
**Describe the bug**
I want to make Stateless Switch in Apple HomeKit.
for that iam using the Generic Switch and enabling the single, double, long events.

when the event fires it should send appropriate event to the matter network.
in sending the event i got error:-

E (97081) chip[DL]: Chip stack locking error at './managed_components/espressif__esp_matter/connectedhomeip/connectedhomeip/src/app/EventManagement.cpp:416'. Code is unsafe/racy
E (97111) chip[-]: chipDie chipDie chipDie

**Environment**
- esp matter as a component:  0.0.2
- SoC (eg: ESP32 or ESP32-C3): esp32
- Host Machine OS: HomePod Mini, Iphone 13, all updated.
- Commissioner app and versions if present: Apple Home

**My Code**
[New folder (2).zip](https://github.com/user-attachments/files/15752874/New.folder.2.zip)

ronny-antoon, 2024-06-09:
Using this code alone i got the errors,
```

esp_matter::cluster::switch_cluster::event::send_multi_press_complete(
            endpoint_id, 0, 1);

```

the solution is to add a CHIP::lock/unclock to the event::send*** functions,
a workaround to solve this issue:-
```

esp_matter::lock::chip_stack_lock(portMAX_DELAY);
        ESP_LOGW(__FILENAME__, "SinglePress"); // In run time i got this log, then error.
        esp_matter::cluster::switch_cluster::event::send_multi_press_complete(
            endpoint_id, 0, 1);
        esp_matter::lock::chip_stack_unlock();

```

dhrishi, 2024-06-10:
@ronny-antoon The same example provides a reference for sending events. Please see if [this](https://github.com/espressif/esp-matter/blob/main/examples/generic_switch/main/app_driver.cpp#L130-#L136) helps

ronny-antoon, 2024-06-10:
> @ronny-antoon The same example provides a reference for sending events. Please see if [this](https://github.com/espressif/esp-matter/blob/main/examples/generic_switch/main/app_driver.cpp#L130-#L136) helps

I think it should be internally.
Maybe

closed 2024-06-19 by ronny-antoon