espressif/esp-matter #960: how to integrate ble server example cose to esp matter controller example code

vishnuprasadvp12345, 2024-06-10:
Hi all, I am currently working on the ESP Matter controller example code. I want to provide the SSID and password through BLE. How do I integrate the BLE example code into the Matter controller example code?

wqx6, 2024-06-11:
Currently the controller example can use `matter esp controller pairing ble-wifi <node-id> <ssid> <password> <pincode> <discriminator>` command to send the SSID and password to end devices through BLE.

vishnuprasadvp12345, 2024-06-11:
For local control, this command gets data from Bluetooth and sends it through the controller, so I need to integrate BLE into my controller example code. Please provide instructions on how to implement local control through an app and help me integrate BLE into this controller code

dhrishi, 2024-06-11:
@vishnuprasadvp12345 Didn't quite understand your requirement. In the controller example, it commissions other devices using BLE and then later controls them on the local network (IP) over Matter.

What do you want to implement?

dhrishi, 2024-06-11:
@vishnuprasadvp12345 Sorry. But it is still not clear. Which commands do you want? What is the purpose of those commands?

vishnuprasadvp12345, 2024-06-11:
I want to control all the controller commands through BLE. How can I add BLE code to the controller code if I don't know how to handle all the aspects related to BLE? and path in cmake list

dhrishi, 2024-06-11:
You cannot do that. The Matter controller CLI commands are supposed to be executed on the local network using Matter after the device is commissioned. Only a part of the `ble-wifi` or `ble-thread` command happens over BLE.

These commands expect that a secure CASE session is established between the end device and the controller. And, in a way, are similar to a Matter phone app sending read/write Matter commands to the end device.

vishnuprasadvp12345, 2024-06-11:
I'm implementing MQTT code in this controller code, and I can easily control end devices using this function:

esp_matter::lock::chip_stack_lock(portMAX_DELAY);
int argc = sizeof(argv) / sizeof(argv[0]);
esp_matter::controller::send_invoke_cluster_command(2, endpoint, cluster_id, command_id,
                                                    5 > 4 ? argv[0] : NULL);
esp_matter::lock::chip_stack_unlock();
I provide data in the endpoint and cluster_id variables through MQTT. Similarly, I want to implement BLE in this code and control devices locally through BLE so how to implement ble

vishnuprasadvp12345, 2024-06-11:
Currently, we provide data through the console, so I need to configure BLE in this code to handle all the commands through BLE. How do I configure BLE in this code?

dhrishi, 2024-06-11:
So you want your phone to give Wi-Fi connectivity to the device using BLE, right?

And what is the role of Matter here? Or this phone app is going to be a Matter phone app? If yes, why not use the standard Matter commissioning process that helps a device connect to the Wi-Fi network too?

vishnuprasadvp12345, 2024-06-11:
I am using the Matter controller example code and currently provide all Matter commands through the console. Similarly, I want to send all Matter commands via BLE. How can I integrate BLE into the controller code?

dhrishi, 2024-06-11:
>I am using the Matter controller example code and currently provide all Matter commands through the console.

The console commands entered on the controller finally work on the IP (Wi-Fi) network

Do you want to send the Matter commands over BLE from your phone app to the Matter end device or the Matter controller?

vishnuprasadvp12345, 2024-06-12:
matter controller

dhrishi, 2024-06-12:
Some additional questions -
1. How do you want the Matter controller to connect to Wi-Fi? Using the same phone app, over BLE? You will send the SSID and passphrase
2. For end device commissioning, will you trigger it through the phone app? If yes, how will you get the discriminator and passcode and make it reach the controller?
3. Why not build a Matter phone app instead? The app can directly commission end devices and then you can also use the Matter controller (if required) to control the end devices. Reference implementation - https://github.com/espressif/esp-rainmaker/tree/master/examples/matter/matter_controller

vishnuprasadvp12345, 2024-06-12:
Basically, I want to create a Matter hub, so I chose the Matter controller example code. We have our own app, and I want to remotely control the device through our app, so I need to provide the SSID and password through BLE to the Matter controller. Therefore, I need to integrate BLE into the Matter controller to pass the SSID and password.

dhrishi, 2024-06-17:
Creating your own Matter hub requires a fabric solution too. Is this going to be a production-level solution, the Matter controller? If yes, you will need your own Matter-enabled fabric phone app to set it up. The standard commissioners would not understand the "controller" related functionality that is supported. 

The BLE communication you are talking about, comes after the controller is setup. And, if it is your own fabric, then you do not need this BLE related functionality. Your phone app can locally talk over Matter and then for remote access, using your cloud, with the controller.

closed 2024-07-26 by vishnuprasadvp12345