espressif/esp-matter #961: Mode Select and mfg_tool

jonsmirl, 2024-06-10:
If the labels for mode select aren't set correctly you get an obscure error.... This is because I altered the ep order and now the ep in flash on mode select doesn't match my actual ep. This scheme of ep number being stored in flash also does not work for a bridged device. These label need to be set from code.

The fix for this error is to make the ep given to mfg_tool match the actual ep.
~~~
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x420d76e0  PS      : 0x00060730  A0      : 0x820d9cfe  A1      : 0x3fcd2fc0  
0x420d76e0: chip::app::DataModel::Encode(chip::TLV::TLVWriter&, chip::TLV::Tag, chip::Span<char const>) at /home/jonsmirl/l6/build/esp-idf/chip/../../../../aosp/esp-matter/connectedhomeip/connectedhomeip/config/esp32/third_party/connectedhomeip/src/app/data-model/Encode.h:95
 (inlined by) void chip::app::DataModel::WrappedStructEncoder::Encode<chip::Span<char const> const&>(unsigned char, chip::Span<char const> const&) at /home/jonsmirl/l6/build/esp-idf/chip/../../../../aosp/esp-matter/connectedhomeip/connectedhomeip/config/esp32/third_party/connectedhomeip/src/app/data-model/WrappedStructEncoder.h:45
 (inlined by) void chip::app::DataModel::WrappedStructEncoder::Encode<chip::Span<char const> const&>(unsigned char, chip::Span<char const> const&) at /home/jonsmirl/l6/build/esp-idf/chip/../../../../aosp/esp-matter/connectedhomeip/connectedhomeip/config/esp32/third_party/connectedhomeip/src/app/data-model/WrappedStructEncoder.h:42

A2      : 0x3fcd3000  A3      : 0x00000000  A4      : 0xbaad5678  A5      : 0x00000000  
A6      : 0x00000016  A7      : 0x3fcd323c  A8      : 0x00000000  A9      : 0x3fcd2fb0  
A10     : 0x00000000  A11     : 0x3fcd323c  A12     : 0x00000000  A13     : 0x61d9430e  
A14     : 0x00000000  A15     : 0x3fc9f548  SAR     : 0x00000010  EXCCAUSE: 0x0000001c  
EXCVADDR: 0xbaad5678  LBEG    : 0x40056f5c  LEND    : 0x40056f72  LCOUNT  : 0x00000000  
0x40056f5c: memcpy in ROM
0x40056f72: memcpy in ROM

Backtrace: 0x420d76dd:0x3fcd2fc0 0x420d9cfb:0x3fcd2ff0 0x42083d7d:0x3fcd3030 0x420757bd:0x3fcd3090 0x42076680:0x3fcd3130 0x420767bb:0x3fcd3150 0x42076ea4:0x3fcd3230 0x420770f3:0x3fcd3350 0x42077178:0x3fcd3390 0x420c154d:0x3fcd33b0 0x420c155d:0x3fcd33d0 0x420dcdbd:0x3fcd33f0 0x420dd165:0x3fcd3410 0x420dd1ad:0x3fcd3480 0x40382cda:0x3fcd34a0
0x420d76dd: void chip::app::DataModel::WrappedStructEncoder::Encode<chip::Span<char const> const&>(unsigned char, chip::Span<char const> const&) at /home/jonsmirl/l6/build/esp-idf/chip/../../../../aosp/esp-matter/connectedhomeip/connectedhomeip/config/esp32/third_party/connectedhomeip/src/app/data-model/WrappedStructEncoder.h:44 (discriminator 2)
0x420d9cfb: chip::app::Clusters::ModeSelect::Structs::ModeOptionStruct::Type::Encode(chip::TLV::TLVWriter&, chip::TLV::Tag) const at /home/jonsmirl/l6/build/esp-idf/chip/../../../../aosp/esp-matter/connectedhomeip/connectedhomeip/config/esp32/third_party/connectedhomeip/zzz_generated/app-common/app-common/zap-generated/cluster-objects.cpp:10500
0x42083d7d: chip::ChipError chip::app::DataModel::Encode<chip::app::Clusters::ModeSelect::Structs::ModeOptionStruct::Type, (chip::app::Clusters::ModeSelect::Structs::ModeOptionStruct::Type*)0>(chip::TLV::TLVWriter&, chip::TLV::Tag, chip::app::Clusters::ModeSelect::Structs::ModeOptionStruct::Type const&) at /home/jonsmirl/aosp/esp-matter/connectedhomeip/connectedhomeip/src/app/data-model/Encode.h:117
 (inlined by) chip::ChipError chip::app::AttributeReportBuilder::EncodeValue<chip::app::Clusters::ModeSelect::Structs::ModeOptionStruct::Type const&, true>(chip::app::AttributeReportIBs::Builder&, chip::TLV::Tag, chip::app::Clusters::ModeSelect::Structs::ModeOptionStruct::Type const&) at /home/jonsmirl/aosp/esp-matter/connectedhomeip/connectedhomeip/src/app/AttributeAccessInterface.h:82
 (inlined by) chip::ChipError chip::app::AttributeValueEncoder::EncodeListItem<chip::app::Clusters::ModeSelect::Structs::ModeOptionStruct::Type const&>(chip::app::Clusters::ModeSelect::Structs::ModeOptionStruct::Type const&) at /home/jonsmirl/aosp/esp-matter/connectedhomeip/connectedhomeip/src/app/AttributeAccessInterface.h:278
 (inlined by) chip::ChipError chip::app::AttributeValueEncoder::ListEncodeHelper::Encode<chip::app::Clusters::ModeSelect::Structs::ModeOptionStruct::Type const&, true>(chip::app::Clusters::ModeSelect::Structs::ModeOptionStruct::Type const&) const at /home/jonsmirl/aosp/esp-matter/connectedhomeip/connectedhomeip/src/app/AttributeAccessInterface.h:127
 (inlined by) operator()<chip::app::AttributeValueEncoder::ListEncodeHelper> at /home/jonsmirl/aosp/esp-matter/connectedhomeip/connectedhomeip/src/app/clusters/mode-select-server/mode-select-server.cpp:87
 (inlined by) EncodeList<(anonymous namespace)::ModeSelectAttrAccess::Read(const chip::app::ConcreteReadAttributePath&, chip::app::AttributeValueEncoder&)::<lambda(const auto:5&)> > at /home/jonsmirl/aosp/esp-matter/connectedhomeip/connectedhomeip/src/app/AttributeAccessInterface.h:227
 (inlined by) Read at /home/jonsmirl/aosp/esp-matter/connectedhomeip/connectedhomeip/src/app/clusters/mode-select-server/mode-select-server.cpp:82
0x420757bd: chip::app::ReadSingleClusterData(chip::Access::SubjectDescriptor const&, bool, chip::app::ConcreteReadAttributePath const&, chip::app::AttributeReportIBs::Builder&, chip::app::AttributeValueEncoder::AttributeEncodeState*) at /home/jonsmirl/aosp/esp-matter/connectedhomeip/connectedhomeip/src/app/util/ember-compatibility-functions.cpp:436
 (inlined by) chip::app::ReadSingleClusterData(chip::Access::SubjectDescriptor const&, bool, chip::app::ConcreteReadAttributePath const&, chip::app::AttributeReportIBs::Builder&, chip::app::AttributeValueEncoder::AttributeEncodeState*) at /home/jonsmirl/aosp/esp-matter/connectedhomeip/connectedhomeip/src/app/util/ember-compatibility-functions.cpp:570
0x42076680: chip::app::reporting::Engine::RetrieveClusterData(chip::Access::SubjectDescriptor const&, bool, chip::app::AttributeReportIBs::Builder&, chip::app::ConcreteReadAttributePath const&, chip::app::AttributeValueEncoder::AttributeEncodeState*) at /home/jonsmirl/aosp/esp-matter/connectedhomeip/connectedhomeip/src/app/reporting/Engine.cpp:93
0x420767bb: chip::app::reporting::Engine::BuildSingleReportDataAttributeReportIBs(chip::app::ReportDataMessage::Builder&, chip::app::ReadHandler*, bool*, bool*) at /home/jonsmirl/aosp/esp-matter/connectedhomeip/connectedhomeip/src/app/reporting/Engine.cpp:203 (discriminator 2)
0x42076ea4: chip::app::reporting::Engine::BuildAndSendSingleReportData(chip::app::ReadHandler*) at /home/jonsmirl/aosp/esp-matter/connectedhomeip/connectedhomeip/src/app/reporting/Engine.cpp:546
0x420770f3: chip::app::reporting::Engine::Run() at /home/jonsmirl/aosp/esp-matter/connectedhomeip/connectedhomeip/src/app/reporting/Engine.cpp:666
0x42077178: chip::app::reporting::Engine::Run(chip::System::Layer*, void*) at /home/jonsmirl/aosp/esp-matter/connectedhomeip/connectedhomeip/src/app/reporting/Engine.cpp:619
0x420c154d: chip::System::TimerData::Callback::Invoke() const at /home/jonsmirl/l6/build/esp-idf/chip/../../../../aosp/esp-matter/connectedhomeip/connectedhomeip/config/esp32/third_party/connectedhomeip/src/system/SystemTimer.h:61
 (inlined by) chip::System::TimerPool<chip::System::TimerList::Node>::Invoke(chip::System::TimerList::Node*) at /home/jonsmirl/l6/build/esp-idf/chip/../../../../aosp/esp-matter/connectedhomeip/connectedhomeip/config/esp32/third_party/connectedhomeip/src/system/SystemTimer.h:237
0x420c155d: chip::LambdaBridge::Initialize<chip::System::LayerImplFreeRTOS::ScheduleWork(void (*)(chip::System::Layer*, void*), void*)::{lambda()#1}>(chip::System::LayerImplFreeRTOS::ScheduleWork(void (*)(chip::System::Layer*, void*), void*)::{lambda()#1} const&)::{lambda(std::aligned_storage<24u, 4u>::type const&)#1}::_FUN(std::aligned_storage<24u, 4u>::type const&) at /home/jonsmirl/l6/build/esp-idf/chip/../../../../aosp/esp-matter/connectedhomeip/connectedhomeip/config/esp32/third_party/connectedhomeip/src/system/SystemLayerImplFreeRTOS.cpp:143
 (inlined by) operator() at /home/jonsmirl/l6/build/esp-idf/chip/../../../../aosp/esp-matter/connectedhomeip/connectedhomeip/config/esp32/third_party/connectedhomeip/src/lib/support/LambdaBridge.h:39
 (inlined by) _FUN at /home/jonsmirl/l6/build/esp-idf/chip/../../../../aosp/esp-matter/connectedhomeip/connectedhomeip/config/esp32/third_party/connectedhomeip/src/lib/support/LambdaBridge.h:39
[... cut ...] mManager.h:403
 (inlined by) chip::DeviceLayer::Internal::GenericPlatformManagerImpl_FreeRTOS<chip::DeviceLayer::PlatformManagerImpl>::EventLoopTaskMain(void*) at /home/jonsmirl/l6/build/esp-idf/chip/../../../../aosp/esp-matter/connectedhomeip/connectedhomeip/config/esp32/third_party/connectedhomeip/src/include/platform/internal/GenericPlatformManagerImpl_FreeRTOS.ipp:280
0x40382cda: vPortTaskWrapper at /home/jonsmirl/esp/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:134
~~~

jonsmirl, 2024-06-11:
I just wrote my own ModeOptionsProvider, pretty simple to do. I did not figure out what is messed up with modes and mfg_tool. I will move these strings into an external file for translations. This also allows me to deal with the mode-select clusters ending up on different endpoints in my bridge.
~~~
const char *labels[] = {"Switch", "Dimmer", "ELV Dimmer", "Smartbulb", "No Load"}; 
const int modes[] = {0, 1, 2, 3, 4};

SupportedModesManager::ModeOptionsProvider StaticSupportedModesManager::getModeOptionsProvider(EndpointId endpointId) const
{
    int count = sizeof(modes) / sizeof(int);

    ModeOptionStructType * modeOptionStructList = new ModeOptionStructType[count];
    if (modeOptionStructList == nullptr)
    {
        return ModeOptionsProvider(nullptr, nullptr);
    }
    for (int i = 0; i < count; i++) {
        Structs::ModeOptionStruct::Type option;

        option.label        = chip::CharSpan::fromCharString(labels[i]);
        option.mode         = static_cast<uint8_t>(modes[i]);
        option.semanticTags = DataModel::List<const SemanticTag>((SemanticTag *)NULL, 0);

        modeOptionStructList[i] = option;
    }

    return ModeOptionsProvider(modeOptionStructList, modeOptionStructList + count);   
}
~~~

jadhavrohit924, 2024-06-11:
Can you please share your endpoint composition initially and after change? 

Till we remove mode_select from mfg you can add it from the code. [Here](https://github.com/project-chip/connectedhomeip/blob/master/examples/all-clusters-app/all-clusters-common/src/static-supported-modes-manager.cpp) is the reference implementation that you will have to do in your example.

jonsmirl, 2024-06-11:
There really needs to a smart switch device type.

I moved the eps from being fixed to being on an internal bridge. Doing that let me remove and reconfigure them.
~~~
For a smart switch:
  Bridge descriptor
  Identify
  Groups
  Scenes
  Mode Select
  OnOff
  LevelControl
  ColorControl
  GenericSwitch
  OnOff Switch
  Level control switch
  Color control switch
  Fixed label
  User label
~~~