espressif/esp-matter #963: How can I add the hue circle on the light example ?

Eliostrim, 2024-06-11:
Hello there !

I flashed my esp32c3 with the light example from esp-matter.
I successfully bind this one to the google nest and i can control it with my phone.

All work fine but i see there is only the temperature feature.
I can add others endpoint without any problem.

but i can't add the hue circle to the existing light.

So there is my question : **How can i add the hue feature** ?

When i'm trying to do something it tell me "ATTRIBUTE CAN NOT BE NULL". (given by the attribute::get() function).

Printing with ```attribute::get_id(current_hue_attribute)```  return -1 instead of 0 according to Attribute.h line 5194.
But with ```attribute::get_id(color_temp_attribute)``` return 7 (the right value for the temperature feature still according to Attribute.h line 5222).

There is these lines i add in the app_main.cpp to check that :
```
    printf("light EP : %d, colorcontrol id: %ld , currentHue id: %ld \n", endpoint_id, cluster::get_id(color_control_cluster),attribute::get_id(current_hue_attribute));
    printf("light EP : %d, colorcontrol id: %ld , currenTemp id: %ld \n", endpoint_id, cluster::get_id(color_control_cluster),attribute::get_id(color_temp_attribute));
```
In addition, in the app_driver.cpp, in the app_driver_light_set_default function, the condition :
```val.val.u8 == (uint8_t)ColorControl::ColorMode::kCurrentHueAndCurrentSaturation```
is never right even if i change some things in the app_main.cpp

I think I'm doing it in a wrong way (obviously).
If anybody have an answer, feel free to share it ! 👍 

I can give you some logs if you want it but there is nothing more what i say before.

(If this question is already answered i didn't find it, so ... sorry for that)

Devices :
ESP32c3
Google Nest Hub
Samsung Galaxy A25

dhrishi, 2024-06-12:
@Eliostrim The light example [creates Extended Color light device type](https://github.com/espressif/esp-matter/blob/main/examples/light/main/app_main.cpp#L176). As per the Matter specification, XY and color temperature are mandatory for the extended color light. And hue, saturation are optional and hence not included by default.

If you want hue and saturation, you can add the hue_saturation (HS) feature using the esp-matter data model APIs. Add feature reference usage can be found in the [generic_switch example](https://github.com/espressif/esp-matter/blob/main/examples/generic_switch/main/app_main.cpp#L180-#L188)

[Reference for Matter specifications](https://github.com/espressif/esp-matter?tab=readme-ov-file#matter-specifications)

Eliostrim, 2024-06-14:
Thank you for your answer !
Thank's to your explanation, i see in the generic_switch example in the app_main.cpp line 174 : 
```
cluster_t *cluster = cluster::get(endpoint, Switch::Id);
````
 So to make what i want, i have to add just 3 lines to add this cluster.

I struggle many days with that....

I juste add theses following lines to the app_main.cpp without modifying anything else.

```
cluster_t *cluster_hue = cluster::get(endpoint_light,ColorControl::Id);
attribute_t *attribute_hue = attribute::get(cluster_hue, ColorControl::Attributes::EnhancedCurrentHue::Id);
attribute::set_deferred_persistence(attribute_hue);
```

Thank you again for your answer !

dhrishi, 2024-06-14:
@Eliostrim Can you please check if this helps

```
    /* Add additional features to the node */
    cluster_t *cluster = cluster::get(endpoint, ColorControl::Id);
    cluster::color_control::feature::hue_saturation::config_t hue_saturation_config;
    hue_saturation_config.current_hue = DEFAULT_HUE;
    hue_saturation_config.current_saturation = DEFAULT_SATURATION;
    cluster::color_control::feature::hue_saturation::add(cluster, &hue_saturation_config);
```

Eliostrim, 2024-06-17:
Add theses lines make theses followings warnings and errors

```
E (701) esp_matter_core: Attribute cannot be NULL
W (701) esp_matter_core: Attribute 0x00000000 on cluster 0x00000300 already exists. Not creating again.
W (711) esp_matter_core: Attribute 0x00000001 on cluster 0x00000300 already exists. Not creating again.
W (721) esp_matter_core: Command 0x00000000 on cluster 0x00000300 already exists. Not creating again.
W (731) esp_matter_core: Command 0x00000001 on cluster 0x00000300 already exists. Not creating again.
W (741) esp_matter_core: Command 0x00000002 on cluster 0x00000300 already exists. Not creating again.
W (751) esp_matter_core: Command 0x00000003 on cluster 0x00000300 already exists. Not creating again.
W (761) esp_matter_core: Command 0x00000004 on cluster 0x00000300 already exists. Not creating again.
W (771) esp_matter_core: Command 0x00000005 on cluster 0x00000300 already exists. Not creating again.
W (781) esp_matter_core: Command 0x00000006 on cluster 0x00000300 already exists. Not creating again.
E (791) esp_matter_core: Attribute cannot be NULL
```
And i see the cluster about the temperature of light disapear on my phone.

dhrishi, 2024-06-18:
@Eliostrim Please share the complete patch that can be applied on the standard esp-matter light examples. This will help us recreate and understand the behavior you are seeing

Eliostrim, 2024-06-19:
I don't know if we can call it a patch because it's only partially working .... but there is what i've done

There is the app_main of a modified light example (3 lines 196 to 198)

[app_main.txt](https://github.com/user-attachments/files/15898581/app_main.txt)

But, i tryed to reflash the esp32c3 with the original light but the temperature cluster doesn't come back on my phone. (i'm still using the google home app with a google nest hub) It's stuck on the hue wheel.

I also see there is two times it say "Attribute cannot be NULL"

There is the log
[log.txt](https://github.com/user-attachments/files/15898673/log.txt)

pimpalemahesh, 2024-06-20:
@Eliostrim I noticed that you are attempting to add attributes to the color control
 cluster. However, the hue_saturation feature is not a mandatory feature of the color_control cluster. Therefore, we need to explicitly add it to the cluster. Without this, the attributes related to hue_saturation will not be supported.

Please review the updated code file below. I hope this helps.
[app_main.txt](https://github.com/user-attachments/files/15914493/app_main.txt)

Eliostrim, 2024-06-21:
@pimpalemahesh Thanks for your answer.
But when we explicity use this line :
```
cluster::color_control::feature::hue_saturation::add(cluster_hue, &hue_saturation_config);
```
A lot of warning is appearing at the start of the ESP.
And it's not doing anything more.

And now, on the google home app, i have the hue wheel but the temperature of light just vanish from all of my light on ESP even if i try to back on original example from esp-matter.

There is the log 
[log.txt](https://github.com/user-attachments/files/15925047/log.txt)

pimpalemahesh, 2024-06-21:
@Eliostrim  I see that changes are not correctly reflecting on the google home app unless we recommission the device. Can you please try factory reset and recommission the device. Please comment if the issue persists.

Eliostrim, 2024-06-24:
@pimpalemahesh Hi, there is nothing more with a factory reset.

I notice the light bulb that i bought on amazon works fine but all of theses ESP i have flashed have their temperature cluster disapeared.

pimpalemahesh, 2024-06-28:
Can you provide the version of your Google Nest Hub and also attach any newly found logs.

Eliostrim, 2024-07-01:
@pimpalemahesh 

I don't know exactly what do you want but there is some version number on the Google Nest Hub
Google Nest Hub : 
Fuchsia Version : 18.20240225.3.63
Software version : 58.102.2.613521407
Firmware Chromecast : 3.73.411796

Aaaaaannnnnnd ... I don't know why, what happened and how but the temperature cluster came back

Maybe because of the wifi I work on is ... not stable ? 

But there is still much warning and one error about the attribute at starting

There is the log
[log.txt](https://github.com/user-attachments/files/16049177/log.txt)

pimpalemahesh, 2024-07-02:
@Eliostrim It is good to know that the clusters are working fine. I saw the logs that you provided. There are some warnings, but they are normal and can be ignored. I also noticed that it is working fine after clicking the toggle button.

Regarding the error you mentioned, "Attribute cannot be null," I am trying to reproduce it. Can you confirm if you are using the latest commit? If not, please mention the commit ID.

Additionally, if you have made any changes in the app_driver or any other file, please specify them.

Eliostrim, 2024-07-02:
@pimpalemahesh I think i don't have the latest commit because i saw there is a update.

I don't know really how to get the commit ID of the esp-matter i have but in the readme there is this one : d38a6496c3
And with the command ``git rev-parse HEAD`` or ``git log -1`` i got this one : 91bf33fc23b7673f6b29ff8bc791c655c9fa90c1

How can I update ?

pimpalemahesh, 2024-07-02:
@Eliostrim

I found that the "Attribute cannot be null" error is caused by some hue-saturation handling statements in the app_driver.cpp file. These are just indications and can be ignored. If you want to remove them, you can comment out the following lines:

/* Setting hue */
attribute = attribute::get(cluster, ColorControl::Attributes::CurrentHue::Id);
attribute::get_val(attribute, &val);
err |= app_driver_light_set_hue(handle, &val);
/* Setting saturation */
attribute = attribute::get(cluster, ColorControl::Attributes::CurrentSaturation::Id);
attribute::get_val(attribute, &val);
err |= app_driver_light_set_saturation(handle, &val);

This will work. To get the latest code, you can use the command ***_git pull origin main_***.

closed 2024-07-17 by Eliostrim