espressif/esp-matter #964: Switching Network Interface

supersunil, 2024-06-12:
Hi,
I have ESP32C6  working with Matter Light example.
Dynamic switching between  WiFi to thread network , i referred this gist link - https://gist.github.com/shubhamdp/76723dcedf6cdfe88d7e0c49602ccab2

I have recreated the code and it work as expected , 
The device is commissioned with Thread and it works fine, trying to switch from thread to WiFi without factory-reset (connected the SSID by using `matter esp wifi connect`). The device went to `no response`.  If i switch back to thread case session establishment successful and device is active.

Is it possible to establish the case session for WiFi without factory-reset? 

- esp-matter: 07152684689146eda1fe3ba8a41181823c180c48
- esp-idf: v5.1.2

wqx6, 2024-06-12:
The patches you used will have only one NetworkCommissioning cluster, and we will decide whether this cluster is used for WiFi or Thread before `esp_matter::start()` that's why you cannot dynamically switch network interface between Wi-Fi and Thread during runtime.

To dynamically switch network interface between Wi-Fi and Thread, you need to create two endpoints with different network commissioning clusters(one for Wi-Fi and another for Thread). And two NetworkCommissioning instances should be created for the two endpoint with WiFi and Thread driver. Note that the Generic Thread NetworkCommissioning instance is defined here https://github.com/project-chip/connectedhomeip/blob/master/src/platform/OpenThread/GenericThreadStackManagerImpl_OpenThread.hpp#L80. Then you can switch the network interface.

However, there is an issue when the Thread BR's backbone network is the same as the network to which the ESP32C6's WiFi network interface is connected. The controller will resolve two mdns services with different IP addresses. So currently the device should not be supposed to work with both wifi and thread interfaces enabled. You can disable one interface when you enable another one.

wqx6, 2024-07-26:
Hi, @supersunil, any further question about the multiple network interfaces?

svn-matterize, 2024-08-14:
Hey @wqx6 - Thanks for asking, I'll be trying your mentioned approach in the coming week and let you know.

dhrishi, 2024-11-13:
I am closing this now. Please re-open if you still face issues

closed 2024-11-13 by dhrishi