espressif/esp-matter #967: Error installing Matter

HamzaHajeir, 2024-06-13:
**Describe the bug**
Error at installing esp-matter for the first time, I've followed the documentations regarding installing esp-matter, but it fails with an error.

I've already cloned it recursively.

```

./esp-matter/install.sh

Running Matter Setup

  WELCOME TO...

         █
         █
     ▄   █   ▄                                █     █
     ▀▀█████▀▀      ▄▀▀▀▄ ▄▀▀▀▄    ▄▀▀▀▀▄█  ▀▀█▀▀▀▀▀█▀▀   ▄▀▀▀▀▄    ▄▀▀
   ▀█▄       ▄█▀   █     █     █  █      █    █     █    █▄▄▄▄▄▄█  █   
     ▀█▄   ▄█▀     █     █     █  █      █    █     █    █         █   
  ▄██▀▀█   █▀▀██▄  █     █     █   ▀▄▄▄▄▀█    ▀▄▄   ▀▄▄   ▀▄▄▄▄▀   █   
 ▀▀    █   █    ▀▀

  BOOTSTRAP! Bootstrap may take a few minutes; please be patient.

Downloading and installing packages into local source directory:

  Setting up CIPD package manager...done (4.5s)
  Setting up Project actions........skipped (0.1s)
  Setting up Python environment.....[/]
ERROR at //build/config/linux/pkg_config.gni:116:17: Script returned non-zero exit code.
    pkgresult = exec_script(pkg_config_script, args, "value")
                ^----------
Current dir: /home/hamza/arduino/esp32-arduino-lib-builder/esp-matter/connectedhomeip/connectedhomeip/.environment/gn_out/
Command: python3 /home/hamza/arduino/esp32-arduino-lib-builder/esp-matter/connectedhomeip/connectedhomeip/build/config/linux/pkg-config.py glib-2.0 gio-unix-2.0
Returned 1.
stderr:

Could not run pkg-config.

See //build/chip/linux/BUILD.gn:19:1: whence it was called.
pkg_config("glib") {
^-------------------
See //build/chip/linux/gdbus_library.gni:106:7: which caused the file to be included.
      glib_config,
      ^----------
['gn', 'gen', '/home/hamza/arduino/esp32-arduino-lib-builder/esp-matter/connectedhomeip/connectedhomeip/.environment/gn_out', '--args=chip_crypto="boringssl" dir_pigweed="/home/hamza/arduino/esp32-arduino-lib-builder/esp-matter/connectedhomeip/connectedhomeip/third_party/pigweed/repo"']

Traceback (most recent call last):
  File "/home/hamza/arduino/esp32-arduino-lib-builder/esp-matter/connectedhomeip/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/virtualenv_setup/install.py", line 402, in install_packages
    subprocess.check_call(
  File "/usr/lib/python3.10/subprocess.py", line 369, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['gn', 'gen', '/home/hamza/arduino/esp32-arduino-lib-builder/esp-matter/connectedhomeip/connectedhomeip/.environment/gn_out', '--args=chip_crypto="boringssl" dir_pigweed="/home/hamza/arduino/esp32-arduino-lib-builder/esp-matter/connectedhomeip/connectedhomeip/third_party/pigweed/repo"']' returned non-zero exit status 1.

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/hamza/arduino/esp32-arduino-lib-builder/esp-matter/connectedhomeip/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/env_setup.py", line 995, in <module>
    sys.exit(main())
  File "/home/hamza/arduino/esp32-arduino-lib-builder/esp-matter/connectedhomeip/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/env_setup.py", line 987, in main
    return EnvSetup(**vars(parse())).setup()
  File "/home/hamza/arduino/esp32-arduino-lib-builder/esp-matter/connectedhomeip/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/env_setup.py", line 577, in setup
    result = step(spin)
  File "/home/hamza/arduino/esp32-arduino-lib-builder/esp-matter/connectedhomeip/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/env_setup.py", line 766, in virtualenv
    if not virtualenv_setup.install(
  File "/home/hamza/arduino/esp32-arduino-lib-builder/esp-matter/connectedhomeip/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/virtualenv_setup/install.py", line 443, in install
    install_packages(gn_target)
  File "/home/hamza/arduino/esp32-arduino-lib-builder/esp-matter/connectedhomeip/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/virtualenv_setup/install.py", line 410, in install_packages
    raise subprocess.CalledProcessError(
subprocess.CalledProcessError: Command '['gn', 'gen', '/home/hamza/arduino/esp32-arduino-lib-builder/esp-matter/connectedhomeip/connectedhomeip/.environment/gn_out', '--args=chip_crypto="boringssl" dir_pigweed="/home/hamza/arduino/esp32-arduino-lib-builder/esp-matter/connectedhomeip/connectedhomeip/third_party/pigweed/repo"']' returned non-zero exit status 1.
```

**Environment**
- ESP-Matter Commit Id: latest main
- ESP-IDF Commit Id: 0b7221e
- SoC (eg: ESP32 or ESP32-C3): All.
- Device Logs (Please attach the log file):
- Host Machine OS: Ubuntu 22.04 LTS

**Any additional details**

Executing gn --version gives: 
> 1000 (0725d78)

sayondeep, 2024-06-14:
@HamzaHajeir Please intall the pre requisites as mentioned [here](https://docs.espressif.com/projects/esp-matter/en/latest/esp32/developing.html#id1).

HamzaHajeir, 2024-06-15:
> @HamzaHajeir Can you please confirm that you have installed the pre requisites as mentioned [here](https://docs.espressif.com/projects/esp-matter/en/latest/esp32/developing.html#id1).

Correct, that point was overlooked.

closed 2024-06-15 by HamzaHajeir