espressif/esp-matter #968: Cannot open ESP-MATTER examples is VSCode using ESP-IDF extension

aarbmx6s, 2024-06-13:
**Describe the bug**
After the successful installation of the SDK, the ESP-IDF extension cannot open them and runs into an error that the `esp-matter/examples/common/blemesh_platform/platform/ESP32_custom/WiFiDnssdImpl.cpp` file is not found. 

I installed the SDK into the project folder. I see the file in the explorer. But I cannot open it.

**Environment**
- Host Machine OS: macOS Sonoma 

**Any additional details**
The file is not a file but a symlink. 

[image]
[image]

I also created an issue for the original extension: https://github.com/espressif/vscode-esp-idf-extension/issues/1208

closed 2024-06-14 by espressif-bot

shripad621git, 2024-06-14:
@aarbmx6s, we have addressed the changes [here](https://github.com/espressif/esp-matter/commit/489d78196eceb8d4ddbfc6c099895300d4bf38b3). Can you please confirm and get back to us.

aarbmx6s, 2024-06-15:
@shripad621git, the examples page is displaying correctly. But, as the version was installed manually, because I didn't find any options on how to install the fixed version using the VSCode extension, I cannot build any examples. It always fails on different problems. However, I can build examples from the shell using `idf.py`  after exporting sources into the shell. I tried to configure the VSCode project by setting all needed paths in the `settings.json`, but it seems like the VSCode project setting at some point uses a different environment. Hence, it is a pain to use it all in VSCode, I see a lot of useful UI elements that can be used just by clicking, but it does not.

**How to friend then the manually installed MATTER SDK with the ESP SDK that was installed via VSCode?** 

This is the failed build output for the **Light Switch** example:
```
 *  Executing task: cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 -B /Users/svystun/work/esp/light_switch/build -S /Users/svystun/work/esp/light_switch 

-- Found Git: /usr/bin/git (found version "2.39.3 (Apple Git-146)")
IDF_VERSION: 5.2.2; idf version number: 5.2.2
SDKCONFIG_PATH: /Users/svystun/work/esp/light_switch/sdkconfig.defaults
-- Component directory /Users/svystun/esp/esp-matter/examples/common/blemesh_platform does not contain a CMakeLists.txt file. No component will be added
-- Component directory /Users/svystun/esp/esp-matter/examples/common/cmake_common does not contain a CMakeLists.txt file. No component will be added
-- Component directory /Users/svystun/esp/esp-matter/examples/common/external_platform does not contain a CMakeLists.txt file. No component will be added
-- Component directory /Users/svystun/esp/esp-matter/examples/common/relinker does not contain a CMakeLists.txt file. No component will be added
-- Component directory /Users/svystun/esp/esp-matter/examples/common/utils does not contain a CMakeLists.txt file. No component will be added
-- The C compiler identification is GNU 13.2.0
-- The CXX compiler identification is GNU 13.2.0
-- The ASM compiler identification is GNU
-- Found assembler: /Users/svystun/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/xtensa-esp32-elf-gcc
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /Users/svystun/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/xtensa-esp32-elf-gcc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /Users/svystun/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/xtensa-esp32-elf-g++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Building ESP-IDF components for target esp32
NOTICE: Skipping optional dependency: espressif/cmake_utilities
Processing 29 dependencies:
[1/29] espressif/button (3.2.0)
[2/29] espressif/cbor (0.6.0~1)
[3/29] espressif/cmake_utilities (0.5.3)
[4/29] espressif/esp-serial-flasher (0.0.11)
[5/29] espressif/esp_bsp_generic (1.2.0)
[6/29] espressif/esp_diag_data_store (1.0.1)
[7/29] espressif/esp_diagnostics (1.0.2)
[8/29] espressif/esp_encrypted_img (2.1.0)
[9/29] espressif/esp_insights (1.0.1)
[10/29] espressif/esp_lcd_gc9a01 (1.2.0)
[11/29] espressif/esp_lcd_ili9341 (1.2.0)
[12/29] espressif/esp_lcd_touch (1.1.2)
[13/29] espressif/esp_lcd_touch_cst816s (1.0.3)
[14/29] espressif/esp_lcd_touch_ft5x06 (1.0.6)
[15/29] espressif/esp_lcd_touch_gt1151 (1.0.5~1)
[16/29] espressif/esp_lcd_touch_gt911 (1.1.1)
[17/29] espressif/esp_lcd_touch_tt21100 (1.1.0)
[18/29] espressif/esp_lvgl_port (2.2.2)
[19/29] espressif/esp_rcp_update (1.0.2)
[20/29] espressif/esp_secure_cert_mgr (2.4.1)
[21/29] espressif/jsmn (1.1.0)
[22/29] espressif/json_generator (1.1.2)
[23/29] espressif/json_parser (1.0.3)
[24/29] espressif/led_indicator (0.9.2)
[25/29] espressif/led_strip (2.5.3)
[26/29] espressif/mdns (1.3.2)
[27/29] espressif/rmaker_common (1.4.5)
[28/29] idf (5.2.2)
[29/29] lvgl/lvgl (9.1.0)
-- Found Python3: /usr/local/Frameworks/Python.framework/Versions/3.11/bin/python3.11 (found version "3.11.9") found components: Interpreter
-- LVGL version: 9.1.0
NOTICE: Skipping optional dependency: espressif/cmake_utilities
-- Project sdkconfig file /Users/svystun/work/esp/light_switch/sdkconfig
warning: default on the choice symbol SEC_CERT_DAC_PROVIDER (defined at /Users/svystun/esp/esp-matter/connectedhomeip/connectedhomeip/config/esp32/components/chip/Kconfig:929, /Users/svystun/esp/esp-matter/components/esp_matter/Kconfig:48) will have no effect, as defaults do not affect choice symbols
warning: the choice symbol SEC_CERT_DAC_PROVIDER (defined at /Users/svystun/esp/esp-matter/connectedhomeip/connectedhomeip/config/esp32/components/chip/Kconfig:929, /Users/svystun/esp/esp-matter/components/esp_matter/Kconfig:48) is defined with a prompt outside the choice
Loading defaults file /Users/svystun/work/esp/light_switch/sdkconfig.defaults...
-- Compiler supported targets: xtensa-esp-elf
[... 162 lines cut ...]
-- Components: bootloader bootloader_support efuse esp_app_format esp_bootloader_format esp_common esp_hw_support esp_rom esp_system esptool_py freertos hal log main micro-ecc newlib partition_table soc spi_flash xtensa
[... cut ...]
  File "../../../../../../esp/esp-matter/connectedhomeip/connectedhomeip/config/esp32/third_party/connectedhomeip/build/chip/write_build_time_header.py", line 24
    def utc_time_in_matter_epoch_s(time: datetime):
                                       ^
SyntaxError: invalid syntax
...
```

For the **light** and **Room Air Conditioner** examples the same error.

I checked that the python version is being used the same when VSCode runs the building either the building is being ran from the shell.