espressif/esp-matter #969: Use Access Point along with Station Mode in ESP32 based Matter device

bilalmalik76, 2024-06-14:
**Describe the question/query that you have**
We want to use access point mode along with station mode on an ESP32-based Matter device. Is this possible? If yes, can we simply use ESP-IDF Wi-Fi APIs to enable the access point, or are there any other APIs available in the ESP-Matter SDK?

dhrishi, 2024-06-14:
>We want to use access point mode along with station mode on an ESP32-based Matter device. Is this possible? If yes, can we simply use ESP-IDF Wi-Fi APIs to enable the access point,

Yes and yes. Make sure that you enable `CONFIG_ESP_WIFI_SOFTAP_SUPPORT` through the menuconfig and then use the SoftAP APIs

bilalmalik76, 2024-06-14:
@dhrishi Hi,  we just need to enable this in defconfig CONFIG_ESP_WIFI_SOFTAP_SUPPORT=y and then we can normally use  SoftAP APIs available in esp-idf right?

bilalmalik76, 2024-06-15:
@dhrishi Hi, I tried by add this  CONFIG_ESP_WIFI_SOFTAP_SUPPORT=y in sdkconfig, But network management implmentation change its state from APSTA to STA. Please check this:
I (9681) chip[DL]: Done driving station state, nothing else to do...
I (9681) esp_netif_lwip: DHCP server started on interface WIFI_AP_DEF with IP: 192.168.4.1
I (9681) StarPower Main: Commissioning window closed
I (9701) chip[DL]: WIFI_EVENT_AP_START
I (9711) chip[DL]: WiFi AP state change: NotActive -> Active
I (9711) chip[DL]: Changing ESP WiFi mode: STA+AP -> STA
I (9721) wifi:mode : sta (80:65:99:c9:cd:94)
I (9731) chip[DL]: WiFi AP state change: Active -> Deactivating
I (9731) chip[DL]: WIFI_EVENT_AP_STOP

shubhamdp, 2024-06-19:
Can you check config options in menu `(Top) → Component config → CHIP Device Layer → WiFi AP Options` are enabled? If not please enable them and give a try.

bilalmalik76, 2024-06-19:
@shubhamdp Actally it works with this, First set mode to AP and then enable it. 
`    CHIP_ERROR err_chip = chip::DeviceLayer::Internal::ESP32Utils::SetAPMode(true);
    if (err_chip != CHIP_NO_ERROR)
    {
        ESP_LOGE(TAG, "Failed to Set APMode");
    }

    ConnectivityMgr().SetWiFiAPMode(ConnectivityManager::kWiFiAPMode_Enabled);`

But it sets the AP mode SSID and Password to default value and open, how can we change it SSID and use password instead of open?

shubhamdp, 2024-06-19:
For now you will have to change it here: https://github.com/project-chip/connectedhomeip/blob/master/src/platform/ESP32/ConnectivityManagerImpl_WiFi.cpp#L948 or use the esp-idf raw APIs to perform what you want to do.

As the AP based commissioning mode is not supported in Matter hence its not really being maintained.

bilalmalik76, 2024-06-19:
Ok Thanks, we are not using AP mode for commissioning but to configure our device according to end user requirements with a web interface.

shubhamdp, 2024-06-25:
Okay, If I understood correctly, then AP mode is only required when device is not commissioned.

bilalmalik76, 2024-06-27:
@shubhamdp we want to use AP mode anytime before or after commissioning.

jonsmirl, 2024-07-16:
You can implement custom Matter clusters to support any of your proprietary configuration needs. Doing it that way is secure.

closed 2024-10-20 by bilalmalik76