espressif/esp-matter #970: [TC-OPCREDS-3.6 test fails

tanyuxinfree, 2024-06-15:
when I test TC-OPCREDS-3.6,"TH1 fully commissions the DUT" is ok, "TH1 sends RemoveFabric command with Fabric Index as FabricIndex_TH1 to DUT" is ok, "Verify that the DUT is not advertising as an operational node on DNS-SD with service type _matter._tcp" is ok, but "If the DUT is advertising as a commissionable node with CM=2 over DNS-SD or advertising the Matter service over BLE, TH commissions the device" faild

tanyuxinfree, 2024-06-15:
The expected result is "CM=0"
[image]
But the actual result is "CM=1"
[image]

tanyuxinfree, 2024-06-15:
in example "light", if fabric is 0,CM=1
[image]
So what can I do to make fabric=1 and CM=0?

tanyuxinfree, 2024-06-15:
esp-idf commit :release/v5.2
esp-matter commit :releasev1.2

shubhamdp, 2024-06-19:
@tanyuxinfree  This code you just pointed out, is part of an application, and you can remove the call to `OpenBasicCommissioningWindow()` to make the `CM=1` to `CM=0`.

If you don't see the `CM=0`, then please check this text from the spec.
> The absence of key CM SHALL imply a value of 0 (CM=0).

- CM=0 means device is not in commissioning mode
- CM=1 means device is in commissioning mode and can be commissioned using passcode printed on the packaging
- CM=2 means dynamically generated passcode shall be used (i.e. multi admin scenario case)

shubhamdp, 2024-06-25:
@tanyuxinfree any updates on this one?

dhrishi, 2024-07-26:
Closing this now. Please re-open if you are still seeing this

closed 2024-07-26 by dhrishi