espressif/esp-matter #972: Battery example

Kristian8606, 2024-06-16:
I have an esp32h2 that I loaded the battery example on. I removed the LED so that it does not consume, but the consumption is 3 milliamps. please tell me how to reduce the consumption to >1 milliamp or more if possible.
Sorry for my bad english

wqx6, 2024-06-17:
If you want to develop a battery-power example on ESP32-H2, you can refer to the [icd_app](https://github.com/espressif/esp-matter/tree/main/examples/icd_app) example. The example will enable power management and light sleep on ESP32-H2/ESP32-C6. And the current will be about 100uA on ESP32-H2.

Kristian8606, 2024-06-17:
Excuse my bad English. I wanted to say that exactly this example I use the consumption is about 3 milliamps. Can you share if you managed to reduce the consumption to about 100uA

Kristian8606, 2024-06-17:
@wqx6 Do I need to make any adjustments to the example settings as the consumption is far higher than you mentioned.

Kristian8606, 2024-06-17:
Where can I find more information about these settings and will they lower the consumption?

```# Enable ICD server
CONFIG_ENABLE_ICD_SERVER=y
# ICD Active mode interval(ms)
CONFIG_ICD_ACTIVE_MODE_INTERVAL_MS=1000
# ICD Active mode threshold(ms)
CONFIG_ICD_ACTIVE_MODE_THRESHOLD_MS=1000
# ICD Idle mode interval(s)
CONFIG_ICD_IDLE_MODE_INTERVAL_SEC=60
# ICD Fast Poll interval(ms)
CONFIG_ICD_FAST_POLL_INTERVAL_MS=500
# ICD Slow Poll interval(ms)
CONFIG_ICD_SLOW_POLL_INTERVAL_MS=5000
```

Kristian8606, 2024-06-22:
Does anyone know what I can turn off or adjust to reduce consumption?

wqx6, 2024-06-27:
@Kristian8606 Sorry for the late reply. The 100uA is only the current of chip. Considering that there are still some other elements on the devkit board, the current of the whole devkit board will be higher than the chip. In our device kit, there should be always two pins (J5 in the following picture) which can be used for current measurement. 
[image]

The ICD configurations only affects the behaviors of light sleep. To enable the light sleep you should enable power management in menuconfig, you could refer to the [config file](https://github.com/espressif/esp-matter/blob/main/examples/icd_app/sdkconfig.defaults.esp32h2#L59-L63) in icd_app to enable light sleep on your device.

wqx6, 2024-06-27:
You could run the icd example and measure the current for that example firstly. And then design your own example.

Kristian8606, 2024-06-27:
Thanks for the reply, but did you measure the current on the chip? Here's a little background. I had the same board as in the picture, when I measured the current at J5 I measured 65uA which is fine for me, but when I powered up and measured the current through the VBAT pin there the consumption is around 3-4mA. I assumed that this consumption was due to the other components of the devkit board. For that I sacrificed the board and unsoldered the chip from the devkit board, soldered thin wires to VBAT GND and GPIO_EN to 3.3v and the chip started with the icd_app example but the current was 3mA again. What else can I do to reduce the consumption to >1ma.

wqx6, 2024-07-02:
Did you finish commissioning when you measured the current after unsoldering the chip from the devkit board? The board has a 32.768kHz crystal which is required for BLE sleep. The ICD example will disable BLE after commissioning so the chip can still sleep if it is commissioned. But if the chip is not commissioned and there is not 32kHz crystal, the chip cannot sleep and the current will be much higher.

Kristian8606, 2024-07-02:
@wqx6 Yes, the example is added to the Apple home. on/off switch and it shows high consumption when I switch. Also, when I switch, there is a delay of about 5 seconds, which tells me that the chip is sleeping, but the consumption cannot be less than 3mA

Kristian8606, 2024-07-02:
It's worth noting that the light_sleep example from esp-idf maintains an average draw of around 300uA, which is fine for my needs and makes me think that esp-matter needs some configuration or tuning. you can measure your average consumption via PIN VBAT?

wqx6, 2024-07-26:
Did you enable this option for your example `CONFIG_USE_BLE_ONLY_FOR_COMMISSIONING=y`? For ESP32-H2 without the 32kHz crystal(since you have already unsoldered the chip from the devkit board), the BLE must be deinitialized for light sleep. the option `USE_BLE_ONLY_FOR_COMMISSIONING` will deinitialize the BLE after commissioning.

dhrishi, 2024-09-26:
@Kristian8606 Please close the issue if resolved

Kristian8606, 2024-09-26:
The problem was solved by replacing the old h2 and c6 with new ones. I don't know if the new ones have hardware changes but the new ones now sleep at the same settings while the old ones don't.

closed 2024-09-26 by Kristian8606