espressif/esp-matter #976: "templates not found" when installing

JetForMe, 2024-06-18:
**Describe the bug**

Following these [instructions](https://docs.espressif.com/projects/esp-matter/en/latest/esp32c6/developing.html#esp-matter-repository) to install the Matter support, I get a series of warnings:

```
19:08:45 esp/esp-matter/  (main) % cd ./connectedhomeip/connectedhomeip
19:08:48 connectedhomeip/connectedhomeip/ ((d83caf7fefea...)) % ./scripts/checkout_submodules.py --platform esp32 darwin --shallow
Checking out: nlassert, nlio, nlunit-test, mbedtls, qrcode, m5stack-tft, pigweed, openthread, nanopb, freertos, third_party/pybind11/repo, third_party/jsoncpp/repo, editline, third_party/boringssl/repo/src, third_party/libwebsockets/repo, perfetto
warning: templates not found in /usr/share/git-core/templates
warning: templates not found in /usr/share/git-core/templates
warning: templates not found in /usr/share/git-core/templates
warning: templates not found in /usr/share/git-core/templates
warning: templates not found in /usr/share/git-core/templates
warning: templates not found in /usr/share/git-core/templates
warning: templates not found in /usr/share/git-core/templates
warning: templates not found in /usr/share/git-core/templates
warning: templates not found in /usr/share/git-core/templates
warning: templates not found in /usr/share/git-core/templates
warning: templates not found in /usr/share/git-core/templates
warning: templates not found in /usr/share/git-core/templates
warning: templates not found in /usr/share/git-core/templates
warning: templates not found in /usr/share/git-core/templates
warning: templates not found in /usr/share/git-core/templates
warning: templates not found in /usr/share/git-core/templates
19:09:43 connectedhomeip/connectedhomeip/ ((d83caf7fefea...)) % 
```

**Environment**
- ESP-Matter Commit Id: d83caf7fefea
- ESP-IDF Commit Id: v5.2.2
- SoC (eg: ESP32 or ESP32-C3): ESP32-C6
- Device Logs (Please attach the log file):
- Host Machine OS: macOS 14.5
- Commissioner app and versions if present:
- Commissioner's logs if present:

shubhamdp, 2024-06-18:
After doing a simple google search I see few others facing the similar problem, can you please check if answers from https://stackoverflow.com/questions/34877649/warning-templates-not-found-usr-local-git-share-git-core-templates can be helpful?

JetForMe, 2024-06-19:
Virtually all of the google results refer to SourceTree, which I don't have installed. I'm using the standard command-line tool, and I don't experience this with any other repos.