espressif/esp-matter #977: Installation failed, unclear why (FAILED: python/gen/matter_build_venv/compiled_requirements.txt)

JetForMe, 2024-06-18:
**Describe the bug**
I'm running through the installation [instructions](https://docs.espressif.com/projects/esp-matter/en/latest/esp32c6/developing.html#esp-matter-repository) for Matter support, and ran into an error I don't understand. You can see the lengthy transcript [here](https://docs.espressif.com/projects/esp-matter/en/latest/esp32c6/developing.html#esp-matter-repository). Excerpt:

```
[511/1099] python3 ../../third_party/pigweed/repo/pw_build/py/pw_build/python_runner.py --gn-root ../../ --current-path ../.. --default-toolchain=//build/toolchain/host:mac_arm64_gcc --current-toolchain=//third_party/pigweed/repo/pw_build/python_toolchain:python --capture-output --module piptools --python-virtualenv-config python/gen/matter_build_venv/venv_metadata.json --python-dep-list-files python/gen/matter_build_venv._compile_requirements_metadata_path_list.txt -- compile --resolver=backtracking --allow-unsafe --output-file python/gen/matter_build_venv/compiled_requirements.txt python/gen/matter_build_venv/generated_requirements.txt ../../scripts/setup/requirements.build.txt
FAILED: python/gen/matter_build_venv/compiled_requirements.txt 
python3 ../../third_party/pigweed/repo/pw_build/py/pw_build/python_runner.py --gn-root ../../ --current-path ../.. --default-toolchain=//build/toolchain/host:mac_arm64_gcc --current-toolchain=//third_party/pigweed/repo/pw_build/python_toolchain:python --capture-output --module piptools --python-virtualenv-config python/gen/matter_build_venv/venv_metadata.json --python-dep-list-files python/gen/matter_build_venv._compile_requirements_metadata_path_list.txt -- compile --resolver=backtracking --allow-unsafe --output-file python/gen/matter_build_venv/compiled_requirements.txt python/gen/matter_build_venv/generated_requirements.txt ../../scripts/setup/requirements.build.txt
    error: subprocess-exited-with-error
    
    × Getting requirements to build wheel did not run successfully.
```

**Environment**
- ESP-Matter Commit Id: d83caf7fefea
- ESP-IDF Commit Id: v5.2.2
- SoC (eg: ESP32 or ESP32-C3):
- Device Logs (Please attach the log file):
- Host Machine OS:
- Commissioner app and versions if present:
- Commissioner's logs if present:

**Any additional details**
...

9zigen, 2024-06-20:
I have the same problem. After downgrading Python version from 3.12 to 3.9 it works.

ntwyman, 2024-06-23:
I also had to downgrade python 3.12 to 3.9.2 to get past issues installing esp-matter on an M1 Mac.

farhan-syed, 2024-06-25:
Also would like to add incase someone sees this and still is getting the issue even though you have 3.9. May need to install Rosetta 2 if you don't already have it. Installing Rosetta fixed it for me.

JetForMe, 2024-06-28:
> @JetForMe can you please provide the complete failure logs and what python version are you using?

Oh I'm sorry, I didn't realize I had the wrong link in there. I've edited the original comment, but you can find it here: https://gist.github.com/JetForMe/f410ac9057d2a6c0990f2f53c04c6915

```
% python --version
Python 3.12.3
```

retsyx, 2024-07-15:
Espressif, please fix this. Having to downgrade python to 3.9 is not a solution.

sldr, 2024-07-29:
Another workaround that worked for me is:
```bash
echo "cython<3" > /tmp/constraint.txt
export PIP_CONSTRAINT=/tmp/constraint.txt
cd esp-matter
./install.sh
```
**Note:**
* This seems to get you down the road to another module using imp instead of importlib. I think egg or maybe python has made me hangry.
* Credit for find is really in [pyyaml issue commant](https://github.com/yaml/pyyaml/issues/601#issuecomment-1813963845)
Laters,
SLDR
(Stephen L. De Rudder)

shubhamdp, 2024-07-30:
@JetForMe @retsyx `main` branch was forked into `release/v1.3` and we have updated the submodule for connectedhomeip to [ae52f4e597cd79e956142009b82eb700a0c916f1](https://github.com/espressif/connectedhomeip/commits/ae52f4e597cd79e956142009b82eb700a0c916f1), it has the fix for working with Python 3.12.

Can you folks please give it a try on release/v1.3 branch.

JetForMe, 2024-07-31:
Yes, it seems to work. I followed the instructions [here](https://docs.espressif.com/projects/esp-matter/en/latest/esp32/developing.html#getting-the-repositories), except I did the following:

```
% git clone  https://github.com/espressif/esp-matter.git 
% git switch release/v1.3
% git submodule update --init --depth 1
% cd ./connectedhomeip/connectedhomeip
% ./scripts/checkout_submodules.py --platform esp32 darwin --shallow
% cd ../..
% ./install.sh

Running Matter Setup

  WELCOME TO...

         █
         █
     ▄   █   ▄                                █     █
     ▀▀█████▀▀      ▄▀▀▀▄ ▄▀▀▀▄    ▄▀▀▀▀▄█  ▀▀█▀▀▀▀▀█▀▀   ▄▀▀▀▀▄    ▄▀▀
   ▀█▄       ▄█▀   █     █     █  █      █    █     █    █▄▄▄▄▄▄█  █   
     ▀█▄   ▄█▀     █     █     █  █      █    █     █    █         █   
  ▄██▀▀█   █▀▀██▄  █     █     █   ▀▄▄▄▄▀█    ▀▄▄   ▀▄▄   ▀▄▄▄▄▀   █   
 ▀▀    █   █    ▀▀

  BOOTSTRAP! Bootstrap may take a few minutes; please be patient.

Downloading and installing packages into local source directory:

  Setting up CIPD package manager...done (35.9s)
  Setting up Project actions........skipped (0.1s)
  Setting up Python environment.....done (1m8.7s)
  Setting up pw packages............skipped (0.1s)
  Setting up Host tools.............done (0.1s)

Activating environment (setting environment variables):

  Setting environment variables for CIPD package manager...done
  Setting environment variables for Project actions........skipped
  Setting environment variables for Python environment.....done
  Setting environment variables for pw packages............skipped
  Setting environment variables for Host tools.............done

Checking the environment:

20240731 01:37:19 INF Environment passes all checks!

Environment looks good, you are ready to go!

To reactivate this environment in the future, run this in your 
terminal:

  source ./activate.sh

To deactivate this environment, run this:

  deactivate

Installing pip requirements for all...

[notice] A new release of pip is available: 23.2.1 -> 24.2
[notice] To update, run: pip install --upgrade pip

Building host tools

Done. Made 7794 targets from 368 files in 332ms
ninja: Entering directory `/Users/rmann/Projects/Personal/ESP32-C6/esp/esp-matter/connectedhomeip/connectedhomeip/out/host'
[885/885] ld ./chip-tool
ld: warning: ignoring duplicate libraries: 'liblogging-stdio.a'

Host tools built at: /Users/rmann/Projects/Personal/ESP32-C6/esp/esp-matter/connectedhomeip/connectedhomeip/out/host
[... 84 lines cut ...]
    Uninstalling pycparser-2.22:
      Successfully uninstalled pycparser-2.22
  Attempting uninstall: ecdsa
    Found existing installation: ecdsa 0.19.0
    Uninstalling ecdsa-0.19.0:
      Successfully uninstalled ecdsa-0.19.0
  Attempting uninstall: cffi
    Found existing installation: cffi 1.16.0
    Uninstalling cffi-1.16.0:
      Successfully uninstalled cffi-1.16.0
  Attempting uninstall: cryptography
    Found existing installation: cryptography 41.0.7
    Uninstalling cryptography-41.0.7:
      Successfully uninstalled cryptography-41.0.7
Successfully installed MarkupSafe-2.1.5 PyQRCode-1.2.1 cffi-1.15.0 cryptography-36.0.0 ecdsa-0.18.0 esp-matter-mfg-tool-1.0.2 esp-secure-cert-tool-1.0.1 future-0.18.2 jinja2-3.0.1 lark-1.1.2 pycparser-2.21 pypng-0.0.21 python-stdnum-1.18 stringcase-1.2.0

[notice] A new release of pip is available: 24.0 -> 24.2
[notice] To update, run: pip install --upgrade pip
All done! You can now run:

  . ./export.sh

```

Not sure why `main` doesn’t include the changes, but at least this works. Hopefully I'll find some time to actually build an example.

sj82516, 2024-08-07:
@JetForMe May I ask what is your mac os version ? 
I followed your comments but encountered this error
```
ninja: Entering directory `/Users/yuanchieh-at/Desktop/esp-matter/connectedhomeip/connectedhomeip/out/host'
[3/4] ld ./chip-cert
FAILED: chip-cert chip-cert.map
g++ -target arm64-apple-macos11.0 -O0 -fPIC -Werror -fdiagnostics-color -Wl,-dead_strip -Wl,-map,./chip-cert.map @./chip-cert.rsp -o ./chip-cert
ld: multiple errors: archive member '/' not a mach-o file in '/Users/yuanchieh-at/Desktop/esp-matter/connectedhomeip/connectedhomeip/out/host/obj/src/crypto/lib/libChipCrypto.a'; archive member '/' not a mach-o file i .......
```

Do you have any idea about that ?

JetForMe, 2024-08-08:
> Do you have any idea about that ?

Huh,  not really.  It seems to have failed to build `libChipCrypto.a` correctly, but I couldn't say why. I'm on macOS 14.5, with Xcode 16b5 (although when I tried this it was probably b2). I would guess the `g++` invoked is whatever GCC the Espressif toolchain builds for RISC-V, so I would assume that’s the same for you and me, but I don’t know enough to quickly check, and can’t spend the time to poke through their toolchain right now.

sj82516, 2024-08-08:
I see. Thanks for your reply @JetForMe

shubhamdp, 2024-08-09:
> Yes, it seems to work. I followed the instructions [here](https://docs.espressif.com/projects/esp-matter/en/latest/esp32/developing.html#getting-the-repositories), except I did the following:
>
> ```
> % [...]

We are in the process of updating the connectedhomeip submodule, and it will be done soon...

dhrishi, 2024-08-14:
The connectedhomeip submodule is updated

closed 2024-08-14 by dhrishi