espressif/esp-matter #978: The endpoint device is not displayed in the Apple app

maenkai, 2024-06-18:
**Describe the bug**
1. Matter bridge device create and commision
2. The endpoint device is not displayed in the Apple app
3. chiptool can look up the endpoint
4. Restart homepod or restart the Matter bridge device
5. Still have problems
6. Appple app re-commision the bridge device
7. Have a chance of recovery

**Environment**
- ESP-Matter Commit Id:1a35c738dff33e5e988d92d0028ed7a60d4336f6 (HEAD -> release/v1.2, origin/release/v1.2)
- ESP-IDF Commit Id: e088c3766ba440e72268b458a68f27b6e7d63986 (HEAD, tag: v5.1.1)
- SoC (eg: ESP32 or ESP32-C3): ESP32
- Device Logs (Please attach the log file):
[chiptool.log](https://github.com/user-attachments/files/15881124/chiptool.log)

- Host Machine OS: Ubuntu
- Commissioner app and versions if present: Apple 17.1 HomePod 17.5
- Commissioner's logs if present: null

**Any additional details**
...

jadhavrohit924, 2024-06-21:
@maenkai Its working completely fine for me. Can you please try again by updated iOS?
My iPadOS version is 18.0 and HomePod 17.5

jonsmirl, 2024-06-23:
Working for me on iPadOS 17 Public Beta, Home Pod 16.6.

If there is something wrong with your endpoint it may not display in IOS. For example I messed up the TLV in a custom cluster and that caused by endpoint to disappear.

dhrishi, 2024-07-26:
@maenkai Are you still seeing this issue?

closed 2024-09-04 by maenkai