espressif/esp-matter #982: Bridges and child endpoints.

jonsmirl, 2024-06-23:
What is the plan to save/restore bridged devices with child endpoints? For example a thermostat with embedded sensors.  I restored the thermostat, fff30008. Then when it created the child sensor EPs ( fff30005, fff30005, fff30005). Then parent of the sensor EPs is the thermostat which of course does not contain an aggregator.

Should this error just be removed? Or should I create the child endpoints bypassing the bridge code?
~~~
I (2846) device: EnumeratedEP fff30008 0
I (2846) device: bridge_create_ep
I (2846) device: EnumeratedEP fff30005 0
I (2847) device: bridge_create_ep
I (2847) device: create_ep_by_dt  fff30005
I (2847) device: parent_endpointId dt fff30008 id 0 agg 1
E (2848) esp_matter_bridge: The device types of the parent endpoint must include aggregator
E (2848) esp_matter_bridge: Parent endpoint is invalid
E (2849) device: Failed to create the bridged device
I (2849) device: EnumeratedEP fff30006 0
I (2849) device: bridge_create_ep
I (2850) device: create_ep_by_dt  fff30006
I (2850) device: parent_endpointId dt fff30008 id 0 agg 1
E (2850) esp_matter_bridge: The device types of the parent endpoint must include aggregator
E (2851) esp_matter_bridge: Parent endpoint is invalid
E (2851) device: Failed to create the bridged device
I (2851) device: EnumeratedEP fff30007 0
I (2852) device: bridge_create_ep
I (2852) device: create_ep_by_dt  fff30007
I (2852) device: parent_endpointId dt fff30008 id 0 agg 1
E (2853) esp_matter_bridge: The device types of the parent endpoint must include aggregator
E (2853) esp_matter_bridge: Parent endpoint is invalid
E (2853) device: Failed to create the bridged device
I (2854) device: EnumeratedEP fff30002 1

~~~

jonsmirl, 2024-06-23:
I read through the spec and I believe child endpoints on bridged devices should not have the BridgedBasicInfo cluster. So I added my own code to dynamically make the child endpoint and set the parent.
~~~
static void create_child_ep(Ep_Base *ep, Ep_Base *parent) {
    descriptor::config_t descriptor_config;
    endpoint_t *endpoint = endpoint::create(node::get(), ENDPOINT_FLAG_NONE, ep);
    descriptor::create(endpoint, &descriptor_config, CLUSTER_FLAG_SERVER);
    ep->init_bridged(endpoint);
    set_parent_endpoint(endpoint, parent->getEndpoint());
}
~~~

wqx6, 2024-06-27:
Do you want to create the following the data model?
```
RootNode-----Aggregator------Thermostat-------Sensor1
                                           |     ...
                                           ---SensorN
```
Currently we don't have any plans on restoring the bridged endpoints with child endpoints.
If you wants to create the child endpoint for bridged endpoints, you can `endpoint::create()` and `endpoint::set_parent_endpoint()`. The `esp_matter_bridge::create_device()` is only for the bridged endpoints not the child endpoints of bridged endpoints.

jonsmirl, 2024-06-27:
It is that model. I think I have it working now by using endpoint::create() directly.

closed 2024-06-27 by jonsmirl