espressif/esp-matter #984: how QR gen

cybortd, 2024-06-27:
hi guys. I have a question about how QR code gen from these information:
Version:             0
Vendor ID:           65521    (0xFFF1)
ProductID:           32768    (0x8000)
Custom flow:         0        (STANDARD)
Discovery Bitmask:   0x02     (BLE)
Long discriminator:  3840     (0xf00)
Passcode:            20202021

to  MT:Y.K9042C00KA0648G00 
 thanks

shubhamdp, 2024-06-27:
https://docs.espressif.com/projects/esp-matter/en/latest/esp32/faq.html#a1-10-how-to-generate-matter-onboarding-codes-qr-code-and-manual-pairing-code

Please take a look at this FAQ.

NOTE: `./generate_setup_payload.py` is not present anymore and is replaced by [SetupPayload.py](https://github.com/project-chip/connectedhomeip/tree/master/src/setup_payload/python). Will fix that in that FAQ.

cybortd, 2024-06-27:
I mean how QR is created by those information ?
what is related between those infor and  MT:Y.K9042C00KA0648G00

shubhamdp, 2024-06-27:
Please check section **5.1. Onboarding Payload** in main Matter specification. Specification can be found here: https://csa-iot.org/developer-resource/specifications-download-request/ Or take a look at implementation of spec here: https://github.com/project-chip/connectedhomeip/tree/master/src/setup_payload

closed 2024-06-27 by cybortd