espressif/esp-matter #985: How to integrate Zigbee Bridge and Thread Border Router functions on ESP32-S3 board

Askidea, 2024-06-27:
**Describe the question/query that you have**

I am working on merging the example codes from the `esp-matter` and `esp-thread-br` repositories to make the **Matter Zigbee Bridge** function and the **Thread Border Router** function operate **simultaneously** on the ESP32-S3 board. Based on the `zigbee_bridge` code in `esp-matter`, I merged what I thought was necessary among the `basic_thread_border_router` sources in `esp-thread-br.` The ESP32-S3 was tested in both ways: using a Border router devkit board (Zigbee gateway board) and using a stand alone board.

I succeeded in integrating the source code and building it, but an error occurred while running. This error occurred while initializing the openthread stack inside the function while calling `esp_matter::start()` in `app_main()` of `zigbee_bridge`. The cause was found to be related to the enabled/disabled setting of "**Top -> Component config -> OpenThread**" in menuconfig.
 The _OpenThread_ menu item was not checked in `zigbee_bridge`'s menuconfig, but this item had to be checked in `basic_thread_border_router`, so a conflict was occurring.

**Environment**
- ESP-Matter Commit Id: 6410ea9d
- ESP-IDF Commit Id: 3d813af
- esp-thread-br Commit Id: 54b9549
- SoC : ESP32-S3 & H2, Zigbee gateway devkit (Border router devkit)
- Device Logs (Please attach the log file):
- Host Machine OS: Ubuntu 22.04 LTS
- Commissioner app and versions if present:
- Commissioner's logs if present:

**Additional context**

I would like to ask a few questions regarding this.
-----------------------------
Q1. Is it possible to have the Matter Zigbee Bridge task and the Thread Border Router task run simultaneously on the ESP32-S3 board?

-----------------------------
Q2. When building the zigbee_bridge source by merging the basic_thread_border_router source in the esp-matter repository and running it on the Zigbee gateway devkit (S3 with 4MB flash), the following error occurred. Is there any way to solve this?

```
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x40375a11
0x40375a11: esp_restart_noos at /home/kskim/github/espressif/esp-idf/components/esp_system/port/soc/esp32s3/system_internal.c:158

SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce2810,len:0x178c
load:0x403c8700,len:0x4
load:0x403c8704,len:0xc10
load:0x403cb700,len:0x2dac
entry 0x403c8904
I (26) boot: ESP-IDF v5.4-dev-12-g3d813afa01-dirty 2nd stage bootloader
I (26) boot: compile time Jun 25 2024 20:43:40
I (26) boot: Multicore bootloader
I (27) boot: chip revision: v0.2
I (27) boot.esp32s3: Boot SPI Speed : 80MHz
I (27) boot.esp32s3: SPI Mode       : DIO
I (27) boot.esp32s3: SPI Flash Size : 4MB
I (27) boot: Enabling RNG early entropy source...
I (28) boot: Partition Table:
I (28) boot: ## Label            Usage          Type ST Offset   Length
I (28) boot:  0 esp_secure_cert  unknown          3f 06 0000d000 00002000
I (29) boot:  1 nvs              WiFi data        01 02 00010000 0000e000
I (29) boot:  2 nvs_keys         NVS keys         01 04 0001e000 00001000
I (29) boot:  3 otadata          OTA data         01 00 0001f000 00002000
I (30) boot:  4 phy_init         RF data          01 01 00021000 00001000
I (30) boot:  5 factory          factory app      00 00 00030000 00271000
I (31) boot:  6 fctry            WiFi data        01 02 002a1000 00006000
I (31) boot:  7 zb_storage       Unknown data     01 81 002a7000 00020000
I (31) boot:  8 zb_fct           Unknown data     01 81 002c7000 00001000
I (32) boot:  9 web_storage      Unknown data     01 82 002c8000 00019000
I (32) boot: 10 rcp_fw           Unknown data     01 82 002e1000 000a0000
I (33) boot: End of partition table
I (33) boot: Defaulting to factory image
I (33) esp_image: segment 0: paddr=00030020 vaddr=3c1d0020 size=77d88h (490888) map
I (122) esp_image: segment 1: paddr=000a7db0 vaddr=3fc9ca00 size=06250h ( 25168) load
I (127) esp_image: segment 2: paddr=000ae008 vaddr=40374000 size=02010h (  8208) load
I (129) esp_image: segment 3: paddr=000b0020 vaddr=42000020 size=1ca57ch (1877372) map
I (466) esp_image: segment 4: paddr=0027a5a4 vaddr=40376010 size=169a0h ( 92576) load
I (497) boot: Loaded app from partition at offset 0x30000
I (498) boot: Disabling RNG early entropy source...
[... 100 lines cut ...]
load:0x3fce2810,len:0x178c
load:0x403c8700,len:0x4
load:0x403c8704,len:0xc10
load:0x403cb700,len:0x2dac
entry 0x403c8904
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x403759cd
0x403759cd: esp_restart_noos at /home/kskim/github/espressif/esp-idf/components/esp_system/port/soc/esp32s3/system_internal.c:158

SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce2810,len:0x178c
load:0x403c8700,len:0x4
load:0x403c8704,len:0xc10
load:0x403cb700,len:0x2dac
entry 0x403c8904
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x403759cd
0x403759cd: esp_restart_noos at /home/kskim/github/espressif/esp-idf/components/esp_system/port/soc/esp32s3/system_internal.c:158

```

wqx6, 2024-06-27:
We have run the ZigBee Gateway(Bridge) and Thread Border Router simultaneously on one S3 + one H2 board with an internal branch, but there are still some MRs to be merged in ESP-IDF about spinel changes and multi PAN support for this feature. Note that there is a limitation for running Zigbee Gateway and border router with only one H2, the channel for Thread network and zigbee network should be the same as there is only one radio on H2.

Askidea, 2024-06-27:
@wqx6 
Couldn't I refer to the internal branch you mentioned? It seems that simply merging the `basic_thread_border_router` code with the `zigbee_bridge` source code does not solve the problem, and deep knowledge of the ESP SDK is needed to solve this problem.

wqx6, 2024-06-28:
You could not simply merging the basic_thread_border_router code and configurations with the zigbee_bridge to use them simultaneously, since both the ZigbeeBridge and TBR will use ESP32-H2 as RCP, but the RCP firmware on ESP32-H2 for Zigbee bridge is different from the one for TBR, we are going to release a new RCP firmware based on Spinel protocol and the RCP can be use for both the Zigbee bridge and TBR. and then we will make the host of Zigbee bridge uses Spinel to transmit the Zigbee packets to the RCP. Currently it is not Spinel.

wqx6, 2024-06-28:
If you want to require the branch that can be used for integrating TBR and Zigbee Bridge, you can reach our business team for it https://www.espressif.com/en/contact-us/technical-inquiries

Askidea, 2024-06-28:
> You could not simply merging the basic_thread_border_router code and configurations with the zigbee_bridge to use them simultaneously, since both the ZigbeeBridge and TBR will use ESP32-H2 as RCP, [...]

I tried connecting `ot_rcp`(esp-idf v5.1) running on H2 to each Zigbee bridge(esp-matter & idf v5.4) and `basic_thread_border_router`(esp-thread-br & idf v5.4). And I confirmed that Matter and Zigbee devices were controlled. Looking at this, it seems that the Zigbee bridge also supports the Spinel interface.

wqx6, 2024-07-02:
Sorry, I forgot that the Zigbee Bridge has switched to using `ot_rcp` now. However, to use one rcp for Zigbee Bridge and TBR, there is still a feature named Multi-PAN which is required for integrating TBR and Zigbee Bridge. With that feature, the RCP will be able to know whether the packets transferred by Spinel are for Thread or Zigbee and the host chip will be able to transfer Zigbee&Thread with Sipnel on single UART port.

jonsmirl, 2024-07-03:
Do note that most of the Google Home, Apple Home, Amazon Alexa devices contain a Thread Border router so you might not need to build one.

dhrishi, 2024-07-26:
@Askidea Please close this if your questions are answered

closed 2024-07-29 by Askidea

omersiar, 2026-02-14:
Hi,

Any ETA for Multi-PAN support?